Dr-Gigavolt/dbus-aggregate-batteries #1: Does not find my batteries

pos-ei-don, 2022-12-28:
Great, thank you for all your work!
Wantet to try it, but unfortunately i cannot get it to find my batteries.

There is also no log which i can find. no ./aggregatebatteries.log file.

I do have 2 jk bms connected. Checking the values with mqtt i see this:

N/1234/system/0/ServiceMapping/com_victronenergy_battery_1=com.victronenergy.battery.ttyUSB1
N/1234/system/0/ServiceMapping/com_victronenergy_battery_2=com.victronenergy.battery.ttyUSB0

to you have any ideas?

[image]

Dr-Gigavolt, 2022-12-28:
Try to reload the ./service directory in your SCP tool to let the log file to appear. I use WinSCP and have to do it. If no log file appears, try to start the aggregatebatteries.py in Pyton in a console to show the error message.

pos-ei-don, 2022-12-28:
ok.
First error was this:
I only have 2 batteries connected, and a smartShunt. Don't know what it found.

INFO:root:2022-12-28 23:33:18.776249: Connected to dbus, and switching over to GLib.MainLoop() (= event based)
INFO:root:2022-12-28 23:33:19.777730: Scan of batteries: Trial Nr. 1
INFO:root:2022-12-28 23:33:19.779728: 3 batteries found.
INFO:root:2022-12-28 23:33:20.779381: Scan of batteries: Trial Nr. 2
INFO:root:2022-12-28 23:33:20.781371: 3 batteries found.

pos-ei-don, 2022-12-28:
Test #2:

Tried to set No. of batteries to 3:

INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:2022-12-28 23:34:39.534136: Connected to dbus, and switching over to GLib.MainLoop() (= event based)
INFO:root:2022-12-28 23:34:40.534288: Scan of batteries: Trial Nr. 1
INFO:root:2022-12-28 23:34:40.544409: 3 batteries found.
ERROR:root:2022-12-28 23:34:41.570426: ERROR: DBus Value Error. Read trial nr. 1
ERROR:root:2022-12-28 23:34:42.567903: ERROR: DBus Value Error. Read trial nr. 2
ERROR:root:2022-12-28 23:34:43.569075: ERROR: DBus Value Error. Read trial nr. 3
ERROR:root:2022-12-28 23:34:44.569981: ERROR: DBus Value Error. Read trial nr. 4

Dr-Gigavolt, 2022-12-28:
Test #2 - if the 3rd battery, whatever it is, is not responsive, you get this error.
I have no experience with mqqt, try dbus-spy in the console and look what could be the 3-rd battery. Perhaps some dead service?

pos-ei-don, 2022-12-28:
ok, did not know this!
It is the SmartShunt from Victron! It is actually no battery. It just measures the total ampers going to all batteries!

Services
com.victronenergy.battery.ttyUSB0             SerialBattery(Jkbms)
com.victronenergy.battery.ttyUSB1             SerialBattery(Jkbms)
com.victronenergy.battery.ttyUSB4             SmartShunt 500A/50mV

Dr-Gigavolt, 2022-12-28:
No Problem, I made a new commit of aggregatebatteries.py

pos-ei-don, 2022-12-29:
Still same error, it  still finds 3 batteries regarding the logs.

Dr-Gigavolt, 2022-12-29:
I know, my mistake. I'm trying to fix it now.

Dr-Gigavolt, 2022-12-29:
Now it should work:
for service in self._dbusConn.list_names():
            if SERVICE_KEY_WORD in service:
                if PRODUCT_NAME_KEY_WORD in (VeDbusItemImport(self._dbusConn, service, '/ProductName').get_value()):
                    self._batteries.append(service)
                    batteriesCount += 1 

Update the aggregatebatteries.py and settings.py

And I go to bed, it is 1:30 AM in Germany :-)

pos-ei-don, 2022-12-29:
thank you for the update! it works!!!

I will test this the next days!

closed 2022-12-29 by pos-ei-don