Dr-Gigavolt/dbus-aggregate-batteries #2: Don't use "except:"

smurfix, 2022-12-29:
Always use "except Exception:" if you need a catch-all.

Otherwise you can't even interrupt the program.

https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/blob/af2e5203dcb4a31fdf47547d981f00354c470284/aggregatebatteries.py#L245

Dr-Gigavolt, 2022-12-29:
I haven't understood what you exactly mean. I don't want to interrupt the program at line 245 but at line 250 when all allowed trials fail. Should I put there Exception instead of sys.exit() ?

smurfix, 2022-12-29:
You simply want to use `except Exception:` instead of `except:`. `SystemExit` and `KeyboardInterrupt` are not subclasses of `Exception` but of `BaseException` thus they're not caught by this statement which is exactly what you want.

Dr-Gigavolt, 2022-12-29:
I modified as you suggested, thanks. But even with except: it did stop on Ctrl+C if started from console by python3 aggregatebatteries.py, without jumping into the code following it.

closed 2022-12-29 by Dr-Gigavolt