Dr-Gigavolt/dbus-aggregate-batteries #3: Reduce number of messages

smurfix, 2022-12-29:
Updating every value on its own generates a lot of Dbus messages.

You can fix this by using a context manager:
```
with self._dbusservice as bus:
    bus["/some/path"] = some_value
    bus["/other/path"] = other_value
    pass # updates all paths using a single dbus one message

```

https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/blob/af2e5203dcb4a31fdf47547d981f00354c470284/aggregatebatteries.py#L355-L403

Dr-Gigavolt, 2022-12-29:
Thanks for the proposal, I'll implement it soon.

Dr-Gigavolt, 2022-12-29:
Done.

closed 2022-12-29 by Dr-Gigavolt

Dr-Gigavolt, 2022-12-30:
Hi Matthias, do you know why this does not work?
with self._dbusConn as bus:
    Voltage += VeDbusItemImport(bus, self._batteries[i], '/Dc/0/Voltage').get_value()

smurfix, 2022-12-31:
Why do you think it should work? dbus.SessionBus is not a context manager, it doesn't have `__enter__` and `__exit__` methods.

Dr-Gigavolt, 2022-12-31:
Because I tried it 😀

Dr-Gigavolt, 2022-12-31:
Sorry, I haven't read correctly, you wrote:
Why do you think it should work? ... not shouldn't ...
Now I've understood. Thanks.
I don't know by heart what which methods has. I write programs only occasionally, not enough experience.

Dr-Gigavolt, 2023-01-08:
Hi Matthias, is there some option to read all paths at once (e.g. as a list) instead of using VeDbusItemImport many times? Normally is the readout fast enough, but I had some cases when all imports took many seconds.

smurfix, 2023-01-09:
Umm … you shouldn't. (I was intending to open an issue for that.) Instead you create a `DbusMonitor` and register for these messages, so you'll get them automatically and with significantly less system load.

You can see how to do that e.g. in `opt/victronenergy/dbus-systemcalc-py/dbus_systemcalc.py`

Dr-Gigavolt, 2023-01-10:
Thanks, Matthias. I just wrote a class for monitoring serial batteries and it works. I will integrate it into my tool.

Dr-Gigavolt, 2023-01-11:
Hello, I need your help again. The class I wrote yesterday (see below) works perfectly as test in main(). Logging shows only what I ask for. But if I create an instance of it as a member of class DbusAggBatService in aggregatebatteries.py I get the following messages from the dbusmonitor and it is time consuming. My service is shown in the device list, but ESS does not accept it. What is wrong? Thanks a lot.

INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 1
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB3, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB3 has device instance 2
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:2023-01-11 22:22:55.628857: Initial Ah read from file: 55Ah
INFO:root:2023-01-11 22:22:55.643935: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate was skipped because it has no device instance
INFO:root:2023-01-11 22:23:20.708781: Scan of batteries: Trial Nr. 1
INFO:root:2023-01-11 22:23:20.842474: 2 batteries found.
INFO:root:2023-01-11 22:23:21.851847: Searching Multi/Quatro VEbus: Trial Nr. 1
INFO:root:2023-01-11 22:23:21.855934: MultiPlus-II 48/5000/70-50 found.
INFO:root:2023-01-11 22:23:22.880010: Searching of MPPTs: Trial Nr. 1
INFO:root:2023-01-11 22:23:22.884309: 1 MPPTs found.
INFO:root:2023-01-11 22:23:24.491648: Max. charging voltage: 55.0V
INFO:root:2023-01-11 22:23:24.492589: Max. charging current: 100.0A
INFO:root:2023-01-11 22:23:24.493268: Max. dircharging current: 200.0A

[batterymonitor.py.txt](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/files/10396322/batterymonitor.py.txt)

Dr-Gigavolt, 2023-01-13:
@smurfix 
Matthias, excuse me for annoying, could you have a look above? I want to update my code ASAP, some guys are waiting. Thanks.

smurfix, 2023-01-13:
Well, please keep in mind that I have my own  paying job that takes precedence.

At first glance I see that your `_find_batteries` does an unconditional lookup on `/ProductName` which obviously deadlocks because your code is not in the GLib mainloop and thus you can't answer that request to yourself.

So you need to skip your own service if you do any import calls.

NB, this kind of deadlock nonsense is exactly why I'm writing my own dbus wrapper, based on async Python. No GLib. It's not quite ready for public consumption unfortunately.

Dr-Gigavolt, 2023-01-13:
Should I avoid using VeDbusItemImport and collect all data through dbusmonitor? Or would you mind making a phone call in the evening, after finishing our payed work? Thanks.

smurfix, 2023-01-13:
No, you simply need to add a test to the loop at line 122+ that skips your own battery. The list returned by `self._dbusConn.list_names` contains the virtual battery you created yourself.

Dr-Gigavolt, 2023-01-13:
The test is already there:

_if BATTERY_KEY_WORD in service:
try:                                                                                  
productName = VeDbusItemImport ..._

and BATTERY_KEY_WORD = 'com.victronenergy.battery.tty',
the name of my own service is 'com.victronenergy.battery.aggregate'

smurfix, 2023-01-13:
Mmh. OK. In that case you'll have to do more intrusive logging and tracing to find the place where you deadlock yourself.

I don't have a better idea here than to sprinkle the code with `print` or `logger.debug` statements.

Dr-Gigavolt, 2023-01-13:
> I don't have a better idea here than to sprinkle the code with `print` or `logger.debug` statements.

That is exactly what I'm doing. Please see the log above (2 days ago). This deadlock happens only when I create an instance of dbusmonitor. It finds the physical batteries immediatelly, before my service is createt. Than my service is created, detected by dbusmonitor and after long time _com.victronenergy.battery.aggregate was skipped because it has no device instance_

Dr-Gigavolt, 2023-01-13:
It seems the dbusmonitor cannot read the service created by the same program as itself started. Even not if another thread (another GLib.timeout) used. It works only from another console.

Here (dbusmonitor.py) happens the exception:
			try:
				di = self.dbusConn.call_blocking(serviceName,
					'/DeviceInstance', None, 'GetValue', '', [])
			except dbus.exceptions.DBusException:
				logger.info("       %s was skipped because it has no device instance" % serviceName)
				return False # Skip it
even if '/DeviceInstance' has a value.

After some 10s the exception is handled, my service sends its path values, they are visible in the Device List, but usually ignored by ESS. 

A workaround could be to add the test into dbusmonitor.py, but it is not a good idea to modify the Venus own stuff.

Dr-Gigavolt, 2023-01-13:
Now I learned that GLib.timeout doesn't start a thread. But this works:

        _x = Thread(target = self._startMonitor)
        x.start()
        GLib.timeout_add(1000, self._find_batteries)  # search connected batteries   
    
    def _startMonitor(self):
        logging.info('%s: Starting battery monitor.' % dt.now())
        self._batteryMonitor = BatteryMonitor()_