Dr-Gigavolt/dbus-aggregate-batteries #4: Max/min charge/discharge voltages

smurfix, 2022-12-29:
This may be a stupid question, but why are you doing this bulk/absorption/float dance?

IMHO you should simply use the lowest charge voltage you get from the batteries. No configuration required.
Same for discharge (using the highest voltage of course).

Current is a bit more complicated, as the batteries' internal resistances vary. However, when you see some nontrivial charge current you can calculate the per-battery relation between the reported actual and maximum currents, and use that to calculate a safe overall maximum. Same for discharge.

https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/blob/af2e5203dcb4a31fdf47547d981f00354c470284/aggregatebatteries.py#L288

Dr-Gigavolt, 2022-12-29:
This dance is just to be able to measure absorption time before float. But there are different opinions whether it is necessary or at least useful in case of Li batteries.

smurfix, 2022-12-29:
Those concepts are from lead batteries where they are useful for getting a good charge into them. IMHO they're entirely useless, if not harmful, on Lithium.

What's more, there are adaptive BMSes out there which recognize that the charge curve on LiFePo4 is seriously non-linear. One effect of this is that "all cells are at 3.35V while one is unbalanced and at 3.5V" *requires* a substantially lower max charge voltage than "all cells are at 3.4V" – assuming that you want to prevent the cell overvoltage protection to trigger the protective relay / charge FET. Disconnecting a single battery is potentially dangerous because the current destined for N batteries now goes through N-1 …

As to what's good for the batteries, well, again, for lead batteries those charge dances make sense because of their chemistry, and it basically works because the charge curves are uniform and the cells self-balance when you charge them to 100%, they simply heat up / create hydrogen+oxygen, and thus burn off the extra charge.

Lithium? not so. Lithium batteries die when you overcharge them, that's why you need that balancer. They degrade with (a) too high cell voltage (b) lots of charge cycles. There's a reason why the typical LiFePo4 BMS is set for max cell voltage of 3.6V but the overall maximum voltage is 3.45*16 (or however many cells you do have).

Dr-Gigavolt, 2022-12-29:
OK, I can remove the time measurement and keep the Vbat <= 0.96*VcellMax*N and Vcell <= VcellMax. So it would be N*3.45V / 3.6V for LiFePO4 and N*2.6V / 2.7V for LTO.

Dr-Gigavolt, 2022-12-29:
Done.

closed 2022-12-29 by Dr-Gigavolt