Dr-Gigavolt/dbus-aggregate-batteries #5: Amps from the batteries

pos-ei-don, 2023-01-06:
Why do your own calculation for allowed amps?
The BMS has a quiet a good value, why not only sum them up together

Dr-Gigavolt, 2023-01-06:
I don't understand your question. BMS has no allowed value. Do you mean the measured current? You can disable the Victron current measurement if your BMS measures the current preisely. My don't.

Dr-Gigavolt, 2023-01-06:
I close because it is not an issue.

pos-ei-don, 2023-01-06:
It is an issue. The bms tells exactly how much current is allowed for charging and discharging. It respects the Single cell voltage.
Yout driver is great. But it ignores the values He gets from the Single BMS and calculate its own Max current value.
This brougjt me in teoibles today as it blocked  charging much too early and my inverter kept priducing too muxh Power.

Dr-Gigavolt, 2023-01-06:
Which BMS do you have? My JK does not provide such value. How should it? It doesn't know with which cells it is put together. And if your BMS does it somehow, it is not re-transmitted by Louis driver. He does calculate is as well. Therefore perhaps he can help you.

pos-ei-don, 2023-01-06:
I can Set it in the jk App settings. And yes, Louis driver can send it. But He cal also change it regardibg the settings.
And These values make sense anw ork for me quiet well.
So I which to use these settings also for the Aggregated virtual battery. Just sum them up.

Dr-Gigavolt, 2023-01-06:
I double checked the BMS app and the transmission from Louis driver:
- in the BMS you adjust the charge/discharge current which if exceeded, the BMS switches the charge/discharge MOSFETs off after adjusted period of time. This is an emergency brake, not a control parameter for the charger. The charger set point must be always lower
- Louis driver does not send it to the DBus. Try dbus-spy command, you will not find it
- the driver sends the max. charge/discharge current and max. charge voltage which is calculated from SoC and this is in case of JK BMS extremely unprecise. And in my case the driver calculation didn't take the cell with max. voltage into account. I don't know why, his code is too complex for me.
- the only what I can do is to implement an option to sum and transmit this values (from all instances of driver) bus not from the BMS. Is this what you mean?

pos-ei-don, 2023-01-06:
Yes, this is what I meant.
If i dont let Louise driver change the current I the settings, it transmits the value from the bms.
I have several battery Packs. One is currently not running well (a defect cell). If this Pack blocks charging, the other packs  can take the load. I just need  to change the maximum current from 800A to 750A. The sum of These values will be perfect for this.

Dr-Gigavolt, 2023-01-06:
I can implement it, but it will not work well if you disable the current management in driver or my program. First, you will charge up to fuse threshold of BMS. They can switch off even if you don't want it. Second, the current will not be reduced if close to full. I can propose the following solution:

- selection of current and voltage management either from Louis or from me
- selection whether the charge/discharge current is set to zero (as in the upgrade I made today) or reduced in ratio of connected and blocking batteries.

Marvo2011, 2023-01-06:
I also think that we should use the amps from the BMS and sum it. I have some DC load and MPPTs, so the value from the Multi(s) are wrong.

Dr-Gigavolt, 2023-01-06:
MPPTs are included. If your BMS current is precise, you can de-activate the feature in settings.py. Or send me the service name of your DC load measurement and I can include it.

pos-ei-don, 2023-01-06:
My "DC load measurement" is the victron smartshunt. This is quiet precise, but i remember the first version of 
your plugin did not start with the smartshunt. Is there anything i can help to implement it?

Dr-Gigavolt, 2023-01-07:
The problem with the smart shunt was its service name "battery", therefore the nr. of batteries found was one more than expected. Now after identifying "batteries" I check the product name to exclude everything except of serial-batteries.
I will add these features in the next version:
- select whether own or battery driver's charge parameters
- select whether stop charging/discharging or reduce current if some battery is blocking
- smart shunt for DC loads

pos-ei-don, 2023-01-07:
Thank you! Them I close this issue!

closed 2023-01-07 by pos-ei-don

Dr-Gigavolt, 2023-01-07:
No problem, but it could take a bit time, my holiday is over 😀

Dr-Gigavolt, 2023-01-08:
Hello, I've just implemented new features (incl. Issue #6) and let it run to test it a bit before commit&push. But I couldn't do the current reduction according blocking by some BMSs. The reason is, it would cause mess for CAN batteries having multiple modules inside of one battery. So the modules online and offline needed to be taken into account and I have no possibility to test how it works. I keep reducing current to zero if something happens, this is the best for safety. If you need that option it is better if you modify your copy of the code or find someone with CAN batteries to do it correctly.

pos-ei-don, 2023-01-09:
I just found out that the SmartShunt hast 2 seperate Volt sensors.
The Sensor i need is #2, currently (if i use it as "Inverter"), it uses the first Sensor.
Can I change it somewhere to use the second?
((The first is used to power the smartshunt, so it is connected to a UPS))