Dr-Gigavolt/dbus-aggregate-batteries #7: value-updates only after 30-40 sec. in Venus OS or VRM-Portal

superkrischi, 2023-01-12:
Hello and many thanks for the nice driver.

I have 3 packs with 16x280Ah and JkBMS with RasPi and Venus OS.
The driver works an every battery, also the virtual aggregate-battery is shown in the Venus OS.
After a while, the values are updated only after 30-40 seconds.
Thats a little problem to find the correct gridpoint near zero.
Is it possible to fix that by my own? 

Thanks and greetings
Kristian

Dr-Gigavolt, 2023-01-12:
Hello, I know about it and work on it. @smurfix supports me with his knowledge. It is a ressource issue. For now you can restart the program, it usually helps.

Dr-Gigavolt, 2023-01-15:
Dbusmonitor used instead VeDbusItemImport. Solution discussed under isuue #5.

closed 2023-01-15 by Dr-Gigavolt