Dr-Gigavolt/dbus-aggregate-batteries #9: Error in aggregatebatteries.py

kakariki1, 2023-01-29:
I think there's a small error in aggregatebatteries.py

https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/blob/6b66f8e8293a12e2320dc0330ef0bd7742d4970d/aggregatebatteries.py#L355-L357

must be changed to (remove the .get_value() at the end of the line):

`MaxChargeCurrent += self._dbusMon.dbusmon.get_value(self._batteries[i], '/Info/MaxChargeCurrent')`
`MaxDischargeCurrent += self._dbusMon.dbusmon.get_value(self._batteries[i], '/Info/MaxDischargeCurrent')`
`MaxChargeVoltage.append(self._dbusMon.dbusmon.get_value(self._batteries[i], '/Info/MaxChargeVoltage'))`

Dr-Gigavolt, 2023-01-29:
Hello, are you sure? I found this member function in examples and it works. I can try to remove it and if it works without I'll change it in the next version.

kakariki1, 2023-01-29:
Hello,

I'm sure for at least my configuration (where OWN_CHARGE_PARAMETERS ist set to False).
Without those changes it does not work.
I don't get any error messages, but there are also no values in Venus OS.

Best regards,
lakeroe

Dr-Gigavolt, 2023-01-30:
Perhaps some version incompatibility. I'll try it without get_value.

Dr-Gigavolt, 2023-02-04:
Hi Lakeroe,

now I'm back at my computer. Of course it is a mistake :-). I've overlooked (on the small screen of my mobile phone when reading your message couple of days ago) that the get_value() is two times there. It cannot give an error when I use own charge parameters and this part of the code is not executed. As promised, I'll publish the correction ASAP together with some other improvements.

BR,
Anton

closed 2023-02-09 by Dr-Gigavolt