Dr-Gigavolt/dbus-aggregate-batteries #10: Calculation of MaxChargeCurrent and MaxDischargeCurrent for only one Multiplus II

kakariki1, 2023-01-30:
The current driver only works for systems having more than one Multiplus II like yours (3 phase system).

My system is one Multiplus II connected to two parallel-connected batteries.
In this case the calculation for maximum charge/discharge currents and maybe other values is different.

MaxChargeCurrent and MaxDischargeCurrent should be the lowest of all values.

Adding this option would be great ...

Dr-Gigavolt, 2023-01-30:
I don't understand. If you disable own charge/discharge parameters calculation you receive the parameters from your both instances of the battery driver. Each one should be configured for only one battery. Therefore sum of currents and minimum of voltages. Why should it be dependend on quantity of multis?

kakariki1, 2023-01-30:
I'll try to explain using following example:

The system is charging and allmost full, but the state-of-charge is slightly different.
The maximum cell voltage of battery 1 is 3.38V -> this results in a maximum charge current of i.e. 50A
The maximum cell voltage of battery 2 is 3.43V -> this results in a maximum charge current of i.e. 10A

If using the sum of currents (i.e. 60A) battery 2 is charged with too much current.
In this case I would like to charge using 2x 10A

Does this make sense ?

Dr-Gigavolt, 2023-01-30:
Thanks for the details, I understand now. I can change it in this way. This week I have no time for it, but I'll do my best.

kakariki1, 2023-01-30:
Okay, thanks and no hurry ;-)

kakariki1, 2023-01-31:
If you're interested I attached the changed files.
Changes are marked by "lakeroe", hardcoded and the files are renamed to .txt
It's not fully tested yet, but works so far ...

[aggregatebatteries.txt](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/files/10543761/aggregatebatteries.txt)
[dbusmon.txt](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/files/10543762/dbusmon.txt)
[settings.txt](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/files/10543760/settings.txt)

maxicus, 2023-02-01:
@kakariki1 
In such a different battery packs, I'd expect that internal resistance growth of cell will cause that once charger push 60A, it won't split it 1:1. i.e.:
- battery1 in your case will take lets say 40A, and 
- battery 2 will take 20A (since it's nearly charged and it's resistance start to grow)
Did you check that ratio in that case for your setup?

If my expectations are true and difference is substantial, limiting total current to 20A will leave a lot of space for optimization.

Other than that I agree that using min current will cause safer calculations for a generic case.

kakariki1, 2023-02-01:
Here is a screenshot showing the charge current dependent on maximum cell voltage:

[image]

It's not clearly visible but (for example) if maximum cell voltage of battery 1 is 3.43V then charge current is 4,92A.
At the same time maximum cell voltage of battery 2 is 3.38V then charge current is 12,91A.
So I agree, charge current does not take 1:1 !

How would you calculate charge current ?

Dr-Gigavolt, 2023-02-01:
I see no reason to adapt the charge current. Your cells are not balanced yet. When they will be ballanced, both batteries will be full at the same time. So taking the lower current or own charge parameters is safer.

kakariki1, 2023-02-02:
I'm sorry but I have to disagree. All cells were balanced as good as I could. I connected them all in parallel and charged them to 3.60V since the charge current was 0. I then let them sit for another day without power supply ...
In my opinion there will always be some imbalance due to small capacity differences ...

Dr-Gigavolt, 2023-02-02:
Ok, if ballanced, then everything is fine. It will work well with the lowest current x N. It will be enough to transport the last fraction of charge and the balancers will do their job.

Dr-Gigavolt, 2023-02-04:
Hi Lakeroe,
thanks for your proposals. I added it into my code and going to test for couple of days. Except of searching for minima of AllowToCharge/Discharge in case of OWN_CHARGE_PARAMETERS == False I put an additional safety feature for the case of 'True' - AllowToCharge/Discharge will be set to zero if the standard mechanism fails and the Min/MaxCellVoltage continues to be exceeded.

Dr-Gigavolt, 2023-02-09:
AllowToCharge/Discharge has no effect, is only shown in GUI. The max. charge current is very coarse, I had over 10A if limited to zero. Only max. charge voltage has an reasonable accuracy to avoid overcharging.

closed 2023-02-09 by Dr-Gigavolt