Dr-Gigavolt/dbus-aggregate-batteries #12: Weird startup behaviour

kakariki1, 2023-02-05:
Hello,

I've found some weird startup behaviour.
After normal Venus OS power-up the driver does (most of the time) not recognize dbus-serial-battery instances.
AggregateBatteries appears in Venus OS but shows --- ---

Output of dbus-spy:
> com.victronenergy.battery.aggregate
Alarms/CellImbalance                                                                     -
Alarms/HighChargeCurrent                                                                 -
Alarms/HighChargeTemperature                                                             -
Alarms/HighDischargeCurrent                                                              -
Alarms/HighTemperature                                                                   -
Alarms/HighVoltage                                                                       -
Alarms/InternalFailure                                                                   -
Alarms/LowCellVoltage                                                                    -
Alarms/LowChargeTemperature                                                              -
Alarms/LowSoc                                                                            -
Alarms/LowTemperature                                                                    -
Alarms/LowVoltage                                                                        -
Capacity                                                                                 -
Connected                                                                                1
ConsumedAmphours                                                                         -
Dc/0/Current                                                                             -
Dc/0/Power                                                                               -
Dc/0/Temperature                                                                         -
Dc/0/Voltage                                                                             -
DeviceInstance                                                                           0
FirmwareVersion                                                                        2.0
HardwareVersion                                                                        0.0
Info/MaxChargeCurrent                                                                    -
Info/MaxChargeVoltage                                                                    -
Info/MaxDischargeCurrent                                                                 -
InstalledCapacity                                                                        -
Io/AllowToCharge                                                                         -
Io/AllowToDischarge                                                                      -
Mgmt/Connection                                                                    Virtual
Mgmt/ProcessName                      /data/dbus-aggregate-batteries/aggregatebatteries.py
Mgmt/ProcessVersion                                                                    0.0
ProductId                                                                                0
ProductName                                                             AggregateBatteries
Soc                                                                                      -
System/MaxCellTemperature                                                                -
System/MaxCellVoltage                                                                    -
System/MaxVoltageCellId                                                                  -
System/MinCellTemperature                                                                -
System/MinCellVoltage                                                                    -
System/MinVoltageCellId                                                                  -
System/NrOfCellsPerBattery                                                               -
System/NrOfModulesBlockingCharge                                                         -
System/NrOfModulesBlockingDischarge                                                      -
System/NrOfModulesOffline                                                                -
System/NrOfModulesOnline                                                                 -

My solution for now is to restart your service 30s after booting by adding following to /data/rc.local
sleep 30
/data/dbus-aggregate-batteries/restart

Of course this is a ugly hack, but it works.
Once up and running it works perfect, just this weird "cold boot problem".

Do you have any ideas for the root cause of this problem ?

Dr-Gigavolt, 2023-02-05:
Hello, the SerialBattery needs more time to appear on Dbus. On each start of Venus It is copied from /data/etc to /opt/victronenergy and by trial&error the BMS type is identified. The AggregateBatteries appears much sooner on Dbus. I've allready observed that many trials are necessary to find the batteries. And was reported about another issue per eMail:  self._dbusMon.dbusmon.get_value() was called before dbus monitor finished scanning and the object was created. Increasing delay in  GLib.timeout_add(1000, self._find_batteries) did help. I will solve it ASAP.

kakariki1, 2023-02-05:
Hello,

this makes sense so far.
I just removed my hack in /data/rc.local (as described above) and increased your suggested delay to 60000 and it works for me ...

Best regards,
lakeroe

Dr-Gigavolt, 2023-02-05:
I think 5 second would be enough, if necessary it tries couple of times but does not wait for one minute if not necessary.
BR,
Anton

kakariki1, 2023-02-05:
For me 5 seconds is not enough. Even 10s doesn't it make reliable.
I'll keep testing ...

BR,
Berni

Dr-Gigavolt, 2023-02-05:
Hi Berni,
I set now 5s and killed my service as well as the serial batteries. It took about 1/2 minute to collect everything. I keep this setting, it is a good compromise. It is better to run some trials than to wait in any case. If only AggregateBatteries is restarted while SerialBatteries are running, it needs only 1 trial.

INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyUSB1 has device instance 289
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB0 has device instance 288
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:2023-02-05 16:09:04.879287: Searching batteries: Trial Nr. 1
INFO:root:2023-02-05 16:09:04.882396: 0 batteries found.
INFO:root:2023-02-05 16:09:09.880679: Searching batteries: Trial Nr. 2
INFO:root:2023-02-05 16:09:09.883523: 0 batteries found.
INFO:root:2023-02-05 16:09:14.881145: Searching batteries: Trial Nr. 3
INFO:root:2023-02-05 16:09:14.890996: 0 batteries found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 1
INFO:root:2023-02-05 16:09:19.881819: Searching batteries: Trial Nr. 4
INFO:root:2023-02-05 16:09:19.884665: SerialBattery(Jkbms) found.
INFO:root:2023-02-05 16:09:19.885670: 1 batteries found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB3, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB3 has device instance 2
INFO:root:2023-02-05 16:09:24.882660: Searching batteries: Trial Nr. 5
INFO:root:2023-02-05 16:09:24.900246: SerialBattery(Jkbms) found.
INFO:root:2023-02-05 16:09:24.901333: SerialBattery(Jkbms) found.
INFO:root:2023-02-05 16:09:24.902063: 2 batteries found.

Dr-Gigavolt, 2023-02-05:
[restart_all.txt](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/files/10611514/restart_all.txt)
[restart_dbus-serial-battery.txt](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/files/10611515/restart_dbus-serial-battery.txt)

You can use these shell files to restart easily (remove .txt)

kakariki1, 2023-02-05:
Hi Anton,

so you set GLib.timeout_add(5000, self._find_batteries) ?
Does this also reliable work after power-up ?

I set this to 20000 and after a coldstart it worked 20 out of 20 times ...
Your shell scripts above simulate another scenario.

Dr-Gigavolt, 2023-02-05:
I've just tried the reboot, it looks similar to the case with restart of services. 2 trials more were needed. See below. Even before I increased the timeout to 5s it was OK - 10 trials x 1s were not enough, the program was terminated and restarted until the battery instances were ready.

BR,
Anton

INFO:root:2023-02-05 17:30:11.957469: Starting AggregateBatteries.
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:2023-02-05 17:30:11.993065: Initial Ah read from file: 60Ah
INFO:root:2023-02-05 17:30:12.017195: Starting battery monitor.
INFO:root:2023-02-05 17:30:12.021784: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyUSB1 has device instance 289
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB0 has device instance 288
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:2023-02-05 17:30:17.022062: Searching batteries: Trial Nr. 1
INFO:root:2023-02-05 17:30:17.026324: 0 batteries found.
INFO:root:2023-02-05 17:30:22.022608: Searching batteries: Trial Nr. 2
INFO:root:2023-02-05 17:30:22.025861: 0 batteries found.
INFO:root:2023-02-05 17:30:27.023549: Searching batteries: Trial Nr. 3
INFO:root:2023-02-05 17:30:27.026980: 0 batteries found.
INFO:root:2023-02-05 17:30:32.024030: Searching batteries: Trial Nr. 4
INFO:root:2023-02-05 17:30:32.027049: 0 batteries found.
INFO:root:2023-02-05 17:30:37.025226: Searching batteries: Trial Nr. 5
INFO:root:2023-02-05 17:30:37.028055: 0 batteries found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 1
INFO:root:2023-02-05 17:30:42.025910: Searching batteries: Trial Nr. 6
INFO:root:2023-02-05 17:30:42.029013: SerialBattery(Jkbms) found.
INFO:root:2023-02-05 17:30:42.030035: 1 batteries found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB3, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB3 has device instance 2
INFO:root:2023-02-05 17:30:47.026511: Searching batteries: Trial Nr. 7
INFO:root:2023-02-05 17:30:47.029775: SerialBattery(Jkbms) found.
INFO:root:2023-02-05 17:30:47.030660: SerialBattery(Jkbms) found.
INFO:root:2023-02-05 17:30:47.031398: 2 batteries found.
INFO:root:2023-02-05 17:30:48.032553: Searching Multi/Quatro VEbus: Trial Nr. 1
INFO:root:2023-02-05 17:30:48.035360: MultiPlus-II 48/5000/70-50 found.
INFO:root:2023-02-05 17:30:49.036905: Searching MPPTs: Trial Nr. 1
INFO:root:2023-02-05 17:30:49.039788: SmartSolar MPPT VE.Can 250/100 rev2 found.
INFO:root:2023-02-05 17:30:49.040772: 1 MPPT(s) found.
INFO:root:2023-02-05 17:30:50.044732: Max. charging voltage: 55.0V
INFO:root:2023-02-05 17:30:50.045611: Max. charging current: 200.0A
INFO:root:2023-02-05 17:30:50.046215: Max. dircharging current: 200.0A

kakariki1, 2023-02-05:
Okay, thanks for your infos.
I'll do more testing later this week and let you know.

BR,
Berni

kakariki1, 2023-02-09:
Hi Anton,

today I did some more testing and found something strange.
If I set the timeout to 5000 and do a coldstart your AggregateBatteries appears in Venus OS but only shows --- --- (but only sometimes).

The appropriate log looks like this:
INFO:root:2023-02-09 09:52:47.051366: Starting AggregateBatteries.
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:2023-02-09 09:52:47.184599: Initial Ah read from file: 304Ah
INFO:root:2023-02-09 09:52:47.287862: Starting battery monitor.
INFO:root:2023-02-09 09:52:47.301865: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:2023-02-09 09:52:52.305312: Searching batteries: Trial Nr. 1
INFO:root:2023-02-09 09:52:52.339622: 0 batteries found.
INFO:root:2023-02-09 09:52:57.344584: Searching batteries: Trial Nr. 2
INFO:root:2023-02-09 09:52:57.385346: 0 batteries found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 2
INFO:root:2023-02-09 09:53:02.758919: Searching batteries: Trial Nr. 3
INFO:root:2023-02-09 09:53:02.835450: SerialBattery(Jkbms) found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 1

This ONLY happens after a coldstart sometimes.
By restarting the services (using your restart_all script) in can't reproduce the problem (I tried 20 times).

For me this looks like another problem.
Do you have any idea, should I try something ?

Thanks and best regards,
Berni

kakariki1, 2023-02-09:
If AggregateBatteries is in the state described above it clearly hangs and does not respond to anything.
Killing the dbus-serialbattery services (by running restart_dbus-serial-battery) does not change anything (no output in the log).
After running restart_all everything works as expected ...

Dr-Gigavolt, 2023-02-09:
Hi Berni,
I make new commit this evening or tommorow, please try with the new one.
BR,
Anton

kakariki1, 2023-02-10:
I think this issue can also be closed.
I tried your latest commit and it works flawlessly (on 20 tried coldstarts).

Thanks for your great work !

Best regards,
Berni

Dr-Gigavolt, 2023-02-10:
Hi Berni, 
nice to hear it :-). Thanks for your input.
Best regards, 
Anton

closed 2023-02-10 by Dr-Gigavolt