Dr-Gigavolt/dbus-aggregate-batteries #15: no values after reboot

transistorgit, 2023-03-08:
Hi,

I use this drivers since some days for my 2 daly 16s batteries. It's working fine in general, but often after reboot it won't come up properly and the remote console shows only "---" "---".
If I ./restart on the console it seems to work always and shows correct values for currents/voltage etc. I found just one hint in the log.

The log shows one startup around 19:19 by reboot and a manual restart around 19:33.
There is 1 single "ERROR" line in the first startup. From the code, I would think it has nothing to do with the problem.

INFO:root:2023-03-08 19:19:56.105017: Starting AggregateBatteries.
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:2023-03-08 19:19:56.187863: Initial Ah read from file: 286Ah
INFO:root:2023-03-08 19:19:56.212418: Starting battery monitor.
INFO:root:2023-03-08 19:19:56.244676: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB3, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB3 has device instance 289
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB1 has device instance 289
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 3
INFO:root:2023-03-08 19:20:18.887406: Searching batteries: Trial Nr. 1
INFO:root:2023-03-08 19:20:18.905823: SerialBattery(Daly) found.
INFO:root:2023-03-08 19:20:18.906897: 1 batteries found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB4, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB4 has device instance 5
INFO:root:2023-03-08 19:20:23.887792: Searching batteries: Trial Nr. 2
INFO:root:2023-03-08 19:20:23.890575: SerialBattery(Daly) found.
INFO:root:2023-03-08 19:20:23.891499: SerialBattery(Daly) found.
INFO:root:2023-03-08 19:20:23.892263: 2 batteries found.
INFO:root:2023-03-08 19:20:24.895094: Searching Multi/Quatro VEbus: Trial Nr. 1
INFO:root:2023-03-08 19:20:24.915631: MultiPlus-II 48/3000/35-32 found.
ERROR:root:2023-03-08 19:20:25.928236: Victron current read error. Using BMS current and power instead.
INFO:root:2023-03-08 19:20:25.929184: Max. charging voltage: 56.0V
INFO:root:2023-03-08 19:20:25.929783: Max. dircharging current: 100.0A Min. cell voltage: 3.476V
INFO:root:2023-03-08 19:33:09.331976: Starting AggregateBatteries.
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:2023-03-08 19:33:09.364182: Initial Ah read from file: 286Ah
INFO:root:2023-03-08 19:33:09.379929: Starting battery monitor.
INFO:root:2023-03-08 19:33:09.382504: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 3
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB3, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB3 has device instance 289
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB4, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB4 has device instance 5
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB1 has device instance 289
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:2023-03-08 19:33:14.382805: Searching batteries: Trial Nr. 1
INFO:root:2023-03-08 19:33:14.385792: SerialBattery(Daly) found.
INFO:root:2023-03-08 19:33:14.386702: SerialBattery(Daly) found.
INFO:root:2023-03-08 19:33:14.387490: 2 batteries found.
INFO:root:2023-03-08 19:33:15.389070: Searching Multi/Quatro VEbus: Trial Nr. 1
INFO:root:2023-03-08 19:33:15.391817: MultiPlus-II 48/3000/35-32 found.
INFO:root:2023-03-08 19:33:16.396971: Max. charging voltage: 56.0V
INFO:root:2023-03-08 19:33:16.397821: Max. dircharging current: 100.0A Min. cell voltage: 3.464V
root@venus:/data/dbus-aggregate-batteries#

brgds
Bernd

Dr-Gigavolt, 2023-03-08:
Hello guys,
I cannot identify the problem. Do you use the last version of the code? According the Bernd's log everything is fine. The Multi was unresponsive in one case, not bad, otherwise there were multiple error messages. If all hardware is found and the charge/discharge limits are calculated, I expect the problem somewhere else. If you see more errors, please let me know.
BR,
Anton

Dr-Gigavolt, 2023-03-12:
Is it the same as #17? Can I close?

transistorgit, 2023-03-19:
I'm not totally sure what's going on. I'm not used to the dbus stuff.
The cause may be sporadic connection problems like in #17 or maybe releated to dbus-monitor like in #20. However, I think aggregate-batteries seem to have init-reinit problems on startup and on disappearing/reappearing devices. Good propability that these problems have the same root cause.

transistorgit, 2023-04-07:
seems to be solved

closed 2023-04-07 by transistorgit