Dr-Gigavolt/dbus-aggregate-batteries #16: bad/slow data propagation

transistorgit, 2023-03-10:
This is more like a question or maybe an improvement suggestion

My ESS tries to regulate the grid injection to zero (grid feed in setpoint 50W) what is working quite nice. As soon I switched over to Aggregate batteries, the feed in regulation is much worse.

What can be the cause? DC values from the battery shoudn't be used for grid regulation, but maybe the system gets slower with AggregateBatteries, 2*Daly Serial Battery and Smartshunt running in parallel, compared to just only having smarshunt alone before?

Graph: without Aggregate Batteries until 06:30, and with AB afterwards
[image]

[image]

Dr-Gigavolt, 2023-03-10:
Hello, since I use the DBusMonitor instead of VEdbusItemImport I haven't observe any delays. The serial-batteries send once per second, AggregateBatteries as well. My ESS set point is zero and it regulates a bit negative, apx. -20...-50W. Have you set both to AggregateBatteries? System setup -》Battery monitor and DVCC -》 Controling BMS? If everything ok, please check in device list whether AggregateBatteries updates values with the same speed as the SerialBatteries. Optionally you can add a command logging the time at the beginning of _update() and check if it writes once per second.
Best regards, 
Anton

Dr-Gigavolt, 2023-03-10:
I've just tried to bypass my service - no difference in grid set point observed.

transistorgit, 2023-03-11:
I found a bug in the serial port implementation of serialbattery. Will test a bit more and add it to my pull request there.
Problem was that the Dalys send broken packages quite often and then and the serial_read function waits for up to 15s for the missing bytes. I think this may be the cause for the slow update.

closed 2023-03-11 by transistorgit