Dr-Gigavolt/dbus-aggregate-batteries #17: doesn´t work in my setup

kr0815, 2023-03-11:


Dr-Gigavolt, 2023-03-11:
Hello, according the log both batteries and Multi are found. But some parameter read returns None. Please send me the DBus transmission of your batteries. Command dbus-spy in console.

kr0815, 2023-03-11:
thanks for your fast reply

This is what i get after dbus-spy
[image]

transistorgit, 2023-03-11:
I think this is the same Problem like in issue #15.

Some values are not on the bus, thus the error messages and bailing out.
I fixed the primary error by changing aggregatebatteries.py@367 to:
                        `if (cellVoltage is not None and cellVoltage > MAX_CELL_VOLTAGE):`
but the initial problem is that the serialbatteries are not up yet. you can add a sleep in rc.local or at begining of main() in aggregatebatteries.py as a work around as described in #15

kr0815, 2023-03-11:
my problem is much bigger
I don´t know what i did , almost nothing, just installed the second battery

Now i get almost no messages from MQTT, looks like the same as keep_alive is 0 - very strange

kr0815, 2023-03-12:
I found the fault - dbus-serialbattery has problems reading highest / lowest cell on a Daly BMS if connected by RS485
dbus-aggregate-batteries needs these values

I switched from RS485 to UART, now it´s working

transistorgit, 2023-03-12:
yes, the RS485 handling is buggy. I opened a pull request for a solution there
https://github.com/Louisvdw/dbus-serialbattery/pull/484

Dr-Gigavolt, 2023-03-12:
> thanks for your fast reply
> 
> This is what i get after dbus-spy [image]

You can navigate by arrows in dbus-spy and see the path values:
[image]

Dr-Gigavolt, 2023-03-12:
> I think this is the same Problem like in issue #15.
>
> Some values are not on the bus, thus the error messages and bailing out. I fixed the primary error by changing aggregatebatteries.py@367 to: `if [...]

This is not safe. It is better to raise an exception than to keep program running with old and invalid values. The ESS could try to over/undercharge the batteries. It is not correct to allow the normally running program to rely on emergency brake of the BMS. This should be understood as the ultima ratio to avoid disaster.

The best solution would be to disable charging and discharging completely in such a case, but the parameters AllowToCharge and AllowToDischarge are just indicators and are ignored by ESS.

Dr-Gigavolt, 2023-03-12:
> I found the fault - dbus-serialbattery has problems reading highest / lowest cell on a Daly BMS if connected by RS485 dbus-aggregate-batteries needs these values
> 
> I switched from RS485 to UART, now it´s working

Excellent. Now we can close this issue :-)

closed 2023-03-12 by Dr-Gigavolt