Dr-Gigavolt/dbus-aggregate-batteries #19: charge current jumping?

kr0815, 2023-03-14:
Maybe an issue, maybe my fault:

Charging 2 batteries - (SOC at the moment over 90%) - current ist jumping around
If i understand all settings - they should load with max current

I don´t see any wrong limotations, cell voltages are way under criticial points

I expect the charger to fully load with it´s maximum of 70A - it´s jumping between about 30 to 45A

One of my Daly BMS is not very accurate regarding Amps, but in my understanding, dbus-aggregate-batteries only uses the internal shunt?

Dr-Gigavolt, 2023-03-14:
Please check if both instances of dbus-serial-battery send /System/NrOfModulesBlockingCharge equal to zero. If not, it is the issue https://github.com/Louisvdw/dbus-serialbattery/issues/458 and workaround is to set the parameters for dbus-serial-battery more liberal than for  dbus-aggregate-batteries, even if OWN_CHARGE_PARAMETERS = True. Otherwise review the settings.py and the logfile.

Dr-Gigavolt, 2023-03-14:
dbus-aggregate-batteries sums currents of all Victron chargers/inverters and adds or subtracts max. one Smart shunt measuring DC consumption.

kr0815, 2023-03-14:
NrOfModulesBlockingCharge  was always zero

I thought dbus-aggregate-batteries is using the victron-internal shunt to meassure current, calculating culoumb instead of using the BMS Values?  wouldn´t that way make more sense?

Dr-Gigavolt, 2023-03-14:
I was confused by the expression "shunt". There is no shunt in the MPPT, it is a Hall sensor. In Multi/Quattro ... I don't know. But you understood correctly, it uses the internal current sensors of all registered Victron devices.

kr0815, 2023-03-14:
Now i´m confused :-)

The marked sensor - this is where your software calculates the SOC from or from the 2 BMS?

You have much more experience than me

Does it make sense to add a smartshunt?
Is the internal shunt / hall-sensor, whatever inaccurate?

[image]

Dr-Gigavolt, 2023-03-14:
AggregateBatteries on Virtual is correct. The marked sensor is the one read by my program. And according my experience it is accurate. MPPT's sensor as well.

[image]
[image]

If you set up settings.py correctly, all internal current sensors are found by my program.

Dr-Gigavolt, 2023-03-14:
I sent you my phone number per eMail, you can call me. I speak German too.

transistorgit, 2023-05-11:
is this fixed?
we worked a lot on the daly driver in the last weeks. please try the latest version.

concerning jumping currents, I read that daly had a firmware fix, so maybe this is a daly internal problem?

kr0815, 2023-05-11:
where could i get this firmware update?

where to get updates for the daly driver?

transistorgit, 2023-05-11:
for the driver, just install the current https://github.com/Louisvdw/dbus-serialbattery

for firmware: don't know, there are some files in newsgroups floating around like here. but never tried it myself
https://www.microcharge.de/forum/forum/index.php?thread/1102-sammelthread-daly-bms-firmware-tools/

kr0815, 2023-05-12:
was not that easy, but i managed it :-)
To install the actual version, i had to choose the nightly build

closed 2023-10-25 by Dr-Gigavolt