Dr-Gigavolt/dbus-aggregate-batteries #20: Freeze after Multi disapear/reapear

transistorgit, 2023-03-19:
Hi,

I found today, that when the Multiplus disappears shortly (because settings update by VRM remote settings upload), the aggregate batteries driver just freezes its values. All other parts like SerialBatteries work normal.

Log (in UTC):
INFO:root:2023-03-19 07:58:44.612685: Max. charging voltage: 56.0V
INFO:root:2023-03-19 07:58:44.613781: Max. charging current: 21.0A Max. cell voltage: 3.279V
INFO:root:2023-03-19 07:58:44.614596: Max. dircharging current: 98.0A Min. cell voltage: 3.271V
ERROR:root:2023-03-19 09:43:01.669795: Victron current read error. Using BMS current and power instead.
INFO:dbusmonitor:com.victronenergy.vebus.ttyUSB2 disappeared from the dbus. Removing it from our lists
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB2 has device instance 290
INFO:dbusmonitor:com.victronenergy.vebus.ttyUSB2 disappeared from the dbus. Removing it from our lists
INFO:dbusmonitor:com.victronenergy.battery.ttyUSB1 disappeared from the dbus. Removing it from our lists
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB2 has device instance 290
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 3
INFO:dbusmonitor:com.victronenergy.vebus.ttyUSB2 disappeared from the dbus. Removing it from our lists
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB2 has device instance 290

Data Monitoring (in UTC+1)
[image]

Remote Console: (all values updating, but not aggragate bat)
[image]

after a manual ./restart.sh its ok again.

Dr-Gigavolt, 2023-03-19:
Hello, I found the mistake. I'll make a new commit ASAP.

Dr-Gigavolt, 2023-03-19:
Please try. Thanks.

Dr-Gigavolt, 2023-03-20:
Please try without Smart Shunt. I don't have it and cannot test.

transistorgit, 2023-03-22:
I unplugged the Smartshunt and removed it from the system in VRM.
After some minutes I forced a Multiplus reset by sending a new config. In the instance that the Multiplus disappears the AggregateBattery freezes its values in remote console.
This is the log:

INFO:root:2023-03-22 17:16:54.519274: Max. dircharging current: 0.0A; Min. cell voltage: 0.117V
INFO:root:2023-03-22 17:16:55.523458: Max. dircharging current: 140.0A; Min. cell voltage: 3.268V
INFO:dbusmonitor:com.victronenergy.battery.ttyUSB1 disappeared from the dbus. Removing it from our lists
INFO:root:2023-03-22 17:17:18.550437: Max. dircharging current: 0.0A; Min. cell voltage: 0.117V
INFO:root:2023-03-22 17:17:20.549922: Max. dircharging current: 140.0A; Min. cell voltage: 3.272V
INFO:root:2023-03-22 17:17:33.558515: Max. dircharging current: 0.0A; Min. cell voltage: 0.117V
INFO:root:2023-03-22 17:17:34.559231: Max. dircharging current: 140.0A; Min. cell voltage: 3.270V
INFO:root:2023-03-22 17:18:33.602991: Max. dircharging current: 0.0A; Min. cell voltage: -18.432V
INFO:root:2023-03-22 17:18:34.603059: Max. dircharging current: 140.0A; Min. cell voltage: 3.268V
INFO:root:2023-03-22 17:18:37.608634: Max. dircharging current: 0.0A; Min. cell voltage: 0.117V
INFO:root:2023-03-22 17:18:39.611544: Max. dircharging current: 140.0A; Min. cell voltage: 3.270V
INFO:root:2023-03-22 17:19:12.635970: Max. dircharging current: 0.0A; Min. cell voltage: -8.448V
INFO:root:2023-03-22 17:19:13.636879: Max. dircharging current: 140.0A; Min. cell voltage: 3.268V
ERROR:root:2023-03-22 17:20:09.681779: Victron current read error. Using BMS current and power instead.
INFO:dbusmonitor:com.victronenergy.vebus.ttyUSB2 disappeared from the dbus. Removing it from our lists
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB2 has device instance 290
INFO:dbusmonitor:com.victronenergy.vebus.ttyUSB2 disappeared from the dbus. Removing it from our lists
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB2 has device instance 290
INFO:dbusmonitor:com.victronenergy.battery.ttyUSB4 disappeared from the dbus. Removing it from our lists
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB4, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB4 has device instance 1

What are these min cell voltage logs? Seems broken. I never see such values in the gui.
[image]

Dr-Gigavolt, 2023-03-25:
Hello, the minimum cell voltage you can see under AggregateBatteries->Details:
[image]
It is the minimum value of paths '/System/MinCellVoltage' of all battery instances. Your values are negative, it means corrupt. Please check by dbus-spy.
[image]

transistorgit, 2023-04-07:
* broken values: I see them occasionally. The Daly driver is not so save implemented. I created a pull request with added checksum checks, but also this is not perfect. Not the fault of Aggregate Batteries anyhow
* frozen values: since these changes and the use of a installer that created proper service links the system seems to work flawless. maybe the fact that I now don't restart it as often as in the begining of the project also helps ;-)

for me this is no problem any more. close it for now.

closed 2023-04-07 by transistorgit