Dr-Gigavolt/dbus-aggregate-batteries #21: mppt not found

pit65535, 2023-03-22:
I have one SmartSolar MPPT VE.Can 250/100 rev2. If I activate NR_OF_MPPTS = 1 the logfile shows the following error:

INFO:root:2023-03-22 14:35:26.117637: Searching batteries: Trial Nr. 1
INFO:root:2023-03-22 14:35:26.121446: SerialBattery(LLT/JBD) found.
INFO:root:2023-03-22 14:35:26.122813: SerialBattery(LLT/JBD) found.
INFO:root:2023-03-22 14:35:26.123848: 2 batteries found.
INFO:root:2023-03-22 14:35:27.126188: Searching Multi/Quatro VEbus: Trial Nr. 1
INFO:root:2023-03-22 14:35:27.129719: MultiPlus-II 48/5000/70-50 found.
INFO:root:2023-03-22 14:35:28.132319: Searching MPPTs: Trial Nr. 1
INFO:root:2023-03-22 14:35:28.136054: 0 MPPT(s) found.
INFO:root:2023-03-22 14:35:29.133154: Searching MPPTs: Trial Nr. 2
INFO:root:2023-03-22 14:35:29.136687: 0 MPPT(s) found.
INFO:root:2023-03-22 14:35:30.134368: Searching MPPTs: Trial Nr. 3
INFO:root:2023-03-22 14:35:30.137727: 0 MPPT(s) found.
INFO:root:2023-03-22 14:35:31.135503: Searching MPPTs: Trial Nr. 4
INFO:root:2023-03-22 14:35:31.138936: 0 MPPT(s) found.
INFO:root:2023-03-22 14:35:32.136151: Searching MPPTs: Trial Nr. 5
INFO:root:2023-03-22 14:35:32.139733: 0 MPPT(s) found.
INFO:root:2023-03-22 14:35:33.136809: Searching MPPTs: Trial Nr. 6
INFO:root:2023-03-22 14:35:33.140542: 0 MPPT(s) found.
INFO:root:2023-03-22 14:35:34.137083: Searching MPPTs: Trial Nr. 7
INFO:root:2023-03-22 14:35:34.140517: 0 MPPT(s) found.
INFO:root:2023-03-22 14:35:35.137965: Searching MPPTs: Trial Nr. 8
INFO:root:2023-03-22 14:35:35.141504: 0 MPPT(s) found.
INFO:root:2023-03-22 14:35:36.139478: Searching MPPTs: Trial Nr. 9
INFO:root:2023-03-22 14:35:36.142942: 0 MPPT(s) found.
INFO:root:2023-03-22 14:35:37.140326: Searching MPPTs: Trial Nr. 10
INFO:root:2023-03-22 14:35:37.143796: 0 MPPT(s) found.
INFO:root:2023-03-22 14:35:38.140700: Searching MPPTs: Trial Nr. 11
INFO:root:2023-03-22 14:35:38.144563: 0 MPPT(s) found.
ERROR:root:2023-03-22 14:35:38.145664: Required number of MPPTs not found. Exiting.

Dr-Gigavolt, 2023-03-22:
Hello, please look in dbus-spy if the MPPT is visible.

pit65535, 2023-03-22:
in dbus-spy the MPPT is visible:
Services
com.victronenergy.adc
com.victronenergy.battery.aggregate                                                                                                                                                                                        AggregateBatteries
com.victronenergy.battery.ttyACM0                                                                                                                                                                                      SerialBattery(LLT/JBD)
com.victronenergy.battery.ttyACM1                                                                                                                                                                                      SerialBattery(LLT/JBD)
com.victronenergy.fronius
com.victronenergy.hub4
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.modbustcp
com.victronenergy.platform
com.victronenergy.qwacs
com.victronenergy.settings
com.victronenergy.solarcharger.socketcan_can0_vi0_uc229774                                                                                                                                                SmartSolar MPPT VE.Can 250/100 rev2
com.victronenergy.system
com.victronenergy.vebus.ttyS4                                                                                                                                                                                      MultiPlus-II 48/5000/70-50
com.victronenergy.vecan.can0

pit65535, 2023-03-22:
Yes, it is connected by CAN, I can order a VE-direct Cable :-)

details from dbus-spy:
com.victronenergy.solarcharger.socketcan_can0_vi0_uc229774
Alarms/HighVoltage                                                                                                                                                                                                                          0
Alarms/LowVoltage                                                                                                                                                                                                                           0
Connected                                                                                                                                                                                                                                   1
CustomName
Dc/0/Current                                                                                                                                                                                                                              0.6
Dc/0/Temperature                                                                                                                                                                                                                            -
Dc/0/Voltage                                                                                                                                                                                                                            52.83
DeviceInstance                                                                                                                                                                                                                              0
DeviceOffReason                                                                                                                                                                                                                             -
Devices/0/CustomName
Devices/0/DeviceInstance                                                                                                                                                                                                                    0
Devices/0/FirmwareVersion                                                                                                                                                                                                              201727
Devices/0/Nad                                                                                                                                                                                                                              36
Devices/0/ProductId                                                                                                                                                                                                                     41237
Devices/0/ProductName                                                                                                                                                                                     SmartSolar MPPT VE.Can 250/100 rev2
Devices/0/ServiceName                                                                                                                                                              com.victronenergy.solarcharger.socketcan_can0_vi0_uc229774
Devices/0/VregLink

Dr-Gigavolt, 2023-03-22:
Wait with ordering, I'll have a look.

Dr-Gigavolt, 2023-03-25:
Hello, it should be easy. Write in settings.py:
MPPT_KEY_WORD = 'com.victronenergy.solarcharger.socketcan_can0'

pit65535, 2023-03-27:
Yes, thank you very much, now my mppt ist found!

Dr-Gigavolt, 2023-03-27:
Closed as complete.

closed 2023-03-27 by Dr-Gigavolt