Dr-Gigavolt/dbus-aggregate-batteries #22: SoC calculation incorrect

Leupi123, 2023-03-29:
It seems the SoC calculation is not correct.
I´m currently using 2 different sized batteries (280Ah and 600Ah), the resulting SoC is as shown below:

[image]

But i would expect a SoC of 57%.

Dr-Gigavolt, 2023-03-29:
You are right, I didn't think about different batteries. It is easy to correct. But do you trust the SoC measurement of your BMS if they show 10% difference although connected parallel?

Dr-Gigavolt, 2023-04-04:
I changed it to weighted average. Will be available at the next update.

closed 2023-05-16 by Dr-Gigavolt

Leupi123, 2023-05-17:
Thanks, I already installed the latest version. Looks good now!