Dr-Gigavolt/dbus-aggregate-batteries #24: driver not autostarting any more

transistorgit, 2023-04-04:
Aggregate batteries was suddenly not starting automatically any more. I didn't make a venus updates or something (always 2.92), but now the link in /service didn't work any more, it just disappears. Seems that I got problems with the file system become read only. I found a hint here:
https://github.com/Louisvdw/dbus-serialbattery/issues/38

Seems, that you have to mount the filesystem rw to set the link. As soon I did it, it worked again.
The link command in rc.local is not enough, so I added a install script.
If you like I can add a pull request.

```

#!/bin/bash
set -x
#credits to https://github.com/Louisvdw/dbus-serialbattery

DRIVERNAME=dbus-aggregate-batteries

#handle read only mounts
sh /opt/victronenergy/swupdate-scripts/remount-rw.sh

#install
rm -rf /opt/victronenergy/service/$DRIVERNAME
rm -rf /opt/victronenergy/$DRIVERNAME
mkdir /opt/victronenergy/$DRIVERNAME
ln -s /data/$DRIVERNAME/service /opt/victronenergy/service/$DRIVERNAME

#restart if running
pkill -f "python .*/$DRIVERNAME.py"

# add install-script to rc.local to be ready for firmware update
filename=/data/rc.local
if [ ! -f $filename ]; then
    echo "#!/bin/bash" >> $filename
    chmod 755 $filename
fi
grep -qxF "sh /data/$DRIVERNAME/install.sh" $filename || echo "sh /data/$DRIVERNAME/install.sh" >> $filename

# also set link directly to get it started by svscan at install time
ln -s /data/$DRIVERNAME/service /service/$DRIVERNAME

```

Dr-Gigavolt, 2023-04-04:
I had never such a case. I had 2.92 and now I have 3.00, it starts without any issues. Would you like to try this version?
I'm not experienced with Linux. Louis explained me that /data is persistent some time ago. Thanks for your script. Let's keep it as an option, I would prefer to keep the tool as simple as possible as long as it works in most cases.

transistorgit, 2023-04-07:
probably some kind of timing/race condition on startup. if rc.local is run before daemontools purges the /service folder, the link is gone, otherwise not. For me, the method with the installer is more universal and safer in a long-time view. (once my battery is fully completed I hopefully wont touch it for years. and my future self won't remeber all these quirks then) .
No problem for me if you want to keep the project simple. just close the issue and keep it as a reference for future users.
Thank you for your work.

closed 2023-04-07 by transistorgit