Dr-Gigavolt/dbus-aggregate-batteries #25: Start issue

djkakadu, 2023-04-08:
Sorry for nebwie question. But what exactly i need put into rc.local

I tryed several options to start but nothing happened.
Thank you

djkakadu, 2023-04-08:
If I try start python3 aggregatebatteries.py

These error here:

```
root@einstein:/data/dbus-aggregate-batteries# python3 aggregatebatteries.py
  File "aggregatebatteries.py", line 458
    Current = Current_VE                                                                            # BMS current overwritten only if no exception raised
                                                                                                                                                        ^
TabError: inconsistent use of tabs and spaces in indentation

```

Dr-Gigavolt, 2023-04-08:
Hello, the error seems to be happened by editing directly in github and mixing tabs and spaces. I just corrected it. You can replace the tabs at the begin of the line 458 by spaces.

Dr-Gigavolt, 2023-04-08:
Into rc.local you should add the line
ln -s /data/dbus-aggregate-batteries/service /service/dbus-aggregate-batteries

(see readme.md)

djkakadu, 2023-04-08:
> Hello, the error seems to be happened by editing directly in github and mixing tabs and spaces. I just corrected it in github. You can replace the tabs at the begin of the line 458 by spaces in your mashine.

These help to start via console everything work good but.....
If i close console aggregatebatteries dissapear. How its start on startup rc.local doesnt work thank you.

Dr-Gigavolt, 2023-04-08:
Of course it stops if you close the console. This is good only for tests.
If you follow the readme file it should start automatically after restart of Venus.
There is only one case known when it didn't work (any more) and here is the solution:
https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/24

djkakadu, 2023-04-09:
Everything work now. 💯
After hundred of tries i lost permission on run to 744...... My mystake..

Thank you again.

Best regards from Slovakia 😉

Dr-Gigavolt, 2023-04-09:
Zdravím naspäť na Slovensko 😀

closed 2023-04-09 by Dr-Gigavolt

djkakadu, 2023-04-09:
Tak keď po slovensky tak len ešte raz poďakujem za skvelú robotu. Presne toto som potreboval k mojím dvom jkbms. Chvíľku som používal aggregate batteries čo je k dispozícii v package managery, ale to len spočitavalo hodnoty, toto je premyslené a omnoho lepšie.!
[Screenshot_20230409_164637_nl.victronenergy_edit_96312921096761.jpg](https://user-images.githubusercontent.com/96320986/230779633-75bd6178-0bb1-4d45-9935-be0fa3acf95a.jpg)