Dr-Gigavolt/dbus-aggregate-batteries #31: Calculation of MaxChargeCurrent and MaxDischargeCurrent for different configurations

kakariki1, 2023-04-10:
Hello,

this is an addendum to https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/10

I'm running 3 different systems:

1. single phase system
1x MultiPlus-II 48/5000/70-50
1x Fronius Symo 12.5-3-M
1x 18 LiFePO4 batteries with JK BMS 2A24S20P in 18S1P configuration

2. single phase system
1x MultiPlus-II 48/5000/70-50
1x Fronius Symo 12.5-3-M
2x 18 LiFePO4 batteries with JK BMS 2A24S20P in 18S2P configuration

3. three phase system
3x MultiPlus-II 48/5000/70-50
1x Fronius Symo 20.0-3-M
3x 18 LiFePO4 batteries with JK BMS 2A24S20P in 18S3P configuration

All three systems are running the same Venus OS 2.92, dbus-serialbattery and your dbus-aggregate-batteries.
I know, aggregate-batteries is not needed for the 1st system, but I want to keep the software the same for all.

I always noticed quite a great cell drift in the 2nd system and now found the reason, a wrong charge and discharge current calculation.
In this configuration the currents have to be divided by two, because 2 batteries are connected to one MultiPlus.
So I added a new setting NR_OF_BATTERIES_PER_MULTIPLUS and divide MaxChargeCurrent and MaxDischargeCurrent by this value.
Not sure if this interferes with anything. Does this make sense ?

In addition I added a state-machine to prevent a battery-low warning and added a safety feature, but that's another topic.
It's fully tested in all 3 configurations and works flawless so far ...

The source code is attached, changes are marked by "lakeroe" and files are renamed to .txt
If you're interested feel free to add it (or parts) in your source code.

Best regards,
lakeroe

[settings.txt](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/files/11192016/settings.txt)
[aggregatebatteries.txt](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/files/11192017/aggregatebatteries.txt)

Dr-Gigavolt, 2023-04-10:
Hello and thank you. I'll review your code at the next opportunity.

Dr-Gigavolt, 2023-04-10:
I've just looked through your hardware description. How do you run a 3-phases Symo on single phase? All 3ph parallel? Does the Symo accept it? 
Why don't you add one more Multi (resulting in 2 x 3ph system) and put all batteries in parallel? That is what Victron recommends. What you do cannot work properly. The most full (or most empty) battery will limit another two.

kakariki1, 2023-04-11:
The description is probably a bit unclear. The phases are only regarding the Victron Multiplus, all Fronius Symo are connected on 3 phases ...

Dr-Gigavolt, 2023-04-13:
Back to your system, why don't you connect all batteries in parallel as recommended by Victron?

Dr-Gigavolt, 2023-04-14:
I mean all batteries in one system should be connected in parallel, only one DC busbar per system. If not, how should batteries Nr. 2 and 3 be fully charged if the battery Nr. 1 says "I am full" and the battery driver, either SerialBattery or Aggregate Batteries, stops charging?

Dr-Gigavolt, 2023-04-14:
In the measurement/balancing wires. I see no (small) fuses there.

kakariki1, 2023-04-14:
Yes you're right. I haven't seen anybody using fuses in the balancing wires so far.
In case of a short circuit i reckon the small wires or connectors vaporize within a very short time.
But I agree, it's not the most secure or elegant solution ...

kakariki1, 2023-04-14:
Please find a VERY simplified schematic diagram from all my 3 systems below:

[image]

Dr-Gigavolt, 2023-04-14:
After all it is as I understood initially. What is the reason not to make 3x ((18s x 2p) + BMS)) and connect all in parallel? Are they faraway each from other? In this case they should have separate Venus devices.

kakariki1, 2023-04-15:
I really don't understand what you mean. All of my 3 systems work perfectly and I won't change anything. I also think I wired everything exactly as Victron recommends in there "Wiring unlimited" book.
Maybe that's not clear for you: All 3 systems are at 3 different locations a few km apart ...

My intention when opening this post was to share my experience with different configurations and if you want, to make your driver more flexible to various configurations, not to discuss different wirings.

Anyway thanks for your great work !

Dr-Gigavolt, 2023-04-15:
I need to understand why you needed do change the code. So all three systems are completelly independend, in different locations. That is what I wanted to need. I thought they are in one house with one common Venus device. OK, now I can look on your code modification. Thanks.

Dr-Gigavolt, 2023-04-16:
I'm reviewing your code modification now and cannot understand the problem. You reuse charge/discharge parameters from SerialBattery, let's make an example:

Batteries = 2
Battery1: I_charge_max1 = I_discharge_max1 = 100A
Battery2: I_charge_max2 = I_discharge_max2 = 80A

Results:
I_charge_max = I_discharge_max = 2*min(100A, 80A) = 160A
If the resistances of batteries and cables are equal, both batteries will be charged or discharged by max. 80A.

Further, I don't understand PER_MULTIPLUS. If Multis are connected in parallel or as 3 phases they act as a single device.

Thanks for the 0,1A limitation of discharge current, I'll implement it on next commit.

Dr-Gigavolt, 2023-04-16:
I see you use VeDbusItemImport. I don't recommend it due to its inefficiency. Better is to get all what you need by Dbus monitor. It was a solution of one of first issues (the program was too slow).

kakariki1, 2023-04-16:
Okay, let me try to explain:

The settings for my 3 different systems are:
`System                         1  2  3`
`NR_OF_BATTERIES                1  2  3`
`NR_OF_BATTERIES_PER_MULTIPLUS  1  2  1`

Regarding the calculation of MaxChargeCurrent/MaxDischargeCurrent the only thing I changed is dividing by NR_OF_BATTERIES_PER_MULTIPLUS:
MaxChargeCurrent     = self._fn._min(MaxChargeCurrent)     * NR_OF_BATTERIES / NR_OF_BATTERIES_PER_MULTIPLUS
MaxDischargeCurrent = self._fn._min(MaxDischargeCurrent) * NR_OF_BATTERIES / NR_OF_BATTERIES_PER_MULTIPLUS

Try to think about all 3 configurations and it should make sense (in any system the current should be the lowest allowed).
NR_OF_BATTERIES_PER_MULTIPLUS is only needed in my 2nd system.

Everything else (the state machine below) is just adding different security levels to prevent charge, discharge and battery-low warning, but that's a different story.

Regarding VeDbusItemImport:
Like you, this is my first Venus OS / Python project and I still don't understand everything. I "stole" some ideas from here:
https://github.com/t0bias-r/venusos_acload_prioritize

The "utils import *" is to get access to MIN_CELL_VOLTAGE and MAX_CELL_VOLTAGE.
I know it's not the most elegant solution but works for me.

Dr-Gigavolt, 2023-04-16:
> Okay, let me try to explain:
>
> The settings for my 3 different systems are: `System 1 2 3` `NR_OF_BATTERIES 1 2 3` `NR_OF_BATTERIES_PER_MULTIPLUS 1 2 1`
>
> Regarding the calculation of [...]

In the case 1 and 3 it is clear, of course. But my case is similar as yours, I have 3 Multis and 2 batteries. I just disabled my own parameter calculation to reproduce your preferences.

[image]
Each physical battery:
[image]
AggregateBatteries:
[image]

The max. 200A is spread into 2 batteries, each gets or gives max. 100A - correct.

Dr-Gigavolt, 2023-04-16:
> Everything else (the state machine below) is just adding different security levels to prevent charge, discharge and battery-low warning, but that's a different story.
>
> Regarding VeDbusItemImport: [...]

Thanks a lot for this proposal, I'll test later when will have some time for it.

> The "utils import *" is to get access to MIN_CELL_VOLTAGE and MAX_CELL_VOLTAGE. I know it's not the most elegant solution but works for me.

I prefer to collect data from SerialBatteries and calculate everything by my tool.

Dr-Gigavolt, 2023-04-16:
And now I'm thinking about Low Battery warning if max. discharge current is zero. Why should we avoid this warning? Is there any reason to set it to zero if the battery is not low?

kakariki1, 2023-04-16:
In my case the "Low Battery warning" often pops up if the battery is far away from empty and for sure not low ...

Dr-Gigavolt, 2023-04-16:
> In my case the "Low Battery warning" often pops up if the battery is far away from empty and for sure not low ...

But then there is no reason to set max. discharge current to zero. Or what do I misunderstand?

kakariki1, 2023-04-16:
That's exactly the point.
If MaxDischargeCurrent is zero (or below 0.1A) the warning occurs, if greater 0.1 then not.

That's my solution:
MaxDischargeCurrent = max (0.1 * NR_OF_BATTERIES / NR_OF_BATTERIES_PER_MULTIPLUS, MaxDischargeCurrent) # to prevent battery-low warning MaxDischargeCurrent is not allowed to be below 0.1A per Multiplus

Dr-Gigavolt, 2023-04-16:
> That's exactly the point. If MaxDischargeCurrent is zero (or below 0.1A) the warning occurs, if greater 0.1 then not.
>
> That's my solution: MaxDischargeCurrent = max (0.1 * NR_OF_BATTERIES / [...]

I understand the calculation. But not the reason. How can happen MaxDischargeCurrent = 0 if the battery is not empty? Bug in SerialBattery? In my case the alarm was always on before I set everything to False in SerialBattery. Even if I calculate parameters myself, I had some features of SerialBattery active to observe the results on the Dbus.

kakariki1, 2023-04-17:
> How can happen MaxDischargeCurrent = 0 if the battery is not empty? Bug in SerialBattery?

MaxDischargeCurrent = 0 happens when MIN_CELL_VOLTAGE (i.e. 3.00V) in dbus-serialbattery is reached. This only means to stop discharge, but not an empty or low batterie which is for example 2.60V.

There are many posts about this "problem" in the Victron Community, for example:
https://community.victronenergy.com/questions/191406/false-low-battery-voltage-alerts.html
https://community.victronenergy.com/questions/190864/multiplus-2-483000-unexplainable-low-battery-alarm.html

This is a "bug" or behaviour of the Multiplus II firmware not dbus-serialbattery.

Mark21041981, 2024-12-15:
> Yes you're absolutely right, and that's how it is connected, one DC busbar per system ... For example have a look at the three-phase system, I think the bus bar is clearly visible. And it works just perfect !
> 
> [image]

Hello, nice setup you have there. I noticed you are using an other BMS from JK than i'm currently using. 

But have you got a clue how to connect them to the Cerbo? I now use an canbus type-A cable from victron.

Link to inverte BMS from JK:
https://nl.aliexpress.com/item/1005007359327060.html?spm=a2g0o.productlist.main.5.55357Hn47Hn4IS&algo_pvid=41955a04-71e4-409e-936a-80b6b5380a09&algo_exp_id=41955a04-71e4-409e-936a-80b6b5380a09-2&pdp_npi=4%40dis%21EUR%21414.57%21129.67%21%21%213084.59%21964.79%21%402103868817342712294145373ed978%2112000040415687914%21sea%21NL%210%21ABX&curPageLogUid=W5U1itOHZ6T1&utparam-url=scene%3Asearch%7Cquery_from%3A

YetAnotherOneGuru, 2025-04-03:
> Hello, nice setup you have there. I noticed you are using an other BMS from JK than i'm currently using.
>
> But have you got a clue how to connect them to the Cerbo? I now use an canbus type-A cable [...]

Hi Mark, There are currently 2 ways to connect the latest JKBMSes to the Cerbo - either with a lan cable (there are 2 types - A and B -, pls search in the victron community pages, which one to use with JKBMS) or the serial connection starting with a lan-like cable going into a small adapter which outputs to a USB port for the cerbo. With this last version, You can use the serialbattery driver (also in GitHub), that the aggregate-batteries relies on. Please search in Youtube, there are many videos covering either of the 2 options.

However, I have the feeling, that this question has nothing to do with the aggregate-batteries driver, nor is it related to the raised issue for the MaxChargeCurrent above.

YetAnotherOneGuru, 2025-04-03:
> > How can happen MaxDischargeCurrent = 0 if the battery is not empty? Bug in SerialBattery?
> 
> MaxDischargeCurrent = 0 happens when MIN_CELL_VOLTAGE (i.e. 3.00V) in dbus-serialbattery is reached. This only means to stop discharge, but not an empty or low battery which is for example 2.60V.
> 
> There are many posts about this "problem" in the Victron Community, for example: https://community.victronenergy.com/questions/191406/false-low-battery-voltage-alerts.html https://community.victronenergy.com/questions/190864/multiplus-2-483000-unexplainable-low-battery-alarm.html
> 
> This is either a "bug" or the intended behaviour of the Multiplus II firmware not dbus-serialbattery.

The issue still exists in latest version of dbus-serialbattery together with current Multiplus firmware and latest cerbo software version.
When MIN_CELL_VOLTAGE is reached, the Multiplus switches to "Charge" and brings the Voltage to approx +1,2V. I also have to work-around the issue with a very low discharge current (0,001 C works, but You report the minimum to be even lower with 0,1A) which will keep the battery there almost forever. I configured the MIN_CELL_VOLTAGE at 2,8V and switch to the very low discharge current at 3V. Note that this setting kicks in when the lowest cell reaches the limits - probably for safety reasons.

Dr-Gigavolt, 2025-04-05:
Multiplus does not care about MIN_CELL_VOLTAGE. When either DCL or SoC is set zo zero, the Multi starts to charge from grid up to the minimum SoC setting of ESS. If my SoC counter is activated, its zeroing can be disabled:

            if (MinCellVoltage <= settings.MIN_CELL_VOLTAGE) and settings.**ZERO_SOC**:
                self._ownCharge = 0  # reset Coulumb counter to 0%