Dr-Gigavolt/dbus-aggregate-batteries #32: Keine Funktion mit Daly BMS

Gr33n93, 2023-04-19:
Hello,
Is it possible that this driver does not work with a Daly BMS?

These are the errors I get when the driver is installed:

INFO:root:2023-04-19 10:18:09.839286: Searching batteries: Trial Nr. 2
INFO:root:2023-04-19 10:18:09.841160: SerialBattery(Daly) found.
INFO:root:2023-04-19 10:18:09.841644: SerialBattery(Daly) found.
INFO:root:2023-04-19 10:18:09.842025: SerialBattery(Daly) found.
INFO:root:2023-04-19 10:18:09.842438: 3 batteries found.
ERROR:root:2023-04-19 10:18:10.844609: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 1
ERROR:root:2023-04-19 10:18:11.845549: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 2
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyUSB8, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyUSB8 has device instance 294
ERROR:root:2023-04-19 10:18:12.845903: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 3
ERROR:root:2023-04-19 10:18:13.846636: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 4
ERROR:root:2023-04-19 10:18:14.847536: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 5

Otherwise, if it does not work, is there a quick uninstall option?

mistermo0404, 2023-04-27:
Hello,
i have the same issue with two Daly BMS. They are both connected with individual USB-UART Adapters and visible in VenusOS via SerialBattery.
The Errors are the same as above.

Best regards
Mo

Dr-Gigavolt, 2023-04-27:
Hello, there were already troubles with Daly, see issues 15, 17 and 20. I cannot test is, I have no such a BMS. It seems the SerialBattery with Daly sends invalid data (NoneType), in this case my driver reacts correctly and raises exception.

If you want to remove: just delete all the stuff you copied in. It is not Windows :-)

transistorgit, 2023-05-11:
we worked a lot on the daly driver in the last weeks. please try the latest version.

Dr-Gigavolt, 2023-05-16:
I close this issue because it does not concern my code.

closed 2023-05-16 by Dr-Gigavolt