Dr-Gigavolt/dbus-aggregate-batteries #33: Error

sqlexception, 2023-05-08:
to reporduce, i restarted serial battery 

`INFO:dbusmonitor:com.victronenergy.battery.c8478ce93652 disappeared from the dbus. Removing it from our lists
ERROR:root:2023-05-08 21:45:57.114693: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'. Read trial nr. 1
ERROR:root:2023-05-08 21:45:58.115224: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'. Read trial nr. 2
ERROR:root:2023-05-08 21:45:59.115960: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'. Read trial nr. 3
ERROR:root:2023-05-08 21:46:00.116702: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'. Read trial nr. 4
ERROR:root:2023-05-08 21:46:01.116993: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'. Read trial nr. 5
ERROR:root:2023-05-08 21:46:02.118055: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'. Read trial nr. 6
ERROR:root:2023-05-08 21:46:03.118167: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'. Read trial nr. 7
ERROR:root:2023-05-08 21:46:04.118337: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'. Read trial nr. 8
ERROR:root:2023-05-08 21:46:05.119015: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'. Read trial nr. 9
ERROR:root:2023-05-08 21:46:06.119053: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'. Read trial nr. 10`

Dr-Gigavolt, 2023-05-11:
Hello, do you have Daly BMS? Then please see the issue 32. I cannot do anything if receiving invalid data.

Dr-Gigavolt, 2023-05-16:
I close this issue because it does not concern my code.

closed 2023-05-16 by Dr-Gigavolt