Dr-Gigavolt/dbus-aggregate-batteries #34: Eintrag Geräteliste erscheint nicht / PV-Ladegerät wird nicht mehr angezeigt!

elektroniktommy, 2023-05-12:
Hello , unfortunately "aggregate-batteries" doesn't work for me. At the moment I only have a DIY battery that is integrated with “dbus-serial-battery” and works. After installing “aggregate-batteries” my PV charger disappears from the “GUI” on the CerboGX. A menu item for the tool does not appear either. If I turn it off again, everything goes back to normal. I have an MPPT RS450/200, 3xMP2/5000, GerboGX and SmartShunt.

Do you have an idea what could be the reason?

Greetings Tommy

Hallo , leider funktioniert “aggregate-batteries” bei mir nicht. Zur Zeit habe ich nur einen DIY-Akku der mit “dbus-serial-battery” eingebunden ist und funktioniert. Nach der Installation von “aggregate-batteries” verschwindet mein PV-Ladegerät von der “GUI” auf dem CerboGX. Ein Menupunkt für das Tool taucht auch nicht auf. Wenn ich es wieder deaktiviere wird wieder alles normal angezeigt. Ich habe einen MPPT RS450/200, 3xMP2/5000, GerboGX und SmartShunt.

Hast Du eine Idee woran das liegen kann?

Gruß Tommy

Dr-Gigavolt, 2023-05-12:
Hi Tommy,
German is OK for me but we use english here to make the stuff readable for as many people as possible 😀.
Please write which BMS you have and attach the schematic diagram of your installation as well as the screenshot of dbus-spy. You are the first guy reporting about MPPT RS. I have no experience with this device yet.

wormsman, 2023-05-12:
Hello I have running a similar system and everything ist fine
3 Multiplus 48/5000
1 MPPT RS 450/100
1 Smartshunt for SOC als Battery Monitor 
2 Serial Battery (JK-BMS) with actual prerelase version 1.0 from 20230505

Following my settings in aggregate battery
```
# Version 2.3

NR_OF_BATTERIES = 2                                     # Nr. of physical batteries to be aggregated
NR_OF_MPPTS = 1                                         # Nr. of MPPTs

BATTERY_KEY_WORD = 'com.victronenergy.battery.tty'      # Key world to identify services of physical Serial Batteries
BATTERY_NAME_KEY_WORD = 'SerialBattery'                 # Key world to identify the name of batteries (to exclude SmartShunt)
BATTERY_NAME_PATH = '/CustomName'                       # What defines the battery name
#BATTERY_NAME_PATH = '/HardwareVersion'

MULTI_KEY_WORD = 'com.victronenergy.vebus.tty'          # Key world to identify service of Multis/Quattros (or cluster of them)
MPPT_KEY_WORD = 'com.victronenergy.solarcharger.tty'    # Key world to identify services of solar chargers (RS not tested, only SmartSolar)
SMARTSHUNT_NAME_KEY_WORD = 'SmartShunt'                 # Key world to identify services of SmartShunt (not tested)
                                                        # specify more precisely if more SmartShunts are in systems

SEARCH_TRIALS = 10                                      # Trials to identify of all batteries before exit and restart
READ_TRIALS = 10                                        # Trials to get consistent data of all batteries before exit and restart

CURRENT_FROM_VICTRON = True                             # If True, the current measurement by Multis/Quattros and MPPTs is taken instead of BMS
DC_LOADS = False                                        # If DC loads with Smart Shunt present, can be used for total current measurement 
INVERT_SMARTSHUNT = False                               # False: Current subtracted, True: Current added
OWN_SOC = True                                          # If True, the self calculated charge indicators are taken instead of BMS
CHARGE_SAVE_PRECISION = 0.0025                          # Trade-off between save precision and file access frequency

OWN_CHARGE_PARAMETERS = False                            # Calculate own charge/discharge control parameters (True) from following settings
                                                        # or use them from battery driver (False)
CHARGE_VOLTAGE = 3.5                                    # Constant voltage charge = this value * nr. of cells
MAX_CELL_VOLTAGE = 3.6                                 # If reached by 1-st cell, the charger voltage is clamped to the measured value
DISCHARGE_VOLTAGE = 2.7                                 # If reached, discharge current set to zero
MIN_CELL_VOLTAGE = 2.5                                  # If reached, discharge current set to zero
VOLTAGE_SET_PRECISION = 0.06                            # To be subtracted from the calculated max. charge voltage if MAX_CELL_VOLTAGE is exceeded

MAX_CHARGE_CURRENT = 250                                # Max. charge current at normal conditions
MAX_DISCHARGE_CURRENT = 250                             # Max. discharge current at normal conditions

# settings limiting charge and discharge current if at least one cell gets full or empty
# the lists may have any length, but the same length for voltage and current
# linear interpolation is used for values between
CELL_FULL_LIMITING_VOLTAGE = [CHARGE_VOLTAGE - 0.1, CHARGE_VOLTAGE, MAX_CELL_VOLTAGE]           # [min, .... ,max]
CELL_FULL_LIMITED_CURRENT =  [1, 0.05, 0]
CELL_EMPTY_LIMITING_VOLTAGE = [DISCHARGE_VOLTAGE - 0.1, DISCHARGE_VOLTAGE, MIN_CELL_VOLTAGE]    # [min, .... ,max]
CELL_EMPTY_LIMITED_CURRENT =  [0, 0.05, 1]

LOGGING = 2                                             # 0: no logging, 1: print to console, 2: print to file
LOG_VOLTAGE_CHANGE = 0.1                                # change of max. charge voltage to be logged
LOG_CURRENT_CHANGE = 5                                  # change of max. charge/discharge current to be logged
```

Settings from serial battery config.ini
```
[DEFAULT]

; If you want to add custom settings, then check the settings you want to change in "config.default.ini"
; and add them below to persist future driver updates.
; Battery Current limits
MAX_BATTERY_CHARGE_CURRENT    = 225.0
MAX_BATTERY_DISCHARGE_CURRENT = 225.0

BLOCK_ON_DISCONNECT = True
; Choose the mode for voltage / current limitations (True / False)
; False is a step mode. This is the default with limitations on hard boundary steps
; True is a linear mode. For CCL and DCL the values between the steps are calculated for smoother values (by WaldemarFech)
;                        For CVL max battery voltage is calculated dynamically in order that the max cell voltage is not exceeded
LINEAR_LIMITATION_ENABLE = True

CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT = 0.040

; -- Cell Voltages
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.90
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.48
FLOAT_CELL_VOLTAGE = 3.40

; --------- SOC limitation (affecting CCL/DCL) ---------
; Description: Maximal charge / discharge current will be increased / decreased depending on State of Charge,
;              see CC_SOC_LIMIT1 etc.
; Example: The SoC limit will be monitored to control the currents.
; Charge current control management enable (True/False).
CCCM_SOC_ENABLE = False
; Discharge current control management enable (True/False).
DCCM_SOC_ENABLE = False
```

elektroniktommy, 2023-05-15:
> Hi Tommy, German is OK for me but we use english here to make the stuff readable for as many people as possible 😀. Please write which BMS you have and attach the schematic diagram of your [...]

Hello,

found the problem. It was me, had a typo in the rc.local.

I have a Solarlock BMS(JBD-SP25S003-L16S-100A-B-U-R)

Unfortunately, there is no circuit diagram at the moment, as everything is still under construction!

Thanks for the help

dbus-spy
`Services
com.victronenergy.adc
com.victronenergy.battery.aggregate                           AggregateBatteries
com.victronenergy.battery.ttyS7                             SmartShunt 500A/50mV
com.victronenergy.battery.ttyUSB0                         SerialBattery(LLT/JBD)
com.victronenergy.fronius
com.victronenergy.grid.cg_BW08700560014 Carlo Gavazzi EM24 Ethernet Energy Meter
com.victronenergy.hub4
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.modbustcp
com.victronenergy.packageManager                                 Package Manager
com.victronenergy.platform
com.victronenergy.pvinverter.http_34                                     OpenDTU
com.victronenergy.pvinverter.http_47                                     OpenDTU
com.victronenergy.qwacs
com.victronenergy.settings
com.victronenergy.solarcharger.socketcan_can0_vi0_uc261303 SmartSolar MPPT RS 45
com.victronenergy.system
com.victronenergy.vebus.ttyS4                         MultiPlus-II 48/5000/70-50
com.victronenergy.vecan.can0
`

closed 2023-05-16 by Dr-Gigavolt