Dr-Gigavolt/dbus-aggregate-batteries #37: Error: '>' not supported between instances of 'float' and 'NoneType'

gnissahr, 2023-05-18:
I am trying to combine 3 serial batteries into one agrgate one. 3x JK BMS. Power measurment by Lynx Shunt.
Currently running with the Lynx Shunt as battery monitor. All JK data seems to be available in Venus OS.

Venus OS 2.93
Serialbattery driver beta 20230508 (3x)

Aggregate battery settings:
# Version 2.4

NR_OF_BATTERIES = 3                                     # Nr. of physical batteries to be aggregated
NR_OF_MPPTS = 0                                         # Nr. of MPPTs

BATTERY_KEY_WORD = 'com.victronenergy.battery.tty'      # Key world to identify services of physical Serial Batteries
BATTERY_NAME_KEY_WORD = 'SerialBattery'                 # Key world to identify the name of batteries (to exclude SmartShunt)
BATTERY_NAME_PATH = '/CustomName'                       # What defines the battery name
#BATTERY_NAME_PATH = '/HardwareVersion'

MULTI_KEY_WORD = 'com.victronenergy.vebus.tty'          # Key world to identify service of Multis/Quattros (or cluster of them)
MPPT_KEY_WORD = 'com.victronenergy.solarcharger.tty'    # Key world to identify services of solar chargers (RS not tested, only SmartSolar)
SMARTSHUNT_NAME_KEY_WORD = 'SmartShunt'                 # Key world to identify services of SmartShunt (not tested)
                                                        # specify more precisely if more SmartShunts are in systems

SEARCH_TRIALS = 10                                      # Trials to identify of all batteries before exit and restart
READ_TRIALS = 10                                        # Trials to get consistent data of all batteries before exit and restart

CURRENT_FROM_VICTRON = True                             # If True, the current measurement by Multis/Quattros and MPPTs is taken instead of BMS
DC_LOADS = False                                        # If DC loads with Smart Shunt present, can be used for total current measurement
INVERT_SMARTSHUNT = False                               # False: Current subtracted, True: Current added
OWN_SOC = True                                          # If True, the self calculated charge indicators are taken instead of BMS
CHARGE_SAVE_PRECISION = 0.0025                          # Trade-off between save precision and file access frequency

OWN_CHARGE_PARAMETERS = False                           # Calculate own charge/discharge control parameters (True) from following settings
                                                        # or use them from battery driver (False)
CHARGE_VOLTAGE = 3.45                                   # Constant voltage charge = this value * nr. of cells
MAX_CELL_VOLTAGE = 3.5                                  # If reached by 1-st cell, the charger voltage is clamped to the measured value
DISCHARGE_VOLTAGE = 2.9                                 # If reached, discharge current set to zero
MIN_CELL_VOLTAGE = 2.7                                  # If reached, discharge current set to zero
VOLTAGE_SET_PRECISION = 0.06                            # To be subtracted from the calculated max. charge voltage if MAX_CELL_VOLTAGE is exceeded

MAX_CHARGE_CURRENT = 180                                # Max. charge current at normal conditions
MAX_DISCHARGE_CURRENT = 180                             # Max. discharge current at normal conditions

Error from log file:

ERROR:root:2023-05-18 13:28:29.048572: Error: '>' not supported between instances of 'float' and 'NoneType'. Read trial nr. 1
... Repeats 10 times, then starts over again.

Dr-Gigavolt, 2023-05-18:
Hello, please copy everything since the start of the service from the log file.

gnissahr, 2023-05-19:
Please find attached the log file.

[aggregatebatteries.log](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/files/11515764/aggregatebatteries.log)

Dr-Gigavolt, 2023-05-19:
There is some issue on the Dbus. The driver needs 4 trials to find all batteries. Already this is suspicious. When all batteries found, the data are read and these are not present (Type None). You can check with dbus-spy command if all instances of Serial Battery driver send valid data and not "None". And you can try to increase READ_TRIALS to e.g. 100 and look what happens.

gnissahr, 2023-05-20:
Could it be related to the fact that sometimes my battery id's change?
I have 3 batteries, but most of the time ID 0-4 with one random one in
between missing. Occasionally it is 0-3 with the Lynx shunt at ID 0. It
changes at every reboot of the cerbo. Perhaps even after
reloading/restarting serial battery driver.
So even in VRM I have to change it everytime to make the advanced graphs
work.

I could try to stop serial battery, remove the devices from VRM and then
reboot. Perhaps then all IDs are 0-3.

gnissahr, 2023-05-20:
I made sure all unused IDs were removed, reinstalled and now it seems to work...

closed 2023-05-21 by Dr-Gigavolt