Dr-Gigavolt/dbus-aggregate-batteries #39: dbus-serialbattery v1.0 does not work

stixif, 2023-06-04:
this time it does not work with 
dbus-serialbattery v1.0

Marvo2011, 2023-06-05:
Hello @stixif,

I think you should change the value in the settings.py to '/ProductName' not in the source code.
`BATTERY_NAME_PATH = '/CustomName'                       # What defines the battery name` 

https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/pull/40

stixif, 2023-06-10:
Hi Marvo, i know... but in the source you do not use this value...
check my changes...

Marvo2011, 2023-06-10:
Hey @stixif,

ah okay, now I understand. I will try dbus-serialbattery v1.0 and will have a look about the missing dbus '/ProductName'.

pit65535, 2023-06-11:
I have exactly th same PRoblem and i don´t understand how to solve it, can you plese help me:
INFO:root:2023-06-11 16:27:47.193831: Starting AggregateBatteries.
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:2023-06-11 16:27:47.307538: Initial Ah read from file: 340Ah
INFO:root:2023-06-11 16:27:47.442479: Starting battery monitor.
INFO:root:2023-06-11 16:27:47.444878: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyACM0 has device instance 1
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyS4, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyS4 has device instance 276
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyACM1 has device instance 2
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.socketcan_can0_vi0_uc229774, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.socketcan_can0_vi0_uc229774 has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:2023-06-11 16:27:52.445687: Searching batteries: Trial Nr. 1
INFO:root:2023-06-11 16:27:52.449233: 0 batteries found.
INFO:root:2023-06-11 16:27:57.446693: Searching batteries: Trial Nr. 2
INFO:root:2023-06-11 16:27:57.450924: 0 batteries found.
INFO:root:2023-06-11 16:28:02.447534: Searching batteries: Trial Nr. 3
INFO:root:2023-06-11 16:28:02.451331: 0 batteries found.
INFO:root:2023-06-11 16:28:07.448134: Searching batteries: Trial Nr. 4
INFO:root:2023-06-11 16:28:07.451892: 0 batteries found.
INFO:root:2023-06-11 16:28:12.448720: Searching batteries: Trial Nr. 5
INFO:root:2023-06-11 16:28:12.453803: 0 batteries found.
INFO:root:2023-06-11 16:28:17.449740: Searching batteries: Trial Nr. 6
INFO:root:2023-06-11 16:28:17.453409: 0 batteries found.
INFO:root:2023-06-11 16:28:22.450018: Searching batteries: Trial Nr. 7
INFO:root:2023-06-11 16:28:22.454556: 0 batteries found.
INFO:root:2023-06-11 16:28:27.450420: Searching batteries: Trial Nr. 8
INFO:root:2023-06-11 16:28:27.454315: 0 batteries found.
INFO:root:2023-06-11 16:28:32.451257: Searching batteries: Trial Nr. 9
INFO:root:2023-06-11 16:28:32.456233: 0 batteries found.
INFO:root:2023-06-11 16:28:37.452156: Searching batteries: Trial Nr. 10
INFO:root:2023-06-11 16:28:37.456824: 0 batteries found.
INFO:root:2023-06-11 16:28:42.452665: Searching batteries: Trial Nr. 11
INFO:root:2023-06-11 16:28:42.457199: 0 batteries found.
ERROR:root:2023-06-11 16:28:42.460423: Required number of batteries not found. Exiting.

Marvo2011, 2023-06-11:
@pit65535 look at @stixif patch, only 2 lines to edit, very simple. I think dbus-serialbattery has remove /ProductName form the dbus. But we have checked it for now. We will take a look as soon as possible.

pit65535, 2023-06-11:
Hi Marvo, looks i´m really too stupid :-(, I tried changing in settings.py:
BATTERY_NAME_PATH = '/CustomName'                       # What defines the battery name
in BATTERY_NAME_PATH = '/ProductName'                       # What defines the battery name

but still the same......

can yo just tell me the two lines I should edit in what file?

Marvo2011, 2023-06-11:
@pit65535 Please look here: https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/pull/40/files and revert settings.py to original /CustomName

pit65535, 2023-06-11:
@Marvo2011 
Ahhh thank you very much! 
This helped, now everthing works again :-)

Dr-Gigavolt, 2023-06-26:
Please try the 3.0, I hope it works now.

closed 2023-06-26 by Dr-Gigavolt