Dr-Gigavolt/dbus-aggregate-batteries #44: versioning

ramack, 2023-06-24:
Readme mentions version 2.4 as current one and 3.0 to come soon. But actually there is no version tag at all in the repo. Would you mind to create releases with tags as soon as you consider a version as stable? Otherwise we get some development version from git, see some version number in the code (and GUI) but it is not necessarily the released version.

Thanks for all your efforts! it's a great piece of work!

Dr-Gigavolt, 2023-06-24:
Hello,, thanks for the tip, but I have no idea about programming conventions. I'm an hardware engineer, more analog than digital, and I learned a bit programming to control my hardware and instruments. I push into git only the stuff I believe it works. If there is some tutorial I can have a look, but at the moment I have no time for perfection.

ramack, 2023-06-24:
maybe the github manual helps: https://docs.github.com/en/repositories/releasing-projects-on-github/managing-releases-in-a-repository but it's really not hard if you can do hardware design you will manage to tag and release here on github :grinning:

For sure it is good to push only code you're confident with, but also putting changes online early can help you by review and testing from others

Dr-Gigavolt, 2023-06-26:
You're right, it was easy. I made it for 3.0, hopefully correctly. Is it possible to make it for the older stuff as well?

closed 2023-06-26 by Dr-Gigavolt

ramack, 2023-06-26:
It is possible to create a tag also on older commits (but maybe not in the github webinterface) but I am not sure if this is worth the effort. It's already very cool if you do it from now on :grinning:
Another great thing would be if you'd commit more often... A big changeset as between v2.x and v3.0 is really hard to walk through and get a rough understanding of the changes. Smaller commits with good commit messages would help a lot here, and you could tag a version v3.00beta and ask for testing before you tag the release, as there might be some users willing to do beta testing and others who prefer an already tested version as release.