Dr-Gigavolt/dbus-aggregate-batteries #45: Batteries not found

coca7, 2023-06-27:
The log reports that the batteries are not found:

INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
INFO:dbusmonitor:	com.victronenergy.settings has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyS5, scanning and storing items
INFO:dbusmonitor:	com.victronenergy.solarcharger.ttyS5 has device instance 278
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyS6, scanning and storing items
INFO:dbusmonitor:	com.victronenergy.solarcharger.ttyS6 has device instance 279
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyS4, scanning and storing items
INFO:dbusmonitor:	com.victronenergy.vebus.ttyS4 has device instance 276
INFO:root:Tue Jun 27 13:51:07 2023: Searching Settings: Trial Nr. 1
INFO:root:Tue Jun 27 13:51:07 2023: com.victronenergy.settings found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyS7, scanning and storing items
INFO:dbusmonitor:	com.victronenergy.battery.ttyS7 has device instance 277
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:	com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
INFO:dbusmonitor:	com.victronenergy.battery.ttyUSB0 has device instance 1
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
INFO:dbusmonitor:	com.victronenergy.battery.ttyUSB1 has device instance 2
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:Tue Jun 27 13:51:12 2023: Searching batteries: Trial Nr. 1
INFO:root:Tue Jun 27 13:51:12 2023: 0 batteries found.
INFO:root:Tue Jun 27 13:51:17 2023: Searching batteries: Trial Nr. 2
INFO:root:Tue Jun 27 13:51:17 2023: 0 batteries found.
INFO:root:Tue Jun 27 13:51:22 2023: Searching batteries: Trial Nr. 3
INFO:root:Tue Jun 27 13:51:22 2023: 0 batteries found.
INFO:root:Tue Jun 27 13:51:27 2023: Searching batteries: Trial Nr. 4
INFO:root:Tue Jun 27 13:51:27 2023: 0 batteries found.
INFO:root:Tue Jun 27 13:51:32 2023: Searching batteries: Trial Nr. 5
INFO:root:Tue Jun 27 13:51:32 2023: 0 batteries found.
INFO:root:Tue Jun 27 13:51:37 2023: Searching batteries: Trial Nr. 6
INFO:root:Tue Jun 27 13:51:37 2023: 0 batteries found.
INFO:root:Tue Jun 27 13:51:42 2023: Searching batteries: Trial Nr. 7
INFO:root:Tue Jun 27 13:51:42 2023: 0 batteries found.
INFO:root:Tue Jun 27 13:51:47 2023: Searching batteries: Trial Nr. 8
INFO:root:Tue Jun 27 13:51:47 2023: 0 batteries found.
INFO:root:Tue Jun 27 13:51:52 2023: Searching batteries: Trial Nr. 9
INFO:root:Tue Jun 27 13:51:52 2023: 0 batteries found.
INFO:root:Tue Jun 27 13:51:57 2023: Searching batteries: Trial Nr. 10
INFO:root:Tue Jun 27 13:51:57 2023: 0 batteries found.
INFO:root:Tue Jun 27 13:52:02 2023: Searching batteries: Trial Nr. 11
INFO:root:Tue Jun 27 13:52:02 2023: 0 batteries found.
ERROR:root:Tue Jun 27 13:52:02 2023: Required number of batteries not found. Exiting.

Dr-Gigavolt, 2023-06-27:
Hello, it is not enough info. Please write down the versions of the Serial Battery and of my tool and attach your settings.py

coca7, 2023-06-27:
serial battery: 1.0.20230613dev
Version 3-beta of your tool
My system setup is:
1- VictronMultiplus
2-Victron MPPT
3- 230ah batteries with JBD Bms
1-Victron Smart Shunt

Now, that I'm thinking of it my Venus OS is 3.0.  I think I remember you having a discussion with someone about this version changing communication speed and causing a problem.

root@einstein:/data/dbus-aggregate-batteries# nano settings.py 

# Version 3.0

#######################################
########## Hardware settings ##########
#######################################

NR_OF_BATTERIES = 3                                     # Nr. of physical batteries to be aggregated
NR_OF_CELLS_PER_BATTERY = 4
NR_OF_MPPTS = 2                                         # Nr. of MPPTs
DC_LOADS = True                                        # If DC loads with Smart Shunt present, can be used for total current measurement
INVERT_SMARTSHUNT = False                               # False: Current subtracted, True: Current added

#######################################
############ DBus settings ############
#######################################

BATTERY_SERVICE_NAME = 'com.victronenergy.battery'	# Key world to identify services of physical Serial Batteries (and SmartShunt if available)
BATTERY_PRODUCT_NAME_PATH = '/Battery'              # Path of Battery Product Name
BATTERY_PRODUCT_NAME = 'SerialBattery'                  # Key world to identify the batteries (to exclude SmartShunt)
BATTERY_INSTANCE_NAME_PATH = '/CustomName'              # if CustomName doesn't exist, set '/ProductName'
MULTI_KEY_WORD = 'com.victronenergy.vebus'              # Key world to identify service of Multis/Quattros (or cluster of them)
MPPT_KEY_WORD = 'com.victronenergy.solarcharger'        # Key world to identify services of solar chargers
SMARTSHUNT_NAME_KEY_WORD = 'SmartShunt'                 # Key world to identify services of SmartShunt

SEARCH_TRIALS = 10                                      # Trials to identify of all batteries before exit and restart
READ_TRIALS = 10                                        # Trials to get consistent data of all batteries before exit and restart

#######################################
############ DBus settings ############
#######################################

CURRENT_FROM_VICTRON = True                             # If True, the current measurement by Multis/Quattros and MPPTs is taken instead of BMS
OWN_SOC = True                                          # If True, the self calculated charge indicators are taken instead of BMS
CHARGE_SAVE_PRECISION = 0.0025                          # Trade-off between save precision and file access frequency

#######################################
##### Charge/Discharge parameters #####
#######################################

# Please note: Victron EES disables CCL if DC-coupled PV feed-in is active

OWN_CHARGE_PARAMETERS = True                            # Calculate own charge/discharge control parameters (True) from following settings
                                                        # or use them from battery driver (False)
CHARGE_VOLTAGE = 2.3                                    # Constant voltage charge = this value * nr. of cells
BALANCING_VOLTAGE = 2.45                                # This voltage per cell will be set periodically and kept until balancing below CELL_DIFF_MAX and next charge cycle
BALANCING_REPETITION = 10                               # in days

MAX_CELL_VOLTAGE = 2.5                                  # If reached by 1-st cell, the CVL is dynamically limited. DC-coupled PV feed-in will be disabled to enable the CCL.
DISCHARGE_VOLTAGE = 2.0                                 # If reached, discharge current set to zero
MIN_CELL_VOLTAGE = 1.9                                  # If reached, discharge current set to zero
CELL_DIFF_MAX = 0.025                                   # If lower: re-enable DC-coupled PV feed in (if was enabled before); go back from BALANCING_VOLTAGE to CHARGE_VOLTAGE

MAX_CHARGE_CURRENT = 300                                # Max. charge current at normal conditions
MAX_DISCHARGE_CURRENT = 200                             # Max. discharge current at normal conditions

# settings limiting charge and discharge current if at least one cell gets full or empty
# the lists may have any length, but the same length for voltage and current
# linear interpolation is used for values between
CELL_FULL_LIMITING_VOLTAGE = [CHARGE_VOLTAGE, BALANCING_VOLTAGE, MAX_CELL_VOLTAGE]           # [min, .... ,max]
CELL_FULL_LIMITED_CURRENT =  [1, 0.05, 0]
CELL_EMPTY_LIMITING_VOLTAGE = [DISCHARGE_VOLTAGE - 0.1, DISCHARGE_VOLTAGE, MIN_CELL_VOLTAGE]    # [min, .... ,max]
CELL_EMPTY_LIMITED_CURRENT =  [0, 0.05, 1]

#######################################
####### Logging and reporting #########
#######################################

SEND_CELL_VOLTAGES = 0                                 # 0: Disable Cell Info in dbus, 1: Format: /Cell/BatteryName_Cell<ID>
LOGGING = 2                                            # 0: no logging, 1: print to console, 2: print to file
LOG_PERIOD = 600                                       # in seconds; if 0, periodic logging is disabled

Dr-Gigavolt, 2023-06-27:
The new Serial Battery has some changes of the Dbus protocol, perhaps it is an identification issue, it could be fixed in settings.py. Please read the transmission of one of your batteries by dbus-spy shell command and share it. I use an old Serial Battery, the new one is not stable in my system. And I see a suspitious combination in your setup: 4 LTO cells in series? Such a battery would be full at less than 10V. Or you have LFP and forgot to set up the cell voltages?

coca7, 2023-06-27:
My batteries are LFP prismatic cells I didn't realize that I needed to setup the cell voltages.

[image]

Dr-Gigavolt, 2023-06-27:
Please navigate in the dbus-spy with arrow keys into one battery to see the details.

Dr-Gigavolt, 2023-06-27:
About voltages ... I use LTOs, therefore such low values. But I have 22S x 5P in each bank.

coca7, 2023-06-27:
[image]

Dr-Gigavolt, 2023-06-27:
There is much more info, please scroll down and screenshot all.

coca7, 2023-06-27:
[image]
[image]
[image]

Dr-Gigavolt, 2023-06-28:
Hi, your dBus setting is not correct. Please read the comments in settings.py first and review your dbus-spy output to understand it. Your '\ProductName' is Battery2, not SerialBattery. The another batteries must be named on the same way, e.g. Battery1 etc. And the new Serial Battery driver has no path '\CustomName'. You sholud set up:

BATTERY_PRODUCT_NAME_PATH = '/ProductName' 
BATTERY_PRODUCT_NAME = 'Battery'
BATTERY_INSTANCE_NAME_PATH = '/ProductName'

Dr-Gigavolt, 2023-07-08:
Is it working now?

coca7, 2023-07-08:
I need my laptop to change the battery name, and haven't been able to.  I have time today, so I will fix the battery name.

coca7, 2023-07-08:
I remained the one battery that was incorrectly named.  So now my batteries are named on the BMS: Battery1, Battery2, and Battery3

I changed settings.py to reflect:
BATTERY_PRODUCT_NAME_PATH = '/ProductName'
BATTERY_PRODUCT_NAME = 'Battery'
BATTERY_INSTANCE_NAME_PATH = '/ProductName'

Aggregate batteries does not show up in the GUI

There is no new log file 

I looked at bus-spy again, and there is something there that was not before.  Namely, there is now a CustomName: SerialBattery(LLT/JBD)

Shouldn't I then change: BATTERY_INSTANCE_NAME_PATH = '/ProductName' 

 BATTERY_INSTANCE_NAME_PATH = '/ SerialBattery(LLT/JBD) 

![Screenshot 2023-07-08 at 8 54 45 AM](https:
[image]
//github.com/Dr-Gigavolt/dbus-aggregate-batteries/assets/2902641/f78f1d81-dd57-4005-8729-cccf1367e29f)

coca7, 2023-07-08:
[image]

Dr-Gigavolt, 2023-07-08:
The BATTERY_INSTANCE_NAME_PATH  must be either '/CustomName' or '/ProductName', the one which carries different names of your batteries. Can you set CustomName for each  battery? That was the way to name then in older Serial Battery versions.

If there is nothing in GUI nor in the log file it means the program either does not start or crashes immediatelly after start.

coca7, 2023-07-08:
I don't think you can change the CustomName in Serial Battery.  I didn't change any of the settings there I just noted that Custom name appeared when I ran dbus-spy

What I can do is name the batteries on the individual BMS For example here is one of my batteries:
[image]

Dr-Gigavolt, 2023-07-08:
You can change CustomName in the GUI of the SerialBattery driver. Please try it and change the BATTERY_INSTANCE_NAME_PATH back to /CustomName. This is my configuration which works without any troubles.

coca7, 2023-07-08:
So you mean to change Name in the GUI from "SerialBattery(LLT/JBD) to something else like "Battery"

[image]

Dr-Gigavolt, 2023-07-08:
Yes, you can set the same name as Product above.

kompenhans, 2023-08-02:
modifying entry in /data/dbus-aggregate-batteries/settings.py
#From:
BATTERY_PRODUCT_NAME = 'SerialBattery'  # Key world to identify the batteries
#To:
BATTERY_PRODUCT_NAME = 'JBD-AP20S006-L20S-200A-B'   # Key world to identify the batteries 

worked for me to unter Victron 3.01 with latest serial battery

Zahnpasta83, 2023-08-24:
> modifying entry in /data/dbus-aggregate-batteries/settings.py #From: BATTERY_PRODUCT_NAME = 'SerialBattery' # Key world to identify the batteries #To: BATTERY_PRODUCT_NAME = [...]

You saved my day. I updated to new Serial Battery version and was desperate because aggregate batteries didn't find my batteries anymore. Now it works again with your help

Dr-Gigavolt, 2023-08-24:
Hello guys,
for this reason the path is parametrized and explained in the readme.