Dr-Gigavolt/dbus-aggregate-batteries #46: No Data from Aggregate Batteries

peregrines2, 2023-07-29:
Yesterday I installed my 4th battery bank (JK BMS EVE280 Serialbattery). I adapted the settings in settings.py accordingly. 
I get the following error :( 
Using Venus V3.10 in Rpi4 8GB.

[image]

`INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB3 has device instance 7
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB4, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB4 has device instance 5
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB5, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB5 has device instance 6
INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyUSB0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyUSB0 has device instance 290
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB1 has device instance 289
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:Sat Jul 29 05:07:09 2023: Searching batteries: Trial Nr. 5
INFO:root:Sat Jul 29 05:07:09 2023: SerialBattery(Jkbms) found, named as: SerialBattery(Jkbms).
INFO:root:Sat Jul 29 05:07:09 2023: SerialBattery(Jkbms) found, named as: SerialBattery(Jkbms)2.
INFO:root:Sat Jul 29 05:07:09 2023: SerialBattery(Jkbms) found, named as: SerialBattery(Jkbms)3.
INFO:root:Sat Jul 29 05:07:09 2023: SerialBattery(Jkbms) found, named as: SerialBattery(Jkbms)4.
INFO:root:Sat Jul 29 05:07:09 2023: 4 batteries found.
ERROR:root:Sat Jul 29 05:07:10 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 1
ERROR:root:Sat Jul 29 05:07:11 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 2
ERROR:root:Sat Jul 29 05:07:12 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 3
ERROR:root:Sat Jul 29 05:07:13 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 4
ERROR:root:Sat Jul 29 05:07:14 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 5
ERROR:root:Sat Jul 29 05:07:15 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 6
ERROR:root:Sat Jul 29 05:07:16 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 7
ERROR:root:Sat Jul 29 05:07:17 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 8
ERROR:root:Sat Jul 29 05:07:18 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 9
ERROR:root:Sat Jul 29 05:07:19 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 10
ERROR:root:Sat Jul 29 05:07:20 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 11
ERROR:root:Sat Jul 29 05:07:21 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 12
ERROR:root:Sat Jul 29 05:07:22 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 13
ERROR:root:Sat Jul 29 05:07:23 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 14
ERROR:root:Sat Jul 29 05:07:24 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 15
ERROR:root:Sat Jul 29 05:07:25 2023: Error: '>' not supported between instances of 'NoneType' and 'float'. Read trial nr. 16
ERROR:root:Sat Jul 29 05:07:25 2023: DBus read failed. Exiting.
INFO:root:Sat Jul 29 05:07:26 2023: Starting AggregateBatteries.
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:Sat Jul 29 05:07:26 2023: Initial Ah read from file: 894Ah
INFO:root:Sat Jul 29 05:07:26 2023: Starting battery monitor.
INFO:root:Sat Jul 29 05:07:26 2023: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 9
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB3, scanning and storing items
INFO:root:Sat Jul 29 05:07:27 2023: Searching Settings: Trial Nr. 1
INFO:root:Sat Jul 29 05:07:27 2023: com.victronenergy.settings found.`

Dr-Gigavolt, 2023-07-31:
Hello, the Venus 3.10 is beta. I use 3.00 and everything works. Please try 3.00 or the last release 3.01. The log says that some Dbus path is invalid. When I'll do the next update of my code I'll add an exact messaging which path fails. But have no time at the moment, I'm sorry.

Dr-Gigavolt, 2023-10-25:
I improved logging, it will be available with soon available with the next update.

closed 2023-10-25 by Dr-Gigavolt