Dr-Gigavolt/dbus-aggregate-batteries #47: Required number of batteries not found. Exiting

kompenhans, 2023-07-31:
hi 

after new installation of Victron OS 3.10-21 in combination of dbus-aggregate-batteries 3.0
I get follwoing repeating messages:

INFO:root:Mon Jul 31 14:37:15 2023: Starting AggregateBatteries.                            
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate              
INFO:root:Mon Jul 31 14:37:15 2023: Initial Ah read from file: 317Ah                        
INFO:root:Mon Jul 31 14:37:15 2023: Last balancing done at the 175. day of the year         
INFO:root:Mon Jul 31 14:37:15 2023: Starting battery monitor.                               
INFO:root:Mon Jul 31 14:37:15 2023: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items         
INFO:dbusmonitor:       com.victronenergy.settings has device instance 0               
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM0, scanning and storing items  
INFO:dbusmonitor:       com.victronenergy.battery.ttyACM0 has device instance 2        
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyACM1, scanning and storing items  
INFO:dbusmonitor:       com.victronenergy.battery.ttyACM1 has device instance 1        
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyUSB0, scanning and storing items    
INFO:dbusmonitor:       com.victronenergy.vebus.ttyUSB0 has device instance 288        
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0     
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:Mon Jul 31 14:37:16 2023: Searching Settings: Trial Nr. 1  
INFO:root:Mon Jul 31 14:37:16 2023: com.victronenergy.settings found.
INFO:root:Mon Jul 31 14:37:21 2023: Searching batteries: Trial Nr. 1
INFO:root:Mon Jul 31 14:37:21 2023: 0 batteries found.              
INFO:root:Mon Jul 31 14:37:26 2023: Searching batteries: Trial Nr. 2                 
INFO:root:Mon Jul 31 14:37:26 2023: 0 batteries found.                         
INFO:root:Mon Jul 31 14:37:31 2023: Searching batteries: Trial Nr. 3                 
INFO:root:Mon Jul 31 14:37:31 2023: 0 batteries found.                         
INFO:root:Mon Jul 31 14:37:36 2023: Searching batteries: Trial Nr. 4
INFO:root:Mon Jul 31 14:37:36 2023: 0 batteries found.              
INFO:root:Mon Jul 31 14:37:41 2023: Searching batteries: Trial Nr. 5
INFO:root:Mon Jul 31 14:37:41 2023: 0 batteries found.              
INFO:root:Mon Jul 31 14:37:46 2023: Searching batteries: Trial Nr. 6
INFO:root:Mon Jul 31 14:37:46 2023: 0 batteries found.              
INFO:root:Mon Jul 31 14:37:51 2023: Searching batteries: Trial Nr. 7 
INFO:root:Mon Jul 31 14:37:51 2023: 0 batteries found.              
INFO:root:Mon Jul 31 14:37:56 2023: Searching batteries: Trial Nr. 8 
INFO:root:Mon Jul 31 14:37:56 2023: 0 batteries found.              
INFO:root:Mon Jul 31 14:38:01 2023: Searching batteries: Trial Nr. 9                 
INFO:root:Mon Jul 31 14:38:01 2023: 0 batteries found.          

The 2 batteries are visible and working as before, but no longer aggregated.
related aggregate GUI secion shows:
connected: yes
type of connection: virtual
product: Aggregatebatteries
Preduct identifier: 0x0
Firmware version: 3.0
Hardware version:  0.0

used batteries: JBD-AP20S006L20S-200A-B

kompenhans, 2023-07-31:
Thx a lot, will downgrade to 3.0

Dr-Gigavolt, 2023-10-25:
Logging improved and will be soon available with the next update.

closed 2023-10-25 by Dr-Gigavolt