Dr-Gigavolt/dbus-aggregate-batteries #48: Freeze when SoC is 0%?

rperpe, 2023-08-26:
Hello

I tried out dbus-aggregate-batteries and it looks like it freezed in this state, values do not change anymore:
[image]

I suspect that it did not survive the 0% SoC value of the single BMS which is connected to this setup.

Is that possible? Otherwise, are there ways to get to the root cause?

Many thanks!

KoljaWindeler, 2023-10-19:
Interesting.. same has happened to me. 
Thanks for your awesome work!

rperpe, 2023-10-19:
Did it happen at your side at 1% SoC too?

KoljaWindeler, 2023-10-19:
Nope, actually at 42% of aggregate battery but both BMS fell from 42 to 0 (I previously reset the capacity while charging, so it was 42% too high. But I was aware of the situation and watched it closely) .. 

Interestingly my Multiplus refused to charge after that even. I had to reboot my Venus.

Dr-Gigavolt, 2023-10-21:
Hello guys, thanks for reporting. I had never zero SoC due to setting ESS minimum greater than 0. My program cannot cause the issue because I never use SoC as an input. CVL, CCL and DCL are calculated of cell voltages (or imported from Serial Batteries). I supose ESS has some problem wenn seeing zero SoC on DBus. You can set the ESS minimum SoC to 5% as workaround.

Dr-Gigavolt, 2023-10-21:
And wenn my JK BMSes run down to zero due to poor current measurement it does not freeze, I have never observed it.

Dr-Gigavolt, 2023-10-21:
> Nope, actually at 42% of aggregate battery but both BMS fell from 42 to 0 (I previously reset the capacity while charging, so it was 42% too high. But I was aware of the situation and watched it closely) ..
> 
> Interestingly my Multiplus refused to charge after that even. I had to reboot my Venus.

I observe similar issue, AC coupled charging is disabled during ESS #1 and #2. I don't know why. One should review the ESS code, O have no time now.

regu-tech, 2024-11-03:
Hello, 

i had the same problem today, caused by the weather all four battery blocks goes to 0% and the aggregate
driver freezes. I know i can enter a minimum SOC with 10% but i want to cycle the batteries 2-3 times to 0%.

[image]

Actually the batteries are charged a little bit but the aggregate driver still shows zero.

[image]

This is what the log says:

`INFO:root:  Bat. voltage: 49.6V, Bat. current: -21A, SoC: 4.1%, Balancing state: 0
INFO:root:  Min. cell voltage: SerialBattery(LLT/JBD)2_C15: 3.043V, Max. cell voltage: SerialBattery(LLT/JBD)3_C12: 3.145V, difference: 0.102V
INFO:root:Sun Nov  3 02:59:34 2024: Repetitive logging:
INFO:root:  CVL: 55.2V, CCL: 800A, DCL: 630A
INFO:root:  Bat. voltage: 49.2V, Bat. current: -23A, SoC: 4.1%, Balancing state: 0
INFO:root:  Min. cell voltage: SerialBattery(LLT/JBD)2_C15: 3.014V, Max. cell voltage: SerialBattery(LLT/JBD)3_C12: 3.128V, difference: 0.114V
INFO:root:Sun Nov  3 03:14:36 2024: Repetitive logging:
INFO:root:  CVL: 55.2V, CCL: 800A, DCL: 576A
INFO:root:  Bat. voltage: 48.9V, Bat. current: -20A, SoC: 3.5%, Balancing state: 0
INFO:root:  Min. cell voltage: SerialBattery(LLT/JBD)2_C15: 2.986V, Max. cell voltage: SerialBattery(LLT/JBD)3_C12: 3.112V, difference: 0.126V
INFO:root:Sun Nov  3 03:29:38 2024: Repetitive logging:
INFO:root:  CVL: 55.2V, CCL: 800A, DCL: 510A
INFO:root:  Bat. voltage: 48.5V, Bat. current: -20A, SoC: 2.9%, Balancing state: 0
INFO:root:  Min. cell voltage: SerialBattery(LLT/JBD)2_C15: 2.952V, Max. cell voltage: SerialBattery(LLT/JBD)3_C12: 3.094V, difference: 0.142V
INFO:root:Sun Nov  3 03:44:40 2024: Repetitive logging:
INFO:root:  CVL: 55.2V, CCL: 800A, DCL: 424A
INFO:root:  Bat. voltage: 48.0V, Bat. current: -21A, SoC: 2.3%, Balancing state: 0
INFO:root:  Min. cell voltage: SerialBattery(LLT/JBD)2_C15: 2.908V, Max. cell voltage: SerialBattery(LLT/JBD)3_C12: 3.069V, difference: 0.161V
INFO:root:Sun Nov  3 03:59:42 2024: Repetitive logging:
INFO:root:  CVL: 55.2V, CCL: 800A, DCL: 262A
INFO:root:  Bat. voltage: 47.3V, Bat. current: -26A, SoC: 1.4%, Balancing state: 0
INFO:root:  Min. cell voltage: SerialBattery(LLT/JBD)2_C15: 2.851V, Max. cell voltage: SerialBattery(LLT/JBD)3_C12: 3.041V, difference: 0.190V
Traceback (most recent call last):
  File "/data/dbus-aggregate-batteries/aggregatebatteries.py", line 1130, in _update
    Soc = Soc / Capacity  # weighted sum
ZeroDivisionError: float division by zero`

Dr-Gigavolt, 2024-11-07:
Hello, now I've found it. I had never the failure because I use my own control (_settings.OWN_SOC = True_) and the lines 1130 ... 1132 are not executed. The mistake was the division by _Capacity_ instead of _InstalledCapacity_ in that piece of code. New commit comes ASAP.

closed 2024-11-07 by Dr-Gigavolt

regu-tech, 2024-11-07:
Hello, 

we are one step ahead, it seems that the aggregate battery is shown correctly in the overview window:
[image]

but the "Aggregate Batteries" menu point is gone when all batteries are empty
[image]

The Log shows many of these errors:

`IndentationError: unindent does not match any outer indentation level
  File "/data/dbus-aggregate-batteries/aggregatebatteries.py", line 832
    sys.exit()
             ^
IndentationError: unindent does not match any outer indentation level
  File "/data/dbus-aggregate-batteries/aggregatebatteries.py", line 832
    sys.exit()
             ^
IndentationError: unindent does not match any outer indentation level
  File "/data/dbus-aggregate-batteries/aggregatebatteries.py", line 832
    sys.exit()
`