Dr-Gigavolt/dbus-aggregate-batteries #50: Aggregate Batteries deactivates DC coupled PV feed in

Zahnpasta83, 2023-09-13:


Dr-Gigavolt, 2023-09-13:
Hello, it is a protection of the cells, is activated only in case of danger, see the readme. I hope it is clear enough, if not, please ask what exactly you don't understand.

Zahnpasta83, 2023-09-13:
I want to feed in DC produced energy. But it always gets deactivated. What has that to do with the battery. As soon as the voltage is at its max load (18S 62,1V) I want to feed in the energy

Zahnpasta83, 2023-09-13:
And it worked since weeks but now I might have changed something and it doesn't work anymore

Dr-Gigavolt, 2023-09-13:
I see now your settings, they are not correct. All voltages are the same. You should set up as follows:
Charge voltage: 80-90% SoC, to prolong battery live
Balancing voltage 100% SoC
Max. cell voltage: Balancing voltage + some margin, e.g. 30...50mV

If at least one cells exceeds Max. cell voltage (you are about losing the control about the cell voltage), DC feed MUST be de-activated. If the DC feed is active, Victron de-activates the CCL.

Dr-Gigavolt, 2023-09-13:
I begun to write my code last winter, I had just few 100W from the DC coupled PV and some kW from AC-coupled Fronius, so I didn't see the problem, even the cells were new and not well balanced. Few months later, if one cell was full and the voltage begun to peak, the dynamic CVL begun terribly unstable (caused BMS cut-off) because the CCL could not be reduced with active DC-feed. This is a Victron ESS feature, is described in the ESS manual.

Zahnpasta83, 2023-09-13:
now it works, thank you very much. However, I have another problem that you might be able to help me with and I've had it for a while. As soon as the balance voltage is switched to the charge voltage, the Victrons discharge the batteries to 61.2V at full power. Is there any way to prevent this from being actively discharged into the net at full power?

Zahnpasta83, 2023-09-13:
Just to clarify, charge voltage is 61,2 (3,4V per cell) balance voltage is 62,1 (3,45V per cell)

Zahnpasta83, 2023-09-16:
It is now the case that as soon as the set voltage is reached and is fed in, the power constantly fluctuates between approx. 300w charging and then 300w discharging again. the battery never goes to rest, is that normal? There is a constant fluctuation between charging and discharging

Zahnpasta83, 2023-09-16:
I might have found the problem but ill have to wait till tomorrow

snow78, 2024-01-29:
adding to that thread. as i observed the same Problem and identified an issue in the aggregatebatteries.py

Line 599 and 600 must be indented to be in the If condition. otherwise the self._DCfeedActive  will be overwritten with false and never reactivated again this is due to the fact that even if self._dynamicCVL is activated the loop might run multiple times into the condition where MaxCellVoltage >= MAX_CELL_VOLTAGE. this then sets self._DCfeedActive to 0/false.
i tested this on my system.

            if MaxCellVoltage >= MAX_CELL_VOLTAGE:                         
                if not self._dynamicCVL:
                    self._dynamicCVL = True
                    logging.info('%s: Dynamic CVL reduction started.'  % (dt.now()).strftime('%c'))
                    
                self._DCfeedActive = self._dbusMon.dbusmon.get_value('com.victronenergy.settings', '/Settings/CGwacs/OvervoltageFeedIn')    # check if DC-feed enabled
                self._dbusMon.dbusmon.set_value('com.victronenergy.settings', '/Settings/CGwacs/OvervoltageFeedIn', 0)                      # disable DC-coupled PV feed-in
                MaxChargeVoltage = min((min(chargeVoltageReduced_list)), ChargeVoltageBattery)                                              # avoid exceeding MAX_CELL_VOLTAGE
                #self._ownCharge = InstalledCapacity

Dr-Gigavolt, 2024-01-29:
Thanks, I've just corrected and commited.

snow78, 2024-01-29:
that was fast :-) thanks

closed 2024-11-07 by Dr-Gigavolt