Dr-Gigavolt/dbus-aggregate-batteries #53: Does this driver work for aggregating only multiple dbus-serialbattery BMS or of any type?

herrep, 2023-10-05:
Hi,

In my setup I have two JK BMS using dbus-serialbattery driver and a CAN-BUS BMS for which I do not need dbus-serialbattery as it is directly recognized in Cerbo GX. Does dbus-aggregate-batteries support aggregating any type of BMS or does it only resolve the issue with multiple dbus-serialbattery instances?

Thanks a lot for this clarification.

Best regards,
Peter

Dr-Gigavolt, 2023-10-05:
Hi Peter, my tool aggregates the battery data available on the dBus. If your GX device publishes the data of the CAN battery, it could work. It should be the case because the ESS communicates with BMS via dBus. You can check it with dbus-spy command.

herrep, 2023-10-10:
Thanks a lot. As you may see from #54 and #55, I am already working on the integration of the CAN-Bus BMS via dbus-aggregate-batteries.

closed 2024-11-07 by Dr-Gigavolt