Dr-Gigavolt/dbus-aggregate-batteries #55: CAN-SMARTBMS-BAT (Aggregation of multiple SEPLOS BMS) has dbus variables that do not match with dbus-aggregate-batteries

herrep, 2023-10-10:
Hi,

My use case concerns two batteries, each equipped with a SEPLOS BMS. These two BMS interact with each other via RS485. One of the two BMS acts as a master and provides BMS CAN-Bus communication with Cerbo GX, visible under product name CAN-SMARTBMS-BAT. The upper protocol level via BMS CAN-Bus provides already an aggregated view on the two batteries. This is the way as defined by the vendor how the battery system that may consist even of more than two battery backs is to be integrated with Victron.

As I have further batteries installed, each equipped with a JK BMS and integrated with Cerbo GX via dbus-serial-battery, I tried to integrate all the BMS, i.e. the 2 JK BMS via RS485/USB and the SEPLOS BMS via BMS CAN-Bus, with dbus-aggregate-batteries.

However, I realized that quite a number of the CAN-SMARTBMS-BAT dbus variables do not match the expected ones retrieved by dbus-aggregate-batteries. The most relevant difference to a BMS driver like dbus-serial-battery is that the CAN-Bus SEPLOS BMS already aggregates values. Here is a listing of all dbus variables of the CAN-SMARTBMS-BAT:

```
com.victronenergy.battery.socketcan_can1
Alarms/CellImbalance                                                                                                            0
Alarms/ChargeBlocked                                                                                                            0
Alarms/DischargeBlocked                                                                                                         0
Alarms/HighChargeCurrent                                                                                                        0
Alarms/HighChargeTemperature                                                                                                    0
Alarms/HighDischargeCurrent                                                                                                     0
Alarms/HighTemperature                                                                                                          0
Alarms/HighVoltage                                                                                                              0
Alarms/InternalFailure                                                                                                          0
Alarms/LowChargeTemperature                                                                                                     0
Alarms/LowTemperature                                                                                                           0
Alarms/LowVoltage                                                                                                               0
Capacity                                                                                                                      560
Connected                                                                                                                       1
CustomName                                                                                                               Bat03/04
Dc/0/Current                                                                                                                   28
Dc/0/Power                                                                                                                   1469
Dc/0/Temperature                                                                                                             28.6
Dc/0/Voltage                                                                                                                52.47
DeviceInstance                                                                                                                512
Diagnostics/Module0/Alarms/CellImbalance                                                                                        -
Diagnostics/Module0/Alarms/HighChargeCurrent                                                                                    -
Diagnostics/Module0/Alarms/HighChargeTemperature                                                                                -
Diagnostics/Module0/Alarms/HighDischargeCurrent                                                                                 -
Diagnostics/Module0/Alarms/HighTemperature                                                                                      -
Diagnostics/Module0/Alarms/HighVoltage                                                                                          -
Diagnostics/Module0/Alarms/InternalFailure                                                                                      -
Diagnostics/Module0/Alarms/LowChargeTemperature                                                                                 -
Diagnostics/Module0/Alarms/LowTemperature                                                                                       -
Diagnostics/Module0/Alarms/LowVoltage                                                                                           -
Diagnostics/Module0/Id                                                                                                          -
Diagnostics/Module1/Alarms/CellImbalance                                                                                        -
Diagnostics/Module1/Alarms/HighChargeCurrent                                                                                    -
Diagnostics/Module1/Alarms/HighChargeTemperature                                                                                -
Diagnostics/Module1/Alarms/HighDischargeCurrent                                                                                 -
Diagnostics/Module1/Alarms/HighTemperature                                                                                      -
Diagnostics/Module1/Alarms/HighVoltage                                                                                          -
Diagnostics/Module1/Alarms/InternalFailure                                                                                      -
Diagnostics/Module1/Alarms/LowChargeTemperature                                                                                 -
Diagnostics/Module1/Alarms/LowTemperature                                                                                       -
Diagnostics/Module1/Alarms/LowVoltage                                                                                           -
Diagnostics/Module1/Id                                                                                                          -
Diagnostics/Module2/Alarms/CellImbalance                                                                                        -
Diagnostics/Module2/Alarms/HighChargeCurrent                                                                                    -
Diagnostics/Module2/Alarms/HighChargeTemperature                                                                                -
Diagnostics/Module2/Alarms/HighDischargeCurrent                                                                                 -
Diagnostics/Module2/Alarms/HighTemperature                                                                                      -
Diagnostics/Module2/Alarms/HighVoltage                                                                                          -
Diagnostics/Module2/Alarms/InternalFailure                                                                                      -
Diagnostics/Module2/Alarms/LowChargeTemperature                                                                                 -
Diagnostics/Module2/Alarms/LowTemperature                                                                                       -
Diagnostics/Module2/Alarms/LowVoltage                                                                                           -
Diagnostics/Module2/Id                                                                                                          -
FirmwareVersion                                                                                                              4100
Info/BatteryLowVoltage                                                                                                       48.8
Info/ChargeRequest                                                                                                              -
[... cut ...]
System/NrOfModulesOffline                                                                                                       0
System/NrOfModulesOnline                                                                                                        2
```

The purpose of this issue is to document the integration of this CAN-SMARTBMS-BAT into dbus-aggregate-batteries and to find out if others have already implemented something.  As soon as time permits, I will adapt aggregatebatteries.py and provide updated information.

Best regards,
Peter

Dr-Gigavolt, 2023-12-25:
Hi Peter, I have the same problem with lack of time. Therefore I cannot imagine to maintain a tool for aggregating different battery systems. I suggest you make a fork, modify it as you need and publish it. Thanks.

MNEWI, 2024-03-03:
Hi Peter, were you able to bring some light in the issue ?
I am a "newbee" and also faced with the same problem with JKMS on serial and a BYD LVS System on Can-Bus.

Cheers,
Michael

closed 2024-11-07 by Dr-Gigavolt