Dr-Gigavolt/dbus-aggregate-batteries #56: AggregateBatteries in Device List but not listed as BMS in DVCC Menu Controlling BMS

herrep, 2023-10-16:
Hi,

After having configured AggregateBatteries to combine two JK BMS managed by dbus-serial-battery, I was able to see AggregateBatteries in the Cerbo GX Device List. Furthermore, when entering the menu DVCC, I could select AggregateBatteries as Controlling BMS.

Today I had major adjustments in my setup (not related to the batteries and the BMS at all) so that I needed to turn the Victron Multiplus off and on again. During the Multiplus being off, AggregateBatteries reported the following in its log file:

```
ERROR:root:Mon Oct 16 09:51:12 2023: Victron current read error. Using BMS current and power instead.
```

After having finished the adjustments, I realized that AggregateBatteries was still in the Cerbo GX Device List, but it was not longer in the list of BMS that could be selected as Controlling BMS in the DVCC menu.

For this reason, I rebooted the Cerbo GX.  However, after the reboot, AggregateBatteries was still not shown in the list of BMS.

```
INFO:root:Mon Oct 16 10:00:35 2023: CVL: 54.4V, CCL: 480A, DCL: 225A
INFO:root:Mon Oct 16 10:00:35 2023: Bat. voltage: 52.9V, Bat. current: 0A, Balancing state: 0
INFO:root:Mon Oct 16 10:00:35 2023: Min. cell voltage: 3.303V, Max. cell voltage: 3.309V, difference: 0.006V
INFO:root:Mon Oct 16 10:04:04 2023: Starting AggregateBatteries.
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:Mon Oct 16 10:04:04 2023: Initial Ah read from file: 225Ah
INFO:root:Mon Oct 16 10:04:04 2023: Last balancing done at the 286. day of the year
INFO:root:Mon Oct 16 10:04:05 2023: Starting battery monitor.
INFO:root:Mon Oct 16 10:04:05 2023: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.socketcan_can0_vi0_uc211136, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.socketcan_can0_vi0_uc211136 has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyS4, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyS4 has device instance 276
INFO:root:Mon Oct 16 10:04:07 2023: Searching Settings: Trial Nr. 1
INFO:root:Mon Oct 16 10:04:07 2023: com.victronenergy.settings found.
INFO:dbusmonitor:Found: com.victronenergy.battery.socketcan_can1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.socketcan_can1 has device instance 512
INFO:dbusmonitor:Found: com.victronenergy.battery.socketcan_can0_vi0_uc171335, scanning and storing items
INFO:root:Mon Oct 16 10:04:12 2023: Searching batteries: Trial Nr. 1
INFO:dbusmonitor:       com.victronenergy.battery.socketcan_can0_vi0_uc171335 has device instance 0
INFO:root:Mon Oct 16 10:04:12 2023: 0 BMS found.
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.socketcan_can0_vi2_uc270152, scanning and storing items
INFO:root:Mon Oct 16 10:04:17 2023: Searching batteries: Trial Nr. 2
INFO:root:Mon Oct 16 10:04:17 2023: 0 BMS found.
INFO:dbusmonitor:       com.victronenergy.solarcharger.socketcan_can0_vi2_uc270152 has device instance 2
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 1
INFO:root:Mon Oct 16 10:04:23 2023: Searching batteries: Trial Nr. 3
INFO:root:Mon Oct 16 10:04:23 2023: 0 BMS found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2 has device instance 2
INFO:root:Mon Oct 16 10:04:28 2023: Searching batteries: Trial Nr. 4
INFO:root:Mon Oct 16 10:04:28 2023: SerialBattery(Jkbms) found, named as: SerialBattery(Jkbms).
INFO:root:Mon Oct 16 10:04:28 2023: SerialBattery(Jkbms) found, named as: SerialBattery(Jkbms)2.
INFO:root:Mon Oct 16 10:04:28 2023: 2 BMS found.
INFO:root:Mon Oct 16 10:04:29 2023: Searching Multi/Quatro VEbus: Trial Nr. 1
INFO:root:Mon Oct 16 10:04:29 2023: MultiPlus-II 48/5000/70-50 found.
INFO:root:Mon Oct 16 10:04:30 2023: Searching MPPTs: Trial Nr. 1
INFO:root:Mon Oct 16 10:04:30 2023: SmartSolar MPPT RS 450/200 found.
INFO:root:Mon Oct 16 10:04:30 2023: SmartSolar MPPT RS 450/100 found.
INFO:root:Mon Oct 16 10:04:30 2023: 2 MPPT(s) found.
INFO:root:Mon Oct 16 10:14:32 2023: CVL: 54.4V, CCL: 480A, DCL: 225A
INFO:root:Mon Oct 16 10:14:32 2023: Bat. voltage: 52.8V, Bat. current: -1A, Balancing state: 0
INFO:root:Mon Oct 16 10:14:32 2023: Min. cell voltage: 3.301V, Max. cell voltage: 3.305V, difference: 0.004V
INFO:root:Mon Oct 16 10:24:33 2023: CVL: 54.4V, CCL: 480A, DCL: 225A
```

According to my understanding, the log file above looks fine. But AggregateBatteries is still not shown for selection in DVCC / Controlling BMS.

This leads me to two questions:

1) What could I do to overcome this problem?
2) Is there a way to improve AggregateBatteries to avoid this problem at all?

Best regards,
Peter

Dr-Gigavolt, 2023-10-16:
Hi Peter, the log looks fine. You can check with dbus-spy console command if the AggregateBatteries is present on dBus, especially the control parameters CCL, DCL and VCL. If everything looks well, the issue is at the Cerbo side.

herrep, 2023-10-16:
Hi Anton,

Thank you very much for your reply. This is what AggregateBatteries presents on dBus:

```
com.victronenergy.battery.aggregate                           AggregateBatteries
Alarms/CellImbalance                                                           0
Alarms/HighChargeCurrent                                                       0
Alarms/HighChargeTemperature                                                   0
Alarms/HighDischargeCurrent                                                    0
Alarms/HighTemperature                                                         0
Alarms/HighVoltage                                                             0
Alarms/InternalFailure                                                         -
Alarms/LowCellVoltage                                                          0
Alarms/LowChargeTemperature                                                    0
Alarms/LowSoc                                                                  0
Alarms/LowTemperature                                                          0
Alarms/LowVoltage                                                              0
Capacity                                                                     364
Connected                                                                      1
ConsumedAmphours                                                             196
Dc/0/Current                                                               -61.2
Dc/0/Power                                                              -3209.02
Dc/0/Temperature                                                           24.25
Dc/0/Voltage                                                              52.435
DeviceInstance                                                                 0
FirmwareVersion                                                              3.0
HardwareVersion                                                              0.0
Info/MaxChargeCurrent                                                        480
Info/MaxChargeVoltage                                                       54.4
Info/MaxDischargeCurrent                                                     225
InstalledCapacity                                                            560
Io/AllowToBalance                                                              1
Io/AllowToCharge                                                               1
Io/AllowToDischarge                                                            1
Mgmt/Connection                                                          Virtual
Mgmt/ProcessName            /data/dbus-aggregate-batteries/aggregatebatteries.py
Mgmt/ProcessVersion                                                          0.0
ProductId                                                                      0
ProductName                                                   AggregateBatteries
Soc                                                                      65.0154
System/MaxCellTemperature                                                     25
System/MaxCellVoltage                                                      3.279
System/MaxVoltageCellId                                 SerialBattery(Jkbms)2_C1
System/MinCellTemperature                                                     24
System/MinCellVoltage                                                      3.277
System/MinVoltageCellId                                  SerialBattery(Jkbms)_C1
System/NrOfCellsPerBattery                                                    16
System/NrOfModulesBlockingCharge                                               0
System/NrOfModulesBlockingDischarge                                            0
System/NrOfModulesOffline                                                      0
System/NrOfModulesOnline                                                       2
Voltages/Diff                                                              0.001
Voltages/Sum                                                             52.4415
```

Dr-Gigavolt, 2023-10-16:
My program has nothing to do with Lynx shunt, it should even be excluded from the battery list. At least Smartshunt is treated in this way and can be used for measurement of DC consumption, not as battery monitor. Only real BMSes transmitting single cell voltages can be aggregated.

closed 2023-12-25 by Dr-Gigavolt