Dr-Gigavolt/dbus-aggregate-batteries #57: Order of voltages for discharge limitation

KoljaWindeler, 2023-10-24:
Hi, is the order here https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/blob/303e95810c45cadde6e1e6416764c9b41b6f9816/settings.py#L61 correct?

I've just reached a voltage below DISCHARGE_VOLTAGE and the current limitation did NOT kick in. 
After reversing the array it works fine for me .. 

Thanks Kolja

Dr-Gigavolt, 2023-10-24:
Hi Kolja,
the current limiting lists are to be defined by user, but you are right. My example is not correct. Thanks for reporting. You can initially set up as follows and than optimize according your requirements. You can add or remove points, at least two are required.

CELL_EMPTY_LIMITING_VOLTAGE = [MIN_CELL_VOLTAGE, DISCHARGE_VOLTAGE, DISCHARGE_VOLTAGE+0.1]
CELL_EMPTY_LIMITED_CURRENT =  [0, 0.05, 1]

if Voltage <= DISCHARGE_VOLTAGE * NR_OF_CELLS_PER_BATTERY, the SoC counter is set to zero.
The limiting list is applied to the most empty cell. In the example above the discharge current is being limited from 100% down to 5% when the most empty cell is falling from DISCHARGE_VOLTAGE+0.1 down to DISCHARGE_VOLTAGE. Further the current falls down to zero when the most empty cell falls down to MIN_CELL_VOLTAGE which is the absolute minimum.

KoljaWindeler, 2023-10-24:
Thanks, works great. You're awesome

Dr-Gigavolt, 2023-10-24:
> You're awesome
Not really. You, as well as any other user of the tool would be able to write it after investing couple of evenings for reading some examples. Before I started I had no experience with Victron control :-)

Dr-Gigavolt, 2023-10-29:
I made a new commit now. There is no DISCHARGE_VOLTAGE any more, it is not relevant. MIN_CELL_VOLTAGE is what protects the weakest cell against deep discharge.

closed 2023-10-29 by Dr-Gigavolt