Dr-Gigavolt/dbus-aggregate-batteries #58: Soc of 0% forces Multiplus to load batteries from grid

jenssd, 2023-10-27:
Hi,
when my batteries gets discharged to MIN_CELL_VOLTAGE or DISCHARGE_VOLTAGE aggregate batteries sets Soc to 0%, then my 2 Multiplus's use all their power to load the batteries from grid (9kW). 
Is that correct behavior? Should the discharge current not be set just to 0 or
why not just disallow discharging instead of setting Soc to 0?

My system:
3x 16S (280AH)
3x JK BMS connected to CerboGX using SerialBattery.

For now I changes my EES to use serial battery for calculating SOC.

Best,
Jens

Dr-Gigavolt, 2023-10-27:
Hi Jens,

neither AggregateBatteries nor any other battery driver gives the charger command to charge. It is done by ESS. I expect your minimum voltage settings are too high. 

DISCHARGE_VOLTAGE should be set somewhere into the knee where the voltage begins to fall quickly), here is SoC reset to zero.
MIN_CELL_VOLTAGE < DISCHARGE_VOLTAGE sets the DCL to zero in order to protect the weakest cell.

I work on improvement, but the is-state is not too bad.

At your DISCHARGE_VOLTAGE your SoC seems to be above the minimum set in ESS. After reseting to zero the ESS does what it has to do: Bring the battery above the in ESS defined minimum using all available means.

You can try this (for LFP):

DISCHARGE_VOLTAGE = 3.1
MIN_CELL_VOLTAGE = 3.0
CELL_EMPTY_LIMITING_VOLTAGE = [MIN_CELL_VOLTAGE, DISCHARGE_VOLTAGE, DISCHARGE_VOLTAGE + 0.1]
CELL_EMPTY_LIMITED_CURRENT =  [0, 0.05, 1]

Of course you can optimize the current limitation according your experience.

Best regards,
Anton

Dr-Gigavolt, 2023-10-27:
And why is the SoC reset to zero: SoC is not measured, only counted, the error is accumulating. If the weather is bad and there is no regular 100% reset, it could happen that battery is empty and the SoC does not know about it.

jenssd, 2023-10-27:
In MP these are my settings 

[image]

I had 
DISCHARGE_VOLTAGE = 2.9
MIN_CELL_VOLTAGE = 2.95
but changed it now to your suggested values.

When I activate OWN_SOC in settings.py the SOC is set to 0, although there are more than 10% left in each battery pack and the MP starts to load the batteries as you can see here.
[image]
 I also get low voltage alarm, but who throws it, where does it come from?
[image]

Using one of the Serial Batteries instances as Batterie Monitor all is working fine.
I use ESS mode optimized without battery life.

Dr-Gigavolt, 2023-10-27:
If you have more than 10% at your original setting you should reduce it, not increase. The 3V I read out from a curve in Internet, I have no experience with LFP, I use LTO.

The issue with the alarm seems to be Victron bug, see [issue 31](https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/31), last comments. I still have no solution.

Dr-Gigavolt, 2023-10-27:
I can make a switch in the new update to disable zeroing of SoC. As explained above, you don't need it for right setup, but it can be helpful for adjustment.

jenssd, 2023-10-27:
It seems that Victron does not like setting SOC to 0, because then it uses all possibilities to load the battery again, which is ok, if the SOC would be really 0% and not as in my case over 10%. I changed now OWN_SOC to false to use the SOC from SerialBatteries. 
It's the first time my batteries are at this state, so I'm a little bit nervous how they will behave.

Dr-Gigavolt, 2023-10-27:
Please try the following:
- voltages as above
- ESS SoC minimum to 5%
- in the aggregatebatteries.py change self._ownCharge = to the number of Ah corresponding your SoC a bit above 1%

if works, set the reseting SoC back to zero. Thanks.

jenssd, 2023-10-27:
`change self._ownCharge = to the number of Ah corresponding your SoC a bit above 1%`

There are several assignments to this variable, which one do you mean?

`if works, set the reseting SoC back to zero`

Sorry, also don't understand where to reset.

Dr-Gigavolt, 2023-10-27:
[image]

Dr-Gigavolt, 2023-10-27:
You should configure as simple as possible for adjustment, the extras you can activate later.

jenssd, 2023-10-27:
Will test it later, have to go for today, family is calling for me :). 
Ok, understand. I'm also German speaking :).

Dr-Gigavolt, 2023-10-27:
I prefer English when speaking with machines, the technical expressions are simplier :-)

jenssd, 2023-11-04:
I'm still testing. For now I switched to SOC of SerialBatteries, but now I get a Low battery alarm as fas as the MaxDischargeCurrent is set to 0
[image]

It does not play any role at which voltage I set the MaxDischargeCurrent to 0, as far as I set it I get the low voltage alarm.
e.g when I set these values, I also get the alarm when 48 V are reached
```
DISCHARGE_VOLTAGE = 3.0                                 # If reached, discharge current set to zero
MIN_CELL_VOLTAGE = 2.85                                  # If reached, discharge current set to zero

CELL_EMPTY_LIMITING_VOLTAGE = [MIN_CELL_VOLTAGE, DISCHARGE_VOLTAGE, DISCHARGE_VOLTAGE + 0.2]   
CELL_EMPTY_LIMITED_CURRENT =  [0, 0, 1]
```

I also tried some small hack in your code, with the same result:

```
        #######################
        # Send values to DBus #
        #######################
        #JEI
        if abs(Voltage) < 49:
            MaxDischargeCurrent = 0

        with self._dbusservice as bus:
```

Why does the MP not just go in idle when MaxDischargeCurrent is set to 0, is this a bug?

As a current workaround I use this settings, 

```
DISCHARGE_VOLTAGE = 3.0                                 # If reached, discharge current set to zero
MIN_CELL_VOLTAGE = 2.85                                  # If reached, discharge current set to zero

CELL_EMPTY_LIMITING_VOLTAGE = [MIN_CELL_VOLTAGE, DISCHARGE_VOLTAGE, DISCHARGE_VOLTAGE + 0.2]   
CELL_EMPTY_LIMITED_CURRENT =  [0, 0.005, 1]
```

which reduce the MaxDischargeCurrent to a very low value (~3 A) when the DISCHARGE_VOLTAGE is reached. This seems to work without raising any alarm. I'll test it this night. But when MIN_CELL_VOLTAGE is reached again and MaxDischargeCurrent will be set to 0, I believe that then the alarm will be raised. 
Is here a state missing? When DISCHARGE_VOLTAGE is reached the MaxDischargeCurrent is linear reduced, and now at some point the charge should stop, when voltage drops below DISCHARGE_VOLTAGE but not yet reached MIN_CELL_VOLTAGE?

Dr-Gigavolt, 2023-11-05:
Hi Jens, in one of the last commits I removed the DISCHARGE_VOLTAGE. It has no relevance because the only what is important to avoid deep discharge is the lowest cell voltage. And I fixed 2 serious bugs, please take the last commit.

About alarm - it seems to be the Victron bug, there was a lot reported about it. The work around should work with DCL >= 0.1A. What was the trigger in your case? Zero SoC? In this case I could send e.g. max(self._own_soc, 0.1) to dBus.

jenssd, 2023-11-06:
I'll take a look on the lastest version, thanks. 

It was as assumed, setting MaxDischargeCurrent to this 3 A over 3 Batteries works for hours, but at 3 AM the cell voltage of one cell drops to MIN_CELL_VOLTAGE and MaxDischargeCurrent was set to 0 and the alarm occurred. And then the "ping pong" started. Discharge stops, cell voltage raised again, discharge started, short after, MIN_CELL_VOLTAGE was reached again, alarm occurred, discharged stopped, cell voltage raised again, and so on. At this time I deactivated the inverter and only allowed charging.

Now it was the MIN_CELL_VOLTAGE and DISCHARGE_VOLTAGE which was reached in some way. SOC was about 20% (collected over all 3 batteries, something about 19%, 24%, 31%). When going under 3V my cell voltages goes apart, sometimes over 100 mV, so some cells are still on 3.0 or 3.1 and some at 2.9 or 2.8. I know that it looks like to do some TopBalancing over all packages, I'll put it on my list :). In past I only did it per pack at different times, over summer that was no issue, but now I get this big cell differences. 
Anyway, setting SOC or MaxDischargeCurrent seem to be values, which should not be set to 0, unless you want Victron to raise alarms.
This night I did some other test. Setting EES SOC MinValue to 25% (with optimized, without battery life), before I had it on 20%, this was to low, with 25% the discharging stops at 25% (the inverter went into Idle state) and my cell voltages were about 2.9-3.0V. Which is ok for me now. When I did the TopBalancing over all packages, I hope that I can reduce the EES SOC MinValue again, and not hit any MinCellVoltage limit.

Dr-Gigavolt, 2023-11-08:
If I understood correctly, the oscillation was the issue. Not the alarm, is it OK if triggered when the battery is empty? Or is there some other problem due to the alarm? I can set a hysteresis to avoid start discharging before some charging done. The workaround by keeping DCL above zero is not the best option, after some time it would discharge the battery critically and disconnect it by BMS.

jenssd, 2023-11-10:
The issue was at first the oscillation of alarm and no alarm depending on the voltage, yes. Setting hysteresis to avoid start discharging before some charging done, sounds good for me and makes also sense in any way. EES does the same (let first charge up to 5% and then allows discharge again).
But the main question for me is, why can the EES stops discharging the battery and say the MP "go into idle" without raising the low battery voltage / low SOC alarm, and if you set either SOC or MaxDischargeCurrent to 0 we always get an alarm?

Dr-Gigavolt, 2023-11-12:
Hi Jens

> The issue was at first the oscillation of alarm and no alarm depending on the voltage, yes. Setting hysteresis to avoid start discharging before some charging done, sounds good for me and makes also sense in any way. 

I made the commit just now.

> EES does the same (let first charge up to 5% and then allows discharge again). 

But this is not working properly, if Peak shaving is set to "Above minimum SoC only" even the AC coupled charging is disabled during this time.

> But the main question for me is, why can the EES stops discharging the battery and say the MP "go into idle" without raising the low battery voltage / low SOC alarm, and if you set either SOC or MaxDischargeCurrent to 0 we always get an alarm?

> EES stops discharging the battery and say the MP "go into idle"
When happens this? Below the MP minimum voltage settings? This is a level below ESS, this could explain ESS does not know about it. But ESS alarm at zero SOC or MaxDischargeCurrent is just a warning, or what else happens?

Dr-Gigavolt, 2023-11-13:
Today I had the low cell voltage case, see below. The low voltage alarm was triggered and it is OK. The battery was charged by the full power of MP from grid (full power is not necessary, I can add a reduction) until the ESS minimum (15%) was reached. IMHO the performance is OK. I will add logging of SoC and max/min voltage cell numbers.

The full discharge happens at some residual SoC, but it is not surprising at this weather when the 100% reset happens seldom. The inaccuracy is caused by the efficiency of the battery which is lower than unity. 

INFO:root:Mon Nov 13 18:29:28 2023: Min. cell voltage: 1.911V, Max. cell voltage: 2.074V, difference: 0.163V
INFO:root:Mon Nov 13 18:39:30 2023: CVL: 53.9V, CCL: 300A, DCL: 1A
INFO:root:Mon Nov 13 18:39:30 2023: Bat. voltage: 45.3V, Bat. current: -1A, Balancing state: 0
INFO:root:Mon Nov 13 18:39:30 2023: Min. cell voltage: 1.905V, Max. cell voltage: 2.074V, difference: 0.169V
INFO:root:Mon Nov 13 18:49:32 2023: CVL: 53.9V, CCL: 300A, DCL: 200A
INFO:root:Mon Nov 13 18:49:32 2023: Bat. voltage: 47.4V, Bat. current: 212A, Balancing state: 0
INFO:root:Mon Nov 13 18:49:32 2023: Min. cell voltage: 2.136V, Max. cell voltage: 2.178V, difference: 0.042V
INFO:root:Mon Nov 13 18:59:33 2023: CVL: 53.9V, CCL: 300A, DCL: 200A
INFO:root:Mon Nov 13 18:59:33 2023: Bat. voltage: 47.8V, Bat. current: 210A, Balancing state: 0
INFO:root:Mon Nov 13 18:59:33 2023: Min. cell voltage: 2.157V, Max. cell voltage: 2.195V, difference: 0.038V
INFO:root:Mon Nov 13 19:09:34 2023: CVL: 53.9V, CCL: 300A, DCL: 200A
INFO:root:Mon Nov 13 19:09:34 2023: Bat. voltage: 47.0V, Bat. current: -1A, Balancing state: 0
INFO:root:Mon Nov 13 19:09:34 2023: Min. cell voltage: 2.126V, Max. cell voltage: 2.142V, difference: 0.016V

jenssd, 2023-12-16:
> When happens this? Below the MP minimum voltage settings? This is a level below ESS, this could explain ESS does not know about it. But ESS alarm at zero SOC or MaxDischargeCurrent is just a warning, or what else happens?

In my first posts here I mentioned:

> when my batteries gets discharged to MIN_CELL_VOLTAGE or DISCHARGE_VOLTAGE aggregate batteries sets Soc to 0%, then my 2 Multiplus's use all their power to load the batteries from grid (9kW).
> Is [...]

After that you've added the switch to not set SOC to 0%. But then I got the other issue:
https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/58#issuecomment-1793567592

During the last weeks I could do another top balancing over all 3 packs. Since then when the SOC reached 15% the EES min soc limit kicks in and stoped discharging (if thats happen I still have a min_cell_value of 3.0 till 3.1V) but the capacity which came out is nearly 40 kWh. So, I use now this approach since weeks, trying not to go under 3.0 V and let do the EES the job.

If nothing else we can do here for my second issue, we can close this ticket.

Thanks for your great work!

closed 2023-12-16 by jenssd