Dr-Gigavolt/dbus-aggregate-batteries #60: Idle discharge during winter

TazerReloaded, 2023-12-25:
Hello,

I've been using this driver for about a year now, and everything works fine, except a little problem during winter:
I have configured `CURRENT_FROM_VICTRON = True` and `OWN_SOC = True` because my two JKs are not very precise.

I have around 25kWh of total LiFePO4 capacity, and my 12kWp system will usually never fully charge my batteries from November to February. I have a Cerbo and and an 48V->12V PSU running directly from the batteries, so those are not accounted for in the inverter and MPPT measurements.

About every 20 days during this time, the SoC will jump from 30% (configured minimum) to 0% due to a low voltage alarm. I think that's correct behavior, considering the battery is actually discharged. I have set conservative voltage limits to avoid damage.

I have also thought about a solution, and the simplest could be to implement a configurable idle drain constant in the config, where I could put for example "20W", and this constant will always be subtracted during SoC calculations. If the constant is too low, the issue persists, but with longer intervals, if it is too high the battery could be fully charged after some time, but with a bit of tuning it should get me through the winter without alarms.

Any opinions on this issue? Or have I missed another, even simpler solution? I am aware that a Smart Shunt would likely also solve this problem, but I don't have a common negative terminal in my setup, everything is bolted to copper bus bars, with the Victron stuff in the middle and the two batteries on each end, so a software solution would be preferred on my end.

Dr-Gigavolt, 2023-12-25:
Hello. I observed the same behaviour. The solution is uploaded. The problem was the efficiency of batteries which is always below unity. The error is being accumulated if the batteries and the charge counter are not reset by 100% charge for longer period of time. I added the parameter BATTERY_EFFICIENCY, 0.98 as default. The charge fed into batteries is now multiplied by this parameter in order not to neglect the losses:

        if Current > 0:
            self._ownCharge += Current * (deltaTime / 3600) * BATTERY_EFFICIENCY                # charging (with efficiency)
        else:    
            self._ownCharge += Current * (deltaTime / 3600)                                     # discharging

Please test. You can adjust the BATTERY_EFFICIENCY parameter by observation of the voltage and SoC.

TazerReloaded, 2023-12-26:
Sounds good, I pulled the newest commit and transferred my config. Will test and report results.
Thank you for your efforts!

TazerReloaded, 2024-02-26:
Update after 2 months of testing: No more alerts, battery voltage stayed above minimum cutoff the entire time. The default 0.98 efficiency factor seems to work well with my setup.
Thank you for the fix and the fine pice of software!

closed 2024-11-07 by Dr-Gigavolt