Dr-Gigavolt/dbus-aggregate-batteries #61: Paralelschaltung von min 2 Battery Packs gefärlich wenn an jedem ein BMS verbaut ist? Is Parallel Connection of at Least 2 Battery Packs Dangerous When Each Has a BMS Installed?

RENE-TS, 2024-01-18:
Sorry ich schreibe hier mal in Deutsch. 

Also, ich habe zwei Batterypacks, jeweils mit einem JBD AP21S002 300 A BMS. Folgendes ist passiert: Ein BMS hatte aufgrund der Sleep-Funktion abgeschaltet, ist aber nicht mehr aus dem Sleep-Modus erwacht und trennte die Batterie vom DC-System. Das andere BMS ging nicht in den Sleep-Modus und trennte die zweite Batterie nicht vom System. Jetzt wurde nur ein Pack geladen, das andere nicht. Pack 1 hatte dann 20% und Pack 2 dann 100%. Wenn jetzt das BMS einfach aus dem Sleep erwacht und sich einschaltet, kracht es mit Sicherheit gewaltig? Gibt es eine Funktion in der Software, die das überwacht? Zum Beispiel ab 0,8 Volt Differenz Victron off. Aber das müsste ja BMS-seitig erfolgen?

So, I have two battery packs, each with a JBD AP21S002 300 A BMS. Here's what happened: One BMS had shut down due to the sleep function, but it did not wake up from sleep mode and disconnected the battery from the DC system. The other BMS did not enter sleep mode and did not disconnect the second battery from the system. Now only one pack was being charged, not the other. Pack 1 then had 20% and Pack 2 had 100%. If the BMS suddenly wakes up from sleep and turns on, it will definitely crash with a bang, right? Is there a function in the software that monitors this? For example, Victron turns off at a 0.8-volt difference. But this would have to be done on the BMS side.   mit Chat GPT übersetzt. Sorry

RENE-TS, 2024-01-18:
Bleibt mir nur die Variante mit Node red? Die Anlage steht etwas weiter weg und der Besitzter ist technisch nicht so fit.

pos-ei-don, 2024-01-18:
Node Red ist bei victron gx quasi mit dabei. Was gefällt dir dran nicht?

RENE-TS, 2024-01-18:
Das gefällt mir sehr gut weil so flexibel. Wenn man über Node Red das BMS noch steuern könnte wäre es perfekt. Grüße René

Dr-Gigavolt, 2024-01-19:
Hallo, mich überrascht dass das BMS eine Sleep-Funktion hat. Am besten deaktivieren wenn es geht.  In aggregatebatteries.py, wenn OWN_CHARGE_PARAMETERS = True ist der Fall abgesichert, Zeilen 615 und 616:

            if NrOfModulesBlockingCharge > 0:
                MaxChargeCurrent = 0

Ich gehe davon aus, dass du mein Alghoritmus deaktiviert hast. Dann werden die Daten aus Serial Battery instances kombiniert und an Victron weiter gegeben, auch die Alarme. Aber es scheint, dass Victron die Alarme nur anzeigt, aber sonst nichts tut. Bei dem nächsten Update kann ich in solchem Fall den Ladestrom auch bei OWN_CHARGE_PARAMETERS = False auf Null setzen.

Um beide Akkus wieder auszugleichen erlaube beiden Akkus Ladung und verbiete Entladung bis beide voll sind. Notfalls 1x aus dem Netz laden wenn die Sonnenleistung nicht ausreicht um es in vernünftiger Zeit zu erledigen.

RENE-TS, 2024-01-19:
Hallo, 
ja, leider hat das JBD AP21S002 eine Sleep-Funktion. Ich habe zwar versucht, mit dem Support zu sprechen, aber da kommt nicht viel außer dem Datenblatt. Die Windows-Software ist sehr schlecht übersetzt, man kann nur raten, was alles bedeutet. In der App habe ich nichts gefunden. Man kann in der Windows-Software bis maximal 65000 Sekunden (ca. 18 Stunden) einstellen, ab wann das BMS in den Sleep-Modus geht. Bei dem schlechten Sonnenwetter war das wohl mehr, und ein BMS ging in den Sleep-Modus, das andere nicht. Ich habe zwei Batteriepacks mit den besagten BMS und in jedem Pack einen Schalter im Plus. Ich konnte das BMS, das nicht im Sleep-Modus war, wieder aufwecken, indem ich es kurz aus- und wieder eingeschaltet habe. Seitdem läuft alles wieder. Aber ruhig schlafen kann ich nicht.

RENE-TS, 2024-01-19:
"wieder aufwecken, indem ich es kurz aus- und wieder eingeschaltet habe."  Sorry ich meinte das Pack mit dem BMS das **NICHT** im Sleep war! Wie es zu dieser Sittuation kommt ist mir noch unklar aber eine Vermutung habe ich. Auch verstehe ich nicht warum es nicht wieder "aufwacht" wenn sie Ströme wieder steigen? Ich habe gesehen das über dem Relais ein 10 Ohm R auf die Ausgangsseite geschlten ist.

transistorgit, 2024-05-04:
Hallo,

die Frage ist schon etwas her, aber ich antworte hier weil es wichtig ist. Parallel schalten zweier Batterien ist sehr gefährlich. Normalerweise verbieten die BMS Hersteller das auch in der Doku. Daly hat zu diesem Zweck zum Beispiel das „Parallel Modul“ im Angebot. Darüber werden zwei BMS hardwaremäßig synchronisiert, so dass immer beide abschalten.
Konkret kann ein BMS aus verschiedensten Gründen abschalten. Ich hatte es zum Beispiel anfänglich öfters, dass ein NTC falsche Werte geliefert hat und so ein BMS wegen Untertemperatur abgeschaltet hat. Weiterhin sind es software gesteuerte Geräte, diese Software kann auch jederzeit Störungen haben.
Eine einfache Parallelschaltung von zwei Batterien mit BMS ist auf jeden Fall gefährlich und nicht fachgerecht, denn euch kann die Bude abbrennen.

Viele Grüße

closed 2024-11-07 by Dr-Gigavolt