Dr-Gigavolt/dbus-aggregate-batteries #62: Log "Spamming" after reactivation of DC Coupled PV Feed in

snow78, 2024-02-01:
when the situation appears that driver deactivates the DC coupled PV feed in. after reactivation is spams the log in seconds 
INFO:root:Thu Feb  1 16:27:00 2024: DC-coupled PV feed-in re-activated.
INFO:root:Thu Feb  1 16:27:01 2024: DC-coupled PV feed-in re-activated.
INFO:root:Thu Feb  1 16:27:02 2024: DC-coupled PV feed-in re-activated.

also it every second reactivates the 'com.victronenergy.settings', '/Settings/CGwacs/OvervoltageFeedIn', 1
on the one hand not sure if that has negativ impact on the system in general when in second interval the setting is set.
second you can't turn it off manually as it always reactivates in a second intervall
i suggest to reset the self._DCfeedActive to false as easiest solution
This recreates the situation as on driver start and will be set again to true when at next occassion when CVL reduction triggers the DC Feed in is still active

would be in  aggregatebatteries.py lines 610 -613

                if ((MaxCellVoltage - MinCellVoltage) < CELL_DIFF_MAX) and self._DCfeedActive:                                              # re-enable DC-feed if it was enabled before
                    self._dbusMon.dbusmon.set_value('com.victronenergy.settings', '/Settings/CGwacs/OvervoltageFeedIn', 1)                  # enable DC-coupled PV feed-in
                    logging.info('%s: DC-coupled PV feed-in re-activated.'  % (dt.now()).strftime('%c'))
                    self._DCfeedActive = False #reset to prevent permanent logging and activation of  /Settings/CGwacs/OvervoltageFeedIn

Thanks

snow78, 2024-02-01:
i think the logic needs to be rethought a bit more. when there is the situation that MAX_CELL_VOLTAGE is exceeded first time. self._DCfeedActive is set to the value that the /Settings/CGwacs/OvervoltageFeedIn has (lets assume its enabled). Under the condition that Cell Voltage falls below  MAX_CELL_VOLTAGE but the CELL_DIFF_MAX is still exceeded the logic will set self._dynamicCVL to false again but keep the /Settings/CGwacs/OvervoltageFeedIn switched off. when now even with the DC Feed in disabled the MAX_CELL_VOLTAGE is exceeded the self._DCfeedActive will be overwriten with false and never reactivated. suggestion is to put another "if" in Line 598
                    if self._DCfeedActive == False
                        self._DCfeedActive = self._dbusMon.dbusmon.get_value('com.victronenergy.settings', '/Settings/CGwacs/OvervoltageFeedIn')    # check if DC-feed enabled

this will only reread the setting if self._DCfeedActive is false. otherwise it will remember that it was true as long as it is not reset.

Dr-Gigavolt, 2024-02-08:
Thank you, it seems correct. But I cannot test now, bad weather.

closed 2024-11-07 by Dr-Gigavolt