Dr-Gigavolt/dbus-aggregate-batteries #63: Quick Question Using smart shunt

Sarah-1331, 2024-03-21:
its been a while since i set up my rpi with venus os and my JBD bms 

so i appoligise for the noob question 

can i use this to display just the current of a smart shunt and the rest of the battery details from my bms? as my bms seems ok for SOC but the current not been accurate kills me like the first 12watts doesnt register..

thanks again

Dr-Gigavolt, 2024-03-21:
The driver is not intended to measure the entire battery current by smart shunt because it would be wasting of the smart shunt. Either by BMSes or as a sum of all chargers and inverters. A smart shunt is intended for DC consumption only. If your BMSes count the the SoC correctly, it is not possible that the current is not precise. SoC(t) = 100% * q(t)/Q(full); q(t) = integral{ i(t)*dt }

Dr-Gigavolt, 2024-03-21:
The first 12W seem to be negligible.

closed 2024-05-28 by Dr-Gigavolt