Dr-Gigavolt/dbus-aggregate-batteries #64: Trouble installing

coca7, 2024-04-06:
Hello,

I have had trouble installing bus-aggregate-batteries.  I think I am making an error.
I wasn't sure how to copy the files to /data/dbus-aggregate-batteries so I just did a wget together the files and then I untried and and renamed the directory and adjusted the settings.  However, it doesn't seem to be running.

My setup is 3 JBD Bms each with 4x 230ah cells
I have a smart shunt, but I have not yet moved it to the DC side of my setup.

coca7, 2024-04-06:
Here is what I put in settings:

> Last login: Sat Apr  6 13:17:14 on ttys001
padre@MacBookPro ~ % ssh root@192.168.2.133
root@192.168.2.133's password: 
Last login: Sat Apr  6 18:19:44 2024 from 192.168.2.171
root@einstein:~# cd /
root@einstein:/# ls
Settings    dev         lost+found  proc        sys
bin         etc         media       run         tmp
boot        home        mnt         sbin        usr
data        lib         opt         service     var
root@einstein:/# cd data/
root@einstein:/data# cd dbus-aggregate-batteries/
root@einstein:/data/dbus-aggregate-batteries# ls
root@einstein:/data/dbus-aggregate-batteries# wget https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/archive/refs/tags/3.1-beta.tar.gz
--2024-04-06 18:22:12--  https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/archive/refs/tags/3.1-beta.tar.gz
Resolving github.com... 140.82.113.3
Connecting to github.com|140.82.113.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Dr-Gigavolt/dbus-aggregate-batteries/tar.gz/refs/tags/3.1-beta [following]
--2024-04-06 18:22:13--  https://codeload.github.com/Dr-Gigavolt/dbus-aggregate-batteries/tar.gz/refs/tags/3.1-beta
Resolving codeload.github.com... 140.82.114.9
Connecting to codeload.github.com|140.82.114.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘3.1-beta.tar.gz’

3.1-beta.tar.gz         [ <=>                ]  15.48K  --.-KB/s    in 0.1s    

2024-04-06 18:22:14 (152 KB/s) - ‘3.1-beta.tar.gz’ saved [15852]

root@einstein:/data/dbus-aggregate-batteries# ls
3.1-beta.tar.gz
root@einstein:/data/dbus-aggregate-batteries# tar xvf 3.1-beta.tar.gz 
dbus-aggregate-batteries-3.1-beta/
dbus-aggregate-batteries-3.1-beta/.gitattributes
dbus-aggregate-batteries-3.1-beta/LICENSE
dbus-aggregate-batteries-3.1-beta/README.md
dbus-aggregate-batteries-3.1-beta/aggregatebatteries.py
dbus-aggregate-batteries-3.1-beta/charge
dbus-aggregate-batteries-3.1-beta/dbusmon.py
dbus-aggregate-batteries-3.1-beta/functions.py
dbus-aggregate-batteries-3.1-beta/last_balancing
dbus-aggregate-batteries-3.1-beta/restart
dbus-aggregate-batteries-3.1-beta/restart_dbus-serial-battery
dbus-aggregate-batteries-3.1-beta/service/
dbus-aggregate-batteries-3.1-beta/service/run
dbus-aggregate-batteries-3.1-beta/settings.py
root@einstein:/data/dbus-aggregate-batteries# ls
3.1-beta.tar.gz                    dbus-aggregate-batteries-3.1-beta
root@einstein:/data/dbus-aggregate-batteries# nano dbus-aggregate-batteries-3.1-beta/
root@einstein:/data/dbus-aggregate-batteries# cd dbus-aggregate-batteries-3.1-beta/
root@einstein:/data/dbus-aggregate-batteries/dbus-aggregate-batteries-3.1-beta# ls
LICENSE                      last_balancing
README.md                    restart
aggregatebatteries.py        restart_dbus-serial-battery
charge                       service
dbusmon.py                   settings.py
functions.py
root@einstein:/data/dbus-aggregate-batteries/dbus-aggregate-batteries-3.1-beta# ls
LICENSE                      last_balancing
README.md                    restart
aggregatebatteries.py        restart_dbus-serial-battery
charge                       service
dbusmon.py                   settings.py
functions.py
root@einstein:/data/dbus-aggregate-batteries/dbus-aggregate-batteries-3.1-beta# cd ..
root@einstein:/data/dbus-aggregate-batteries# ls
3.1-beta.tar.gz                    dbus-aggregate-batteries-3.1-beta
[... 330 lines cut ...]
BATTERY_INSTANCE_NAME_PATH = '/CustomName'              # if CustomName doesn't exist, set '/ProductName'
MULTI_KEY_WORD = 'com.victronenergy.vebus'              # Key world to identify service of Multis/Quattros (or cluster of them)
MPPT_KEY_WORD = 'com.victronenergy.solarcharger'        # Key world to identify services of solar chargers
SMARTSHUNT_NAME_KEY_WORD = 'SmartShunt'                 # Key world to identify services of SmartShunt

SEARCH_TRIALS = 10                                      # Trials to identify of all batteries before exit and restart
READ_TRIALS = 10                                        # Trials to get consistent data of all batteries before exit and restart

#######################################
############## Options ################
#######################################

CURRENT_FROM_VICTRON = True                             # If True, the current measurement by Multis/Quattros and MPPTs is taken instead of BMS.
                                                        # Necessary for JK BMS due to poor precision.
OWN_SOC = True                                          # If True, the self calculated charge indicators are taken instead of BMS
ZERO_SOC = True                                         # Allow zeroing charge counter at MIN_CELL_VOLTAGE. At full battery it is always set to 100%.
CHARGE_SAVE_PRECISION = 0.0025                          # Trade-off between save precision and file access frequency

#######################################
##### Charge/Discharge parameters #####
#######################################

                                                                                     [ Read 88 lines ]
^G Get Help	 ^O Write Out     ^W Where Is	   ^K Cut Text      ^J Justify       ^C Cur Pos       M-U Undo         M-A Mark Text    M-] To Bracket   M-Q Previous     ^B Back
^X Exit          ^R Read File     ^\ Replace	   ^U Paste Text    ^T To Spell      ^_ Go To Line    M-E Redo         M-6 Copy Text    ^Q Where Was     M-W Next         ^F Forward

coca7, 2024-04-07:
There is no log file created after I installed, edited the settings.py and rc.local and rebooted.

Dr-Gigavolt, 2024-04-07:
Hello, have you perhaps forgot to create the symbolic link?
 ln -s /data/dbus-aggregate-batteries/service /service/dbus-aggregate-batteries into /data/rc.local
 
I've just released last commit, here you can download the stuff "install" according Readme file.
https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/releases/tag/3.1.1-beta

Dr-Gigavolt, 2024-04-07:
wget from /data should work too:
sudo wget https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/archive/refs/tags/3.1.1-beta.tar.gz
sudo tar -xvf 3.1.1-beta.tar.gz
sudo mv dbus-aggregate-batteries-3.1.1-beta dbus-aggregate-batteries
cd dbus-aggregate-batteries
sudo chmod 777 restart ./service/run

and then open /data/rc.local with a text editor and add 
_ln -s /data/dbus-aggregate-batteries/service /service/dbus-aggregate-batteries_
line.

Dr-Gigavolt, 2024-04-07:
The log file aggregatebatteries.log should be created in /data/dbus-aggregate-batteries

coca7, 2024-04-07:
I did add the line in to rc.local but I have a question about it.
does the past start with "In" or "-s". I started with -s.  Everything else you suggested I did except I didn't run the command with sudo.  Also the readme had said to chmod 744, whereas here you said to chmod 777.

Dr-Gigavolt, 2024-04-08:
The line begins with ln (command), -s is a parameter. The permission 744 is ok (777 works as well, but unnecessarily too much rights).

coca7, 2024-04-08:
The driver runs, but it doesn't find any of the batteries.  I think I have the DBUS settings wrong:

Here is what the log says:

>  INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
> INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2, scanning and storing items
> INFO:dbusmonitor:	com.victronenergy.battery.ttyUSB2 has device instance 4
> INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
> INFO:dbusmonitor:	com.victronenergy.settings has device instance 0
> INFO:root:Mon Apr  8 13:26:01 2024: Searching Settings: Trial Nr. 1
> INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyS5, scanning and storing items
> INFO:root:Mon Apr  8 13:26:01 2024: com.victronenergy.settings found.
> INFO:dbusmonitor:       com.victronenergy.solarcharger.ttyS5 has device instance 278
> INFO:dbusmonitor:Found: com.victronenergy.solarcharger.ttyS6, scanning and storing items
> INFO:dbusmonitor:	com.victronenergy.solarcharger.ttyS6 has device instance 279
> INFO:dbusmonitor:Found: com.victronenergy.system, scanning and storing items
> INFO:dbusmonitor:	com.victronenergy.system has device instance 0
> INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyS4, scanning and storing items
> INFO:dbusmonitor:	com.victronenergy.vebus.ttyS4 has device instance 276
> INFO:dbusmonitor:Found: com.victronenergy.battery.ttyS7, scanning and storing items
> INFO:dbusmonitor:       com.victronenergy.battery.ttyS7 has device instance 277
> INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
> INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
> INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
> INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 3
> INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
> INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 2
> INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
> INFO:root:Mon Apr  8 13:26:06 2024: Searching batteries: Trial Nr. 1
> INFO:root:Mon Apr  8 13:26:06 2024: 0 batteries found.
> INFO:root:Mon Apr  8 13:26:11 2024: Searching batteries: Trial Nr. 2
> INFO:root:Mon Apr  8 13:26:11 2024: 0 batteries found.
> 

Here is what I put in the settings.py

```
 BATTERY_SERVICE_NAME = 'com.victronenergy.battery'	# Key world to identify services of physical Serial Batteries (and SmartShunt if available)
 BATTERY_PRODUCT_NAME_PATH = '/dev/ttyUSB'                  # Path of Battery Product Name
 BATTERY_PRODUCT_NAME = 'Battery'                  # Key world to identify the batteries (to exclude SmartShunt)
 BATTERY_INSTANCE_NAME_PATH = '/SerialBattery(LLT/JBD)'              # if CustomName doesn't exist, set '/ProductName'
 MULTI_KEY_WORD = 'com.victronenergy.vebus'              # Key world to identify service of Multis/Quattros (or cluster of them)
 MPPT_KEY_WORD = 'com.victronenergy.solarcharger'        # Key world to identify services of solar chargers
 SMARTSHUNT_NAME_KEY_WORD = 'SmartShunt'                 # Key world to identify services of SmartShunt

```

Here is what I have for one of my batteries in dbus-spy

 
[image]
[image]

Dr-Gigavolt, 2024-04-08:
The product name path is wrong. It is a dBus path, not driver path in Linux. The default is correct: BATTERY_PRODUCT_NAME_PATH = '/ProductName'

closed 2024-05-28 by Dr-Gigavolt