Dr-Gigavolt/dbus-aggregate-batteries #65: log grows endlessly

transistorgit, 2024-05-04:
Hi,

Today I Found that the aggregatebatteries.log was hundreds of MB big. Version is 3.0. Please implement standard log file handling like in other drivers.

brgds

Dr-Gigavolt, 2024-05-12:
Hi, I know it is not optimal. I will look how to do it correctly as soon as I will have more time. Please increase the log period as a temporary solution.

Dr-Gigavolt, 2024-05-28:
Hi @mr-manuel , the service/run is not executed automatically any more, have you an idea?

Dr-Gigavolt, 2024-05-28:
It seems there was an issue with the end of lines in your file, it starts again after I copied my old run file back. But the directory in /var/log is not created. 

If I start log/run from shell is shows the message:
: ambiguous redirect

mr-manuel, 2024-05-28:
Can you write me on Discord, so we solve this in no time?

mr-manuel, 2024-05-28:
Fixed with https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/pull/68.

closed 2024-11-07 by Dr-Gigavolt