Dr-Gigavolt/dbus-aggregate-batteries #69: Installation issue "reinstall-local.sh: No such file or directory"

Kletternaut, 2024-06-01:
Hi there,
during the installation on a Raspberry Pi 4B (4GB) with freshly installed VenusOS v3.40~20 large (http://updates.victronenergy.com/feeds/venus/candidate/images/raspberrypi4/venus-image-large-raspberrypi4-20240529173916-v3.40~20.rootfs.wic.gz) an error occurred due to a missing reinstall-local.sh file.
```
root@raspberrypi4:~# wget -O /tmp/install.sh https://raw.githubusercontent.com/Dr-Gigavolt/dbus-aggregate-batteries/main/install.sh
--2024-06-01 11:12:31--  https://raw.githubusercontent.com/Dr-Gigavolt/dbus-aggregate-batteries/main/install.sh
Resolving raw.githubusercontent.com... 185.199.108.133, 185.199.110.133, 185.199.109.133, ...
Connecting to raw.githubusercontent.com|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1792 (1.8K) [text/plain]
Saving to: ‘/tmp/install.sh’

/tmp/install.sh     100%[===================>]   1.75K  --.-KB/s    in 0.001s

2024-06-01 11:12:32 (3.21 MB/s) - ‘/tmp/install.sh’ saved [1792/1792]

root@raspberrypi4:~#
root@raspberrypi4:~# bash /tmp/install.sh
Downloading latest release: 3.1.1-beta
--2024-06-01 11:12:32--  https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/archive/refs/tags/3.1.1-beta.zip
Resolving github.com... 140.82.121.4
Connecting to github.com|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Dr-Gigavolt/dbus-aggregate-batteries/zip/refs/tags/3.1.1-beta [following]
--2024-06-01 11:12:32--  https://codeload.github.com/Dr-Gigavolt/dbus-aggregate-batteries/zip/refs/tags/3.1.1-beta
Resolving codeload.github.com... 140.82.121.9
Connecting to codeload.github.com|140.82.121.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘dbus-aggregate-batteries_latest.zip’

dbus-aggregate-batt     [ <=>                ]  18.93K  --.-KB/s    in 0.004s

2024-06-01 11:12:33 (5.03 MB/s) - ‘dbus-aggregate-batteries_latest.zip’ saved [19386]

bash: /data/dbus-aggregate-batteries/reinstall-local.sh: No such file or directory
root@raspberrypi4:~#
```

Dr-Gigavolt, 2024-06-01:
Hello, please try again. The 3.2 was not released, now I've done it.

Kletternaut, 2024-06-02:
Dankeschön!
Habe es gerade nochmals getestet und die Installation ist nun durchgelaufen.

```
root@raspberrypi4:~# wget -O /tmp/install.sh https://raw.githubusercontent.com/Dr-Gigavolt/dbus-aggregate-batteries/main/install.sh
--2024-06-02 17:18:27--  https://raw.githubusercontent.com/Dr-Gigavolt/dbus-aggregate-batteries/main/install.sh
Resolving raw.githubusercontent.com... 185.199.108.133, 185.199.111.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1792 (1.8K) [text/plain]
Saving to: ‘/tmp/install.sh’

/tmp/install.sh               100%[===============================================>]   1.75K  --.-KB/s    in 0.001s

2024-06-02 17:18:28 (3.20 MB/s) - ‘/tmp/install.sh’ saved [1792/1792]

root@raspberrypi4:~# bash /tmp/install.sh
Downloading latest release: 3.2-beta
--2024-06-02 17:18:39--  https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/archive/refs/tags/3.2-beta.zip
Resolving github.com... 140.82.121.4
Connecting to github.com|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/Dr-Gigavolt/dbus-aggregate-batteries/zip/refs/tags/3.2-beta [following]
--2024-06-02 17:18:39--  https://codeload.github.com/Dr-Gigavolt/dbus-aggregate-batteries/zip/refs/tags/3.2-beta
Resolving codeload.github.com... 140.82.121.10
Connecting to codeload.github.com|140.82.121.10|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘dbus-aggregate-batteries_latest.zip’

dbus-aggregate-batteries_late     [ <=>                                            ]  22.46K  --.-KB/s    in 0.01s

2024-06-02 17:18:40 (1.70 MB/s) - ‘dbus-aggregate-batteries_latest.zip’ saved [23003]

Restore settings.py
chown: /data/etc/dbus-serialbattery: No such file or directory
sed: /data/rc.local: No such file or directory
grep: /data/rc.local: No such file or directory

Installation successful! Please modify the settings.py file in /data/dbus-aggregate-batteries to your needs and then reboot the device.

root@raspberrypi4:~#
```

Kletternaut, 2024-06-02:
Nach dem Anpassen der settings.py in /data/dbus-aggregate-batteries und einem anschliessendem Reboot, wurde wieder die Originalversion hergestellt. Nach der Installation erschien in der RemoteConsole ein entsprechendes Menü, welches nach dem Reboot verschwunden ist.

[image]
Diese Meldung deutete bereits bei der Installation auf ein Problem hin.
```
Restore settings.py
chown: /data/etc/dbus-serialbattery: No such file or directory
sed: /data/rc.local: No such file or directory
grep: /data/rc.local: No such file or directory

Installation successful! Please modify the settings.py file in /data/dbus-aggregate-batteries to your needs and then reboot the device.
```

Dr-Gigavolt, 2024-06-02:
Existiert die Datei /data/rc.local tatsächlich nicht? Dann probiere es zu erstellen.  Kopiere #!/bin/bash in die erste Zeile und setze die Berechtigungen auf 555 (r-xr-xr-x).

Kletternaut, 2024-06-03:
Auf einem frisch installierten System (release: 3.22) ist keine /data/rc.local vorhanden.

Bevor ich jetzt weiter prüfen kann, muss ich erst die Ursache dafür finden, weshalb sich die per can0 angeschlossene Batterien heute Morgen im Notaus-Modus befanden.

closed 2025-03-20 by Dr-Gigavolt