Dr-Gigavolt/dbus-aggregate-batteries #71: Connection to BMS lost, aggregate-driver crashes

regu-tech, 2024-09-01:
Hello,

i have a problem with crashing with the driver. It runs 3-4 weeks without any problems.

But then it could happen that for a short moment the connection to one of my 4 BMS (JBD) 
is lost for a short moment. And then the driver crashes and also a reboot does not solve the problem.
I then have to restart the driver with the restart script when the Cerbo is up again.

It shows me the following error in the log:

`ERROR:root:Occured during step Read battery state, Battery SerialBattery(LLT/JBD).
ERROR:root:Read trial nr. 1
Traceback (most recent call last):
  File "/data/dbus-aggregate-batteries/aggregatebatteries.py", line 842, in _update
    sum(VoltagesSum_dict.values()) / settings.NR_OF_BATTERIES
TypeError: unsupported operand type(s) for +: 'int' and 'NoneType'
Terminated`

My actual version of the Venus OS is 3.41

best regards
Steffen

onemorename, 2024-09-06:
Same here, just as i check VRM Portal for new updates things start to go wrong - what you see at the end 15:32 is when i run 
 /data/dbus-aggregate-batteries/restart.sh manually:

INFO:root:  CVL: 54.0V, CCL: 280A, DCL: 280A
INFO:root:  Bat. voltage: 54.0V, Bat. current: 0A, SoC: 98.2%, Balancing state: 0
INFO:root:  Min. cell voltage: Pack2_C5: 3.372V, Max. cell voltage: Pack2_C3: 3.379V, difference: 0.007V
INFO:dbusmonitor:com.victronenergy.battery.ttyUSB1 disappeared from the dbus. Removing it from our lists
INFO:dbusmonitor:com.victronenergy.battery.ttyUSB0 disappeared from the dbus. Removing it from our lists
ERROR:root:Fri Sep  6 15:21:11 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 1
ERROR:root:Fri Sep  6 15:21:12 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 2
ERROR:root:Fri Sep  6 15:21:13 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 3
ERROR:root:Fri Sep  6 15:21:14 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 4
ERROR:root:Fri Sep  6 15:21:15 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 5
ERROR:root:Fri Sep  6 15:21:16 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 6
ERROR:root:Fri Sep  6 15:21:17 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 7
ERROR:root:Fri Sep  6 15:21:18 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 8
ERROR:root:Fri Sep  6 15:21:19 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 9
ERROR:root:Fri Sep  6 15:21:20 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 10
ERROR:root:Fri Sep  6 15:21:21 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 11
ERROR:root:Fri Sep  6 15:21:21 2024: DBus read failed. Exiting.
INFO:root:Fri Sep  6 15:21:23 2024: Starting AggregateBatteries.
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:Fri Sep  6 15:21:23 2024: Initial Ah read from file: 536Ah
INFO:root:Fri Sep  6 15:21:24 2024: Starting battery monitor.
INFO:root:Fri Sep  6 15:21:24 2024: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.system, scanning and storing items
INFO:root:Fri Sep  6 15:21:25 2024: Searching Settings: Trial Nr. 1
INFO:dbusmonitor:       com.victronenergy.system has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyS4, scanning and storing items
INFO:root:Fri Sep  6 15:21:25 2024: com.victronenergy.settings found.
INFO:dbusmonitor:       com.victronenergy.vebus.ttyS4 has device instance 276
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:Fri Sep  6 15:21:30 2024: Searching batteries: Trial Nr. 1
INFO:root:Fri Sep  6 15:21:30 2024: 0 batteries found.
INFO:root:Fri Sep  6 15:21:35 2024: Searching batteries: Trial Nr. 2
INFO:root:Fri Sep  6 15:21:35 2024: 0 batteries found.
INFO:root:Fri Sep  6 15:21:40 2024: Searching batteries: Trial Nr. 3
INFO:root:Fri Sep  6 15:21:40 2024: 0 batteries found.
INFO:root:Fri Sep  6 15:21:45 2024: Searching batteries: Trial Nr. 4
INFO:root:Fri Sep  6 15:21:45 2024: 0 batteries found.
INFO:root:Fri Sep  6 15:21:50 2024: Searching batteries: Trial Nr. 5
INFO:root:Fri Sep  6 15:21:50 2024: 0 batteries found.
INFO:root:Fri Sep  6 15:21:55 2024: Searching batteries: Trial Nr. 6
[... 57 lines cut ...]
INFO:root:Fri Sep  6 15:32:17 2024: Starting AggregateBatteries.
INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregate
INFO:root:Fri Sep  6 15:32:17 2024: Initial Ah read from file: 536Ah
INFO:root:Fri Sep  6 15:32:17 2024: Starting battery monitor.
INFO:root:Fri Sep  6 15:32:17 2024: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
INFO:root:Fri Sep  6 15:32:19 2024: Searching Settings: Trial Nr. 1
INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
[... cut ...]
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1 has device instance 1
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:Fri Sep  6 15:32:24 2024: Searching batteries: Trial Nr. 1
INFO:root:Fri Sep  6 15:32:24 2024: SerialBattery(Seplos) found, named as: Pack2.
INFO:root:Fri Sep  6 15:32:24 2024: SerialBattery(Seplos) found, named as: Pack1.
INFO:root:Fri Sep  6 15:32:24 2024: 2 batteries found.

Dr-Gigavolt, 2024-09-06:
Hello guys, it is correct to crash and restart if a battery is disconnected or send an invalid value. Otherwise would aggregate batteries provide invalid charge parameters which is dangerous to the battery. Try to monitor the failing instance of the dubs serial battery in order to understand which value goes to none.

regu-tech, 2024-09-08:
Hello, last night i had again a connection loss to one of my four battery packages but there the restart has worked properly.

onemorename, 2024-09-08:
I do agree that it should crash, but why isn‘t it recovering by itself but after a restart only?
I mean the data from serialbattery is the same before and after the crash

Dr-Gigavolt, 2024-09-09:
If my battery crashes and restarts, the aggregate restarts as well. If there is a error message about incompatible operands 'int' and 'none', some battery instance, even found, doesn't send the given quantity.

Fred2938, 2024-09-09:
Hi,
I do have exactly the same issue. Just install both dbus-serialbattery and dbus-aggregate-batteries.
I do have 2 LFP pack (16cells). Both are visible on the Cerbo, with all details. 
So what kind of setup we should apply in serialbattery driver to ensure the aggregate-batteries driver works fine ? (ie in config.default.ini of serialbattery)
Thank you
[image]
[image]
[image]
[image]

Fred2938, 2024-09-09:
here is log of serialbattery.ttyUSB0 (other batterie is same in USB1):
[image]

Dr-Gigavolt, 2024-09-09:
The JPG seems to be broken, please try again. And check with dbus-spy command if all batteries are visible on the dbus and which values do they send.

Dr-Gigavolt, 2024-09-09:
The right guy who can help you with the serial-battery setup is @mr-manuel .

mr-manuel, 2024-09-09:
If you see errors in the dbus-serialbattery open a issue in the repository: https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/new/choose

Please read carefully the instructions and don't skip anything.

Fred2938, 2024-09-10:
Hi
With the corrected link to the missing above jpg:
[image]

Dr-Gigavolt, 2024-09-10:
Your cerbo screenshots contain no relevant info. Please check the aggregate batteries log file and observe dbus by dbus-spy command. Find which path contains 'None' value. If this is the case, my program must crash and restart.

Fred2938, 2024-09-10:
Hi, 
dbus-spy does not show any error message or line with 'None' value. See attached screenshots. I ony share for 1 battery, the 2nd is fully aligned with similar details (well, its name change, ... ;) )
[image]
[image]

Fred2938, 2024-09-10:
Hi, and the Cerbo Gx screenshot

[image]

Dr-Gigavolt, 2024-09-11:
Hi, now I see the problem. Your paths are not as usual. Your voltages:
[image]

My voltages:
[image]

The exception is raised in the 842, where the sum of "Voltages/Sum"  from the lines 661 and 662 are calculated. But you have the path "Cell/Sum" instead. The value of not existing path "Voltages/Sum" is "None".

@mr-manuel : Have you renamed the paths in some update?

mr-manuel, 2024-09-12:
No this is selectable since many years.

https://github.com/Louisvdw/dbus-serialbattery/blob/fd0fc1a193427df6ca0fadd141aa57826f761819/etc/dbus-serialbattery/config.default.ini#L342-L347

Fred2938, 2024-09-12:
Hi,
That is the issue and the solution !
As soon I change the BATTERY_CELL_DATA_FORMAT to 1, I get the dbus-aggregate_battery working on Cerbo !

Many thanks for your kind support and time ! Definitely really appreciate :)

Best regards

onemorename, 2024-09-13:
But isn't that the default, so if it's not changed manually it should be 1 already? 
Cells have the correct format on my end, without any customization - and aggregate_battery works perfectly for me, unless i trigger firmware search from the VRM Portal.
Serial Battery disconnects then, aggregate batteries has no values anymore and crashes as well. What happens then is, as serial battery reconnects, it has no values for 20-30 seconds until the connection is re-established fully again - and at this time aggregate batteries has given up already. I think key would be if serial battery would not post anything to dbus until it's fully up, or aggregatebatteries handles this situation better.

onemorename, 2024-09-13:
Just did look dbus-spy while things happen. The Cell voltages have empty values for around 20-30 seconds. Aggregate batteries is in a dead end afterwards

onemorename, 2024-09-13:
[image]

Dr-Gigavolt, 2024-09-13:
Please read readme. You can increase number of trials before the program gives up. But even given up, it should be restarted automatically by Linux.

onemorename, 2024-09-13:
Please see my post with the log above:

it's exiting after trial 11 to search for batteries and then restarts. Here it finds the batteries then, however with still empty values. At the read thread it just goes down to retry 4 until it fails and doesn't recover from there anymore (even while the settings.py is still set to 10 read retries per default, it doesn't seem to make it)

INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1, scanning and storing items
INFO:dbusmonitor: com.victronenergy.battery.ttyUSB1 has device instance 1
INFO:root:Fri Sep 6 15:22:39 2024: Searching batteries: Trial Nr. 2
INFO:root:Fri Sep 6 15:22:39 2024: SerialBattery(Seplos) found, named as: Pack2.
INFO:root:Fri Sep 6 15:22:39 2024: SerialBattery(Seplos) found, named as: Pack1.
INFO:root:Fri Sep 6 15:22:39 2024: 2 batteries found.
ERROR:root:Fri Sep 6 15:22:40 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack2.
ERROR:root:Read trial nr. 1
ERROR:root:Fri Sep 6 15:22:41 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read battery state, Battery Pack2.
ERROR:root:Read trial nr. 2
ERROR:root:Fri Sep 6 15:22:42 2024: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack1.
ERROR:root:Read trial nr. 3
ERROR:root:Fri Sep 6 15:22:43 2024: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'.
ERROR:root:Occured during step Read V, I, P, Battery Pack1.
ERROR:root:Read trial nr. 4
Traceback (most recent call last):
File "/data/dbus-aggregate-batteries/aggregatebatteries.py", line 842, in _update
sum(VoltagesSum_dict.values()) / settings.NR_OF_BATTERIES
TypeError: unsupported operand type(s) for +: 'float' and 'NoneType'
Terminated

Dr-Gigavolt, 2024-09-18:
> HW with a robust and galvanic isolated interface between batteries and Cerbo.

As I wrote in the dbus-serial-battery discussion about the hardware some time ago, **it is essential to use galvanic insulation**. Otherwise a part of (and in case of BMS disconnection the whole) battery current through ground of the serial line with all consequences you can imagine.

Dr-Gigavolt, 2024-09-18:
> it's exiting after trial 11 to search for batteries and then restarts. Here it finds the batteries then, however with still empty values. > At the read thread it just goes down to retry 4 until it [...]

I think I've found it out. In aggregatebatteries.py, in try block beginning in line 578 in the variable step is written by text describing what is just being done. In the exception bloc beginning in the line 819 this text is printed in order to identify the issue. This is the case here,

> ERROR:root:Fri Sep 6 15:22:43 2024: Error: unsupported operand type(s) for +=: 'float' and 'NoneType'.
> ERROR:root:Occured during step Read V, I, P, Battery Pack1.
some path is 'None'.

After 4 trials the missing value is present. But then an panhandled exception in the line 842 is raised. The voltages of all batteries are in the dictionary, some of them is 'None'. The assignment of 'None' into dictionary seems to be legal, therefore no exception starting new trial. But the line 842 is outside of the trial block, the exception causes immediate crash. I'll correct it in the program at the next opportunity.

But even in this case the program should be restarted again and again, there could be something wrong with ./service/run or the symbolic link, see the lines 12-15 in reinstall-local.sh.

The easiest way to check if it is working is to execute sh ./ restart.sh from the work directory.

regu-tech, 2024-09-30:
An additional feedback from my side, it has also crashed with the new driver but with the help from mr-manuel (Thx) the problem could be localized to a cabeling problem. It seems that the USB cables are laid to close to the batterie cables. 
I will reorganize the cables and hope that the problems are solved then.

Dr-Gigavolt, 2024-10-01:
>  It seems that the USB cables are laid to close to the batterie cables.
The battery stabilizes the voltage, not the current. The AC component from charger/inverter flows through, there is no way to filter it in case of 50Hz. That is the reason why the current measurement of JK BMS is inaccurate. I am not even sure in case of Smart Solar, running at high switching frequency, whether the current ripples are filtered sufficiently. So you have time-variable magnetic field around your battery which coupled into your USB cable. But I wonder that the shielding of the cable didn't do its job. Was the shield on both sides connected to the reference potential (digital ground)? And the galvanic insulation between the USB to serial converter and the BMS is essential.

regu-tech, 2024-10-10:
Hello again,

no i had the same problem again (Connection lost to one BMS, there i have to reorganize the cables) but after it comes back
the driver does not come up again. There where 4 BMS recognized (correct count of BMS, 4 installed) but the driver crashes therefore.

`INFO:root:Thu Oct 10 12:15:01 2024: Connected to DBus, and switching over to GLib.MainLoop()
INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0__0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0__0 has device instance 6
INFO:dbusmonitor:Found: com.victronenergy.system, scanning and storing items
INFO:root:Thu Oct 10 12:15:02 2024: Searching Settings: Trial Nr. 1
INFO:root:Thu Oct 10 12:15:02 2024: com.victronenergy.settings found.
INFO:dbusmonitor:       com.victronenergy.system has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.socketcan_can0_vi2_uc432290, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.socketcan_can0_vi2_uc432290 has device instance 2
INFO:dbusmonitor:Found: com.victronenergy.vebus.ttyS4, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.vebus.ttyS4 has device instance 276
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.socketcan_can0_vi1_uc429774, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.socketcan_can0_vi1_uc429774 has device instance 1
INFO:dbusmonitor:Found: com.victronenergy.solarcharger.socketcan_can0_vi0_uc430176, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.solarcharger.socketcan_can0_vi0_uc430176 has device instance 0
INFO:dbusmonitor:Found: com.victronenergy.battery.aggregate, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.aggregate has device instance 0
INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
INFO:root:Thu Oct 10 12:15:07 2024: Searching batteries: Trial Nr. 1
INFO:root:Thu Oct 10 12:15:07 2024: JBD-AP20S003S-L20S-300A-300A-B found, named as: SerialBattery(LLT/JBD).
INFO:root:Thu Oct 10 12:15:07 2024: 1 batteries found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB1__0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB1__0 has device instance 3
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB2__0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB2__0 has device instance 5
INFO:root:Thu Oct 10 12:15:12 2024: Searching batteries: Trial Nr. 2
INFO:root:Thu Oct 10 12:15:12 2024: JBD-AP20S003S-L20S-300A-300A-B found, named as: SerialBattery(LLT/JBD).
INFO:root:Thu Oct 10 12:15:12 2024: JBD-AP20S003S-L20S-300A-300A-B found, named as: SerialBattery(LLT/JBD)2.
INFO:root:Thu Oct 10 12:15:12 2024: JBD-AP20S003S-L20S-300A-300A-B- found, named as: SerialBattery(LLT/JBD)3.
INFO:root:Thu Oct 10 12:15:12 2024: 3 batteries found.
INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB3__0, scanning and storing items
INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB3__0 has device instance 4
INFO:root:Thu Oct 10 12:15:17 2024: Searching batteries: Trial Nr. 3
INFO:root:Thu Oct 10 12:15:17 2024: JBD-AP20S003S-L20S-300A-300A-B found, named as: SerialBattery(LLT/JBD).
INFO:root:Thu Oct 10 12:15:17 2024: JBD-AP20S003S-L20S-300A-300A-B found, named as: SerialBattery(LLT/JBD)2.
INFO:root:Thu Oct 10 12:15:17 2024: JBD-AP20S003S-L20S-300A-300A-B found, named as: SerialBattery(LLT/JBD)3.
INFO:root:Thu Oct 10 12:15:17 2024: JBD-AP20S003S-L20S-300A-300A-B- found, named as: SerialBattery(LLT/JBD)4.
INFO:root:Thu Oct 10 12:15:17 2024: 4 batteries found.
Traceback (most recent call last):
  File "/data/dbus-aggregate-batteries/aggregatebatteries.py", line 842, in _update
    sum(VoltagesSum_dict.values()) / settings.NR_OF_BATTERIES
TypeError: unsupported operand type(s) for +: 'float' and 'NoneType'`

regu-tech, 2024-10-10:
After a restart.sh the driver comes up again and works as expected, but it dosn't come up automatically

Dr-Gigavolt, 2024-10-10:
Hi, please replace the aggregatebaterries.py by the one I've just pushed and try. Now if '/Voltages/Sum' is None, the adjusted number of trials (and not only one as before) will be done before crash. But if the program after crash does not restart, I cannot help you. It is your local Linux issue and I'm not a Linux specialist.

Here is the code change (from line 662):

# here an exception is raised and new read trial initiated if None is on Dbus
volt_sum_get = self._dbusMon.dbusmon.get_value(self._batteries_dict[i], "/Voltages/Sum")
if volt_sum_get != None:
    VoltagesSum_dict[i] = volt_sum_get
else:
   raise TypeError(f"Battery {i} returns None value of /Voltages/Sum.")

regu-tech, 2024-10-10:
Thanks for the changes, i have replaced the file and rebooted the Cerbo. I will monitor this for the next weeks what happens then when a connection lost appears again.

onemorename, 2024-10-11:
I can confirm that the driver now successfully recovers, when serial battery looses the connection to both Packs for some time. Thanks for this fix!

closed 2024-11-07 by Dr-Gigavolt