Dr-Gigavolt/dbus-aggregate-batteries #72: DBUS product name could be None

db1pf, 2024-11-03:
I had the issue, that dbus-aggregate-batteries was not able to find both connected batteries. The reason for this was, that during iterate through list of available DBUS devices (line 317 in aggregatebatteries.py) the service with name "com.victronenergy.battery.aggregate" was found. For this service the productName read in line 319 is None which throws an exception in Line 322. This leads to abbort of device discovery.
I changed line 322 to `if productName is not None and settings.BATTERY_PRODUCT_NAME in productName:` and line 383 to `productName not None and settings.SMARTSHUNT_NAME_KEY_WORD in productName`, now it works.

I used current main branch.

Dr-Gigavolt, 2024-11-07:
Hello and thanks for reporting. I don't understand why the exception was thrown. 

In my _settings py_, line 24: _BATTERY_PRODUCT_NAME = "SerialBattery"_

If the iteration finds the _com.victronenergy.battery.aggregate_ (itself) and the _productName_ is _None_,
the statement in the line 322 _settings.BATTERY_PRODUCT_NAME in productName_ is false, not invalid.

Otherwise the program could not work at me nor other users.

 The same for the line 383:
in my _settings.py_, line 32: _SMARTSHUNT_NAME_KEY_WORD = "SmartShunt"_
the statement _settings.SMARTSHUNT_NAME_KEY_WORD in productName_ should be false if _productName_ is _None_

Am I missing something?

Dr-Gigavolt, 2024-11-07:
Hello and thanks for reporting. I have never observed it before. I tried to reproduce it in terminal and you're right. I'm going to add your proposal to my code.

[image]

closed 2024-11-07 by Dr-Gigavolt