Dr-Gigavolt/dbus-aggregate-batteries #74: Driver crashes when SOC is 0%

regu-tech, 2024-11-07:
Hello, 

we are one step ahead, it seems that the aggregate battery is shown correctly in the overview window:
[image]

but the "Aggregate Batteries" menu point is gone when all batteries are empty
[image]

The Log shows many of these errors:

`IndentationError: unindent does not match any outer indentation level
  File "/data/dbus-aggregate-batteries/aggregatebatteries.py", line 832
    sys.exit()
             ^
IndentationError: unindent does not match any outer indentation level
  File "/data/dbus-aggregate-batteries/aggregatebatteries.py", line 832
    sys.exit()
             ^
IndentationError: unindent does not match any outer indentation level
  File "/data/dbus-aggregate-batteries/aggregatebatteries.py", line 832
    sys.exit()
`

_Originally posted by @regu-tech in https://github.com/Dr-Gigavolt/dbus-aggregate-batteries/issues/48#issuecomment-2463325530_

coca7, 2024-11-07:
There was an error with this that was corrected earlier today.  Can you navigate to the directory and install it again using ".'install.sh". I think that will resolve your issue.  You might also have to put your settings back in settings.new-version.py and restart your device.

regu-tech, 2024-11-08:
Good morning,

thanks for your help, now it is working.

I have only replaced the new (changed) files since the last install and has done a local-install.sh 

best regards
Steffen

closed 2024-11-08 by regu-tech