Dr-Gigavolt/dbus-aggregate-batteries #76: cannot aggregate batteries and how did you connect it?

Mark21041981, 2024-12-15:
Goodmorning folks,

- I got myself an victron 3phase system, self installed. 
- With 3 banks LFP 205Ah, I used an 16s config.
- The BMS is JK BMS hardware version 14, latest firmware update. 
- I did manage to get one visible on the cerbo with Louisvdw script. I used an CAN type A cable and daisy chained the rest with RJ45

How to see them all on the cerbo? I want to see all 3 banks on the Cerbo. 

How did you manage to get that far?

Dr-Gigavolt, 2025-01-19:
Hello, AFAIK if they are daisy-chained, you see only one battery and you have nothing to aggregate. Can someone confirm it?

Mark21041981, 2025-01-24:
I don't know, but how do you connect the JK BMS to the Cerbo GX?

TheGuru0000, 2025-03-05:
> I don't know, but how do you connect the JK BMS to the Cerbo GX?

I connected my 2 JKBMS Hardware V15 with normal lan cables and a serial adapter using lan connectors on the BMS side to connect it to the cerboGX. You can search for videos from Andy from the offgridgarage - i think he also did videos of the older V14 and he definitely did 1 videos on how to connect the jkbms to the cerbo device / raspi with Venus OS.

Dr-Gigavolt, 2025-03-05:
I'm sorry, I hadn't seen the question about connection. Exactly, the Serial Battery and the Aggregate are designed for simple connection by serial link (TTL, RS485 or RS232). Don't neglect galvanic insulation and EMC.

closed 2025-03-20 by Dr-Gigavolt