Dr-Gigavolt/dbus-aggregate-batteries #77: With "temp" off, unable to start

nicobresseleers, 2024-12-29:
I ran into issue/bug https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/145

to resolve it , i temporary disabled:
CCCM_T_ENABLE = True -> False
DCCM_T_ENABLE = True -> False

Now aggregate will not start:

INFO:root:Sun Dec 29 21:16:01 2024: MultiPlus-II 48/3000/35-32 found.
ERROR:root:Sun Dec 29 21:16:02 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery SerialBattery(JKBMS)C2.
ERROR:root:Read trial nr. 1
.....
ERROR:root:Sun Dec 29 21:16:12 2024: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery SerialBattery(JKBMS)C2.
ERROR:root:Read trial nr. 11
ERROR:root:Sun Dec 29 21:16:12 2024: DBus read failed. Exiting.

Can you please check?

n

nicobresseleers, 2024-12-29:
PS:
the temp values are still there:
[image]

Seems you are reading in the wrong place or something?

Dr-Gigavolt, 2024-12-29:
Hi Nico,
I'm sorry, I have no time to follow the issues of Serial Battery. Does my program raise an exception when the temperature is negative? According the log it receives 'None' value. Please make a screenshot of dbus-spy of the temperatures. Perhaps there is some format problem.

nicobresseleers, 2024-12-29:
Issue with Serial Battery has been identified and resolved.. ( nothing to do with negative temp .. it's when min voltage is not set), many people complained about unidentified "low battery" errors... seems we found that one.(  and is also reported to victron)

Aggregates works, till i change these 2 parameters in serial: ( as a workaround to ignore a negative value -> Serial works)

CCCM_T_ENABLE = True -> False
DCCM_T_ENABLE = True -> False

and/or 

 read 1 temperature from the batteries ( to ignore the negative) 

[image]

So only 1 measurement for temperature (14) is available as Temp 1
[image]
and
[image]

once i start aggregate now it stops, with "NONE", seems it expects more.. quite sure if i disconnect the temp sensors ( or turn of the temperature on the BMS via the IOS app , it will also fail to start.

Can you replicate the issue?

Nico

nicobresseleers, 2024-12-29:
could replicate myself: 

I can confirm it's only the reading 1 temperature that makes aggregate stop, when reading 0 ( all) , aggregates runs again even with parameters:

CCCM_T_ENABLE = True -> False
DCCM_T_ENABLE = True -> False

hope this helps.

nico

Dr-Gigavolt, 2025-01-02:
Hi Nico, sorry, I still have no time. I'll reply in couple of days.

nicobresseleers, 2025-01-02:
> Hi Nico, sorry, I still have no time. I'll reply in couple of days.

No worries.. i have temporary set TEMP_BATTERY = 0 and it works.. but it's still a bug ( i need to set to 1 since i have a defect sensor that i cannot replace in 123).. for the time being: all OK!

Dr-Gigavolt, 2025-01-05:
Hi Nico, I cannot reproduce it, my Aggregate starts always. I don't read the paths 'System/Temperature[n]', there is nothing to aggregate. My instances of SerialBattery give 'None' of two sensors as well because they don't exist. But it doesn't matter because they are not read. The only temperature paths I read are:

"/Dc/0/Temperature"
"/System/MinCellTemperature"
"/System/MaxCellTemperature"

If you have crash during reading of temperature, at least one of these paths is 'None'. Try to identify which one.

nicobresseleers, 2025-04-05:
this came back;

INFO:root:Sat Apr  5 10:57:58 2025: 3 batteries found.
INFO:root:Sat Apr  5 10:57:59 2025: Searching Multi/Quatro VEbus: Trial Nr. 1
INFO:root:Sat Apr  5 10:57:59 2025: MultiPlus-II 48/3000/35-32 found.
ERROR:root:Sat Apr  5 10:58:00 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 1
ERROR:root:Sat Apr  5 10:58:01 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 2
ERROR:root:Sat Apr  5 10:58:02 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 3
ERROR:root:Sat Apr  5 10:58:03 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 4
ERROR:root:Sat Apr  5 10:58:04 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 5
ERROR:root:Sat Apr  5 10:58:05 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 6
ERROR:root:Sat Apr  5 10:58:06 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 7
ERROR:root:Sat Apr  5 10:58:07 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 8
ERROR:root:Sat Apr  5 10:58:08 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 9
ERROR:root:Sat Apr  5 10:58:09 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 10
ERROR:root:Sat Apr  5 10:58:10 2025: Error: unsupported operand type(s) for +=: 'int' and 'NoneType'.
ERROR:root:Occured during step Read temperatures, Battery C1.
ERROR:root:Read trial nr. 11
ERROR:root:Sat Apr  5 10:58:10 2025: DBus read failed. Exiting.

Dr-Gigavolt, 2025-04-05:
The trouble happens in this part of code, starting with line 623:

                # Temperature
                step = "Read temperatures"
                Temperature += self._dbusMon.dbusmon.get_value(
                    self._batteries_dict[i], "/Dc/0/Temperature"
                )

It tries to calculate average temperature of all batteries, but receives 'None' instead of a number. Does your batteries provide the path "/Dc/0/Temperature"? As you can see in the issue #91, some paths in new version of Serial Battery were removed. I have no time to follow that project and I run an old version. I wait for new pull request concerning #91, than I can adapt the temperature paths. But if the authors of Serial Battery will make backwards-incompatible changes, it will be a never ending story.