Dr-Gigavolt/dbus-aggregate-batteries #78: Does not find multiple Batteries

misterm15, 2025-01-04:
Hello,

running DalyBMS via dbusSerial on VRM 3.52.
I´m not able to see my three batteries.
They are named BatteryBlock_1 till _3 as the custom Name. VRM shows them in dbus-spy and also VRM portal itself.
In the log i see that the last battery_3 is recconized. Once i change the config, where _3 was the last declared one to _2 or _1 it profides me that this block has been found. But not all together.
[image]

I´ve checked presiously closed issues for some default config which is idiot proofen but seems that i´m the issue.

Could someone please help me with the change i need to do in the config?
Thanks!

Config:

# Nr. of physical batteries to be aggregated. Smart shunt for battery current is not needed and not supported.
NR_OF_BATTERIES = 3
NR_OF_CELLS_PER_BATTERY = 12
# Nr. of MPPTs
NR_OF_MPPTS = 0
# If DC loads with Smart Shunt present, can be used for total current measurement
DC_LOADS = False
# False: Current subtracted, True: Current added
INVERT_SMARTSHUNT = False

# ######################################
# ########### DBus settings ############
# ######################################

# Key world to identify services of physical Serial Batteries (and SmartShunt if available)
BATTERY_SERVICE_NAME = "com.victronenergy.battery"
#BATTERY_PRODUCT_NAME_PATH = "/CustomName"
BATTERY_PRODUCT_NAME = "BatteryBlock_1"
BATTERY_PRODUCT_NAME_PATH = "/CustomName"
BATTERY_INSTANCE_NAME_PATH = "/CustomName"
BATTERY_PRODUCT_NAME = "BatteryBlock_2"
#BATTERY_PRODUCT_NAME_PATH = "/CustomName"
#BATTERY_INSTANCE_NAME_PATH = "/CustomName"
BATTERY_PRODUCT_NAME = "BatteryBlock_3"
#BATTERY_PRODUCT_NAME_PATH = "/CustomName"
#BATTERY_INSTANCE_NAME_PATH = "/CustomName"
# Key world to identify service of Multis/Quattros (or cluster of them)
[image]
[image]
[image]
[image]

Dr-Gigavolt, 2025-01-05:
Hello, your config is wrong. You product name is 'SerialBattery(Daly)' and it is OK to short it like 'SerialBattery', see the line 322 in aggregatebatteries.py:
_if (productName != None) and **(settings.BATTERY_PRODUCT_NAME in productName**):_
It is not "BatteryBlock_1" etc. and you must not write it three times. The instances and the custom names are found automatically. You don't even need to modify the default. This is my (default) setup, I have  two pcs. with ProductName SerialBattery(Jkbms).

BATTERY_SERVICE_NAME = "com.victronenergy.battery"
# Path of Battery Product Name
BATTERY_PRODUCT_NAME_PATH = "/ProductName"
# Key world to identify the batteries (to exclude SmartShunt)
BATTERY_PRODUCT_NAME = "SerialBattery"
# if CustomName doesn't exist, set '/ProductName'
BATTERY_INSTANCE_NAME_PATH = "/CustomName"
# Key world to identify service of Multis/Quattros (or cluster of them)

misterm15, 2025-01-25:
Hello,

`NR_OF_BATTERIES = 3
NR_OF_CELLS_PER_BATTERY = 12
# Nr. of MPPTs
NR_OF_MPPTS = 0
# If DC loads with Smart Shunt present, can be used for total current measurement
DC_LOADS = False
# False: Current subtracted, True: Current added
INVERT_SMARTSHUNT = False

# ######################################
# ########### DBus settings ############
# ######################################

# Key world to identify services of physical Serial Batteries (and SmartShunt if available)
BATTERY_SERVICE_NAME = "com.victronenergy.battery"
#BATTERY_PRODUCT_NAME_PATH = "/ProductName"
BATTERY_PRODUCT_NAME = "SerialBattery"
BATTERY_INSTANCE_NAME_PATH = "/CustomName"

# Key world to identify service of Multis/Quattros (or cluster of them)
MULTI_KEY_WORD = "com.victronenergy.vebus"
# Key world to identify services of solar chargers
MPPT_KEY_WORD = "com.victronenergy.solarcharger"
# Key world to identify services of SmartShunt`

Gives me less sucess than before.

None of the batteries are detected.

Could it be something with the outdated registration Method?

[image]

misterm15, 2025-01-26:
Output looks different:

[image]

Can i specify the USB Ports directly?

[image]

Dr-Gigavolt, 2025-01-26:
Please read the code of aggregatebatteried.py from the line 307. Then you will understand that there is no reason to search for specific USB ports. I can add more diagnostics when I come back to computer.

Dr-Gigavolt, 2025-01-26:
Now you can make the next trial. If all OK, you should see as follows. If not, the battery is found, but ignored.

INFO:root:Sun Jan 26 15:34:35 2025: Searching batteries: Trial Nr. 1
**INFO:root:Sun Jan 26 15:34:35 2025: Dbusmonitor sees: com.victronenergy.battery.ttyUSB3
INFO:root:Sun Jan 26 15:34:35 2025: Correct battery product name SerialBattery(Jkbms) found in the service com.victronenergy.battery.ttyUSB3**
INFO:root:Sun Jan 26 15:34:35 2025: SerialBattery(Jkbms) found, named as: JK-BMS-1.
**INFO:root:Sun Jan 26 15:34:35 2025: Dbusmonitor sees: com.victronenergy.battery.ttyUSB4
INFO:root:Sun Jan 26 15:34:35 2025: Correct battery product name SerialBattery(Jkbms) found in the service com.victronenergy.battery.ttyUSB4**
INFO:root:Sun Jan 26 15:34:35 2025: SerialBattery(Jkbms) found, named as: JK-BMS-2.

misterm15, 2025-01-26:
Hello,

thanks for your support, unfortunately it´s not helping..
I´m not sure what the issue is feels like i did some stupid mistake but i can not see it anywhere

[image]

settings as provided/default

[image]

[image]

[image]

[image]

misterm15, 2025-01-26:
Got it working!!
Seems that it was my fault..
Config:
BATTERY_SERVICE_NAME = "com.victronenergy.battery"
#BATTERY_PRODUCT_NAME_PATH = "/CustomName"
BATTERY_PRODUCT_NAME = "SerialBattery"
BATTERY_PRODUCT_NAME_PATH = "/ProductName"
BATTERY_INSTANCE_NAME_PATH = "/ProductName"

--> both need to point to ProductName instead of a mix ( BATTERY_PRODUCT_NAME_PATH = "/ProductName" & BATTERY_INSTANCE_NAME_PATH = "/CustomName")

[image]

Thank you!

closed 2025-01-26 by Dr-Gigavolt