Dr-Gigavolt/dbus-aggregate-batteries #80: SOC won´t reset

Rellikon83, 2025-01-05:
Hi,

could someone explain why aggregate don´t reset the SOC? I don´t find my mistake in the configuration.
Thx a lot.

[image]
[image]
[settings.txt](https://github.com/user-attachments/files/18311572/settings.txt)

Dr-Gigavolt, 2025-01-05:
Hi, I don't understand your problem. Please explain.

Rellikon83, 2025-01-06:
as you can see the batteries are both at their lowest point of discharge. Aggregate batteries 10.3% SOC. I thought the script will reset the soc if on cell reach the lowest point. Could you explain how it works? My settings are attached above.

Dr-Gigavolt, 2025-01-07:
Have you set OWN_SOC true? Then the BMS SoC is ignored and my counter runs. You have to configure it correctly, read readme and comments in settings.py. I made my own SoC counter because the current measurement as well as the SoC of my BMS (JK) are useless. If your BMS is accurate, you can set OWN_SOC false.

Rellikon83, 2025-01-07:
Yes i did. in my first post you can See the settings.txt under the last picture.

Dr-Gigavolt, 2025-01-15:
Is it clarified now? Can we close the issue?

Rellikon83, 2025-01-15:
No. The Problem is the same as you can See in the pictures. The soc didnt reset if the battery has reached the minimum voltage.

Dr-Gigavolt, 2025-01-15:
Now I understand, sorry, I thought you expect the reset when the BMS's SoC is zero. What the log file says? The maximum and minimum cell voltage is shown there. You can also check with dbus-spy whether the aggregatebatterie's cell voltages correspond to your BMS app. If the voltages are read correctly it should work, see the lines 1057 and 1058 in aggregatebatteries.py:

            if (MinCellVoltage <= settings.MIN_CELL_VOLTAGE) and settings.ZERO_SOC:
                self._ownCharge = 0  # reset Coulumb counter to 0%

Rellikon83, 2025-02-01:
Yes i can See but it wont work.
The same Situation if Battery is Full. The soc wont reset if the cell voltage is reched in the settings. In my case 3.4V. In dbus-spy i can See 3.48V. 

Whats wrong?

[image]

Dr-Gigavolt, 2025-02-01:
OK, thanks for the update. Please be patient for couple of days, I suffer under terrible lack of time :-(. I will think about adding some further diagnostics into aggregatebatteries.py.

Dr-Gigavolt, 2025-02-03:
I found a mistake in your settings:
BALANCING_VOLTAGE = 3.40
MAX_CELL_VOLTAGE = 3.4
CHARGE_VOLTAGE_LIST = [
    3.45,  # January
    3.45,  # February

The rule is max(CHARGE_VOLTAGE_LIST) <= BALANCING_VOLTAGE < MAX_CELL_VOLTAGE

The SoC is set to 100% if:

- V(batt) = Ncells * BALANCING_VOLTAGE
or
- Vcell(max) = MAX_CELL_VOLTAGE

If the voltage from CHARGE_VOLTAGE_LIST for given month is lower than BALANCING_VOLTAGE, the battery won't be full charged and SoC not reset. Full charge is done only at the day of scheduled balancing.

MAX_CELL_VOLTAGE is the absolute maximum allowed voltage for most charged cell. If this is reached or exceeded, the CVL is automatically limited and DC-coupled feed-in de-activated in order to avoid further increase of that cell voltage resulting in switching off the BMS. MAX_CELL_VOLTAGE can be reached or exceeded in case of strong disbalance.

In your case normal target voltage is higher than BALANCING_VOLTAGE and MAX_CELL_VOLTAGE, therefore the algorithm cannot work properly.

I'm sorry that I haven't found it out sooner, I was always in hurry.

Rellikon83, 2025-02-15:
Thx for looking to my settings. But i think i missunderstand the rule or we should discuss a better way to bring the SoC to 100%. I want to explain.

You write: The CHARGE_VOLTAGE_LIST must be lower than BALANCING_VOLTAGE must be lower than MAX_CELL_VOLTAGE
Otherwise you write: SoC ist set to 100% if Ncells x BALANCING_VOLTAGE is reached.

In my list the Voltage is 3.45V. So this is needed to reach 3.40V (BALANCING_VOLTAGE). And BALANCING_VOLTAGE is needed to set the SoC to 100%. If i set the CHARGE_VOLTAGE_LIST to 3.37V the SoC won´t reset because BALANCING_VOLTAGE is never reached, am I right?

So i think the SoC reset rule must reset if one cell of CHARGE_VOLTAGE_LIST will reach. Otherwise it doesn´t make sense to set voltages in the list lower than BALANCING_VOLTAGE. But this is needed to push the health of batteries on sunny days because the working mode in summer should be 20% - 80%. With the voltage list we could set this very clear. In winter if we need every Ah the battery could work from 10%-90/100%. 
Balancing only make sense if cells are full. A LFP cell is full at approximately 3.37V. So BALANCING_VOLTAGE must be higher than 3.37 to do the job right and avoid cell drifting because a difference is needed to share voltage from a higher cell to the lower cell.
The BALANCING_VOLTAGE shouldn´t work together with CHARGE_VOLTAGE_LIST. The BALANCING_VOLTAGE and REPETITION should be a setting to force balancing on a long sunny period. I understand this as a cell protection mode, which makes really sense, but the combination with CHARGE_VOLTAGE_LIST gets in trouble with SoC reset.

I like the idea of the CHARGE_VOLTAGE_LIST because we can protect the battery AND give us a defined SoC of 100%. The SoC is very important to the rules of a smart house and bring the energy to the right spot such as e cars or heating pumps etc. If the SoC of aggregate_battery is not right the smart system has a lot of problems to calculate the amount of Ah which is needed to load from the grid if dynamic prices from stock echange are used.

Dr-Gigavolt, 2025-02-15:
If your voltage in CHARGE_VOLTAGE_LIST is lower than balancing voltage, the cells won't be fully charged to prolong their useful life. Why should it be reset to 100% if the real SoC is lower? At the balancing day (every BALANCING_REPETITION days) the charger tries to make the battery full (BALANCING_VOLTAGE). In case of success (N * BALANCING_VOLTAGE reached) the SoC is reset and status "balancing" is kept until the voltage difference falls below CELL_DIFF_MAX. If not managed till evening, it continues the next day(s). After that the CHARGE_VOLTAGE_LIST is used again. In haven't managed to charge the batteries full for many weeks this winter. The accumulated SoC error was just 4%. You keep it low by adjusting BATTERY_EFFICIENCY. My default value of 98.5% is evaluated with my LTO batteries. I expect your efficiency is a bit different and has to be found out and adjusted.

Dr-Gigavolt, 2025-02-15:
And of course if at least one cell reaches or exceeds MAX_CELL_VOLTAGE, the SoC is reset to 100% as well. This happens only in case of strong disbalance. The CVL is adjusted dynamically (below BALANCING_VOLTAGE) in such case and it can take days or weeks to get cells balanced. Therefore I reset SoC in such case, the battery is let's say 98-99% full, but needs a lot of time do be charger to 100% in respect to BALANCING_VOLTAGE.

Rellikon83, 2025-02-16:
> If your voltage in CHARGE_VOLTAGE_LIST is lower than balancing voltage, the cells won't be fully charged to prolong their useful life. Why should it be reset to 100% if the real SoC is lower? At the [...]

I understand, but in summer the battery capacity will be enough for 20-80% (just an example). At 80% the SoC wont be reset, because the balancing wont work. In every cycle the SoC drifts more and more.
The software in the house doesnt work with cell voltages. It takes the SoC to decide if it is better to unload the battery or to stop inverting because the price would be higher in the future. Thats why it is important to set the SoC every time from 0% to 100 % as a working mode. It isnt interesting if the battery voltage is reached or not, the software needs to know if the battery is full or not. 
In my case after a few days the real SoC was 99% but aggregate shows 64%. The software said battery shouldnt unload because the price in the evening would be higher. The decision was wrong because there was enough capacity in the battery.
Thats why i said the charge_voltage_list and of course max_voltage is needed for resetting the SoC to 100%. Balancing is just a calibration thing. And of course, if balancing voltage is reached it must reset to 100% because the battery is full.
If the charge List must be higher than balancing voltage the List doesnt make sense. 

I would try the battery efficiency. If the soc wont drift it could help.

Maybe you could arrange the script a Little bit. So the user could decide if the soc should reset by balancing or by charge List.

Rellikon83, 2025-02-19:
Ok i have set the battery efficiency to a high value (1) and to a low value (0.5). Nothing happens.
Settings: BALANCING_REPETITION = 1 day
BALANCING_VOLTAGE = 3.40
MAX_CELL_VOLTAGE = 3.45
CHARGE_LIST_VOLTAGE = 3.37

Nothing happens as you can see in the pictures. Logs ar fine.

[image]

Dr-Gigavolt, 2025-02-19:
Hello, I'm sorry, I think about you but I have desperately no time :-(. I will check it at the next opportunity. What I already wanted to write: Resetting to 100% if the battery is not full won't make you happy. Then you get troubles when the battery is empty but SoC let's say still at 20%. And in winter there is no guarantee anyway that you reach your list voltage.  As I wrote recently I have no troubles with accumulated error during a couple of weeks without reset. I use the Victron current measurement, is much more precise than my JK BMS.

Dr-Gigavolt, 2025-02-21:
Hello, it is suspicious that you set balancing every day but the balancing state is 0. It means it does not try to reach balancing voltage. What is your current CELL_DIFF_MAX? And I see the MAX_CELL_VOLTAGE is exceeded. Do you see "Dynamic CVL reduction started." somewhere in the log?

Dr-Gigavolt, 2025-02-21:
I found it out. You have OWN_CHARGE_PARAMETERS = False. My charging algorithm is deactivated, the tool takes everything from battery driver and combines it. The BALANCING_VOLTAGE, CHARGE_VOLTAGE_LIST, MAX_CELL_VOLTAGE, MIN_CELL_VOLTAGE etc. are ignored.

Rellikon83, 2025-02-22:
> Hello, I'm sorry, I think about you but I have desperately no time :-(. I will check it at the next opportunity. What I already wanted to write: Resetting to 100% if the battery is not full won't [...]

This shouldn´t happen because the MIN_CELL_VOLTAGE reached and the SoC will be set to 0%. Yes my JK is wrong too, thats why i want a SoC Reset by Voltage. It is very important to show the right SoC for a smart home. Every other scripts in my house have a look on the SoC not the Voltages.

Rellikon83, 2025-02-22:
> I found it out. You have OWN_CHARGE_PARAMETERS = False. My charging algorithm is deactivated, the tool takes everything from battery driver and combines it. The BALANCING_VOLTAGE, CHARGE_VOLTAGE_LIST, MAX_CELL_VOLTAGE, MIN_CELL_VOLTAGE etc. are ignored.

Omg. I set this to true and the SoC is 100% now. I love your script and reset functions, but i want to mention a better documentation is needed for the users. It is hard to understand which parameters must be set to let the script work. Or otherwise: If you have a AND setting in the script, think about if whether both triggers are needed. 

I set the BALANCING_REPETITION to 0. Now the Balancing should start all the time, am i right?

Dr-Gigavolt, 2025-02-22:
Shit happens 😂. You are right about the quality of the documentation. As I already mentioned, the time is my at least available goods. I believe it is better for the community to have a code with weak documentation than to have nothing at all. I would very appreciate it if someone who understands my description would rewrite it better.

Dr-Gigavolt, 2025-02-22:
You don't need to set balancing repetition to zero. With any setting it will try to balance as many days as necessary, until the goal is reached. Then charge according the list for the given period of time.

closed 2025-02-25 by Dr-Gigavolt

Rellikon83, 2025-02-26:
> Shit happens 😂. You are right about the quality of the documentation. As I already mentioned, the time is my at least available goods. I believe it is better for the community to have a code with [...]

Youre completely right. Dokumentation follows function. The Problem from the outside is: no one knows the Code. So it is hard to write down if you dont know the dependencies.
Docu is quite easy: enable everything 😜

Thx for your help.

Rellikon83, 2025-02-26:
> You don't need to set balancing repetition to zero. With any setting it will try to balance as many days as necessary, until the goal is reached. Then charge according the list for the given period of time.

This is something i dont understand Till now. The balancing voltage must be lower than the voltages in the charging List. The voltages in the charging List are in the operating Range of the battery. The balancing starts Not in the top voltage Zone. This is not right, because the balancing should be done if charging is Finished and only the differences at the top of the cell voltage should compared (top balancing).

Dr-Gigavolt, 2025-02-28:
>  the balancing should be done if charging is Finished and only the differences at the top of the cell voltage should compared (top balancing)

Exactly so it works. After the BALANCING_REPETITION days the CVL is set to N * BALANCING_VOLTAGE. If the BMS is set up correctly, it starts its job. The CVL = N * BALANCING_VOLTAGE is kept until the difference between maximum and minimum cell voltage is less or equal CELL_DIFF_MAX, solar power falls and the battery is discharged below the voltage from the list.

> The balancing voltage must be lower than the voltages in the charging List.
This is not true. You must make sure that at least one cell is completely full and the voltage begins to peak. Balancing at lower voltage is contra-productive.  Due to tolerances, two cells can have different voltage at same SoC and at this point we don't want to balance. Therefore the BMS start balancing voltage shall not be set too low.

Rellikon83, 2025-03-05:
I Must set the balancing Repetition to 1 day because the soc calculation drifts more and More. After 3 days the soc drifts about 30-50%. So i set the efficiency to 1.0 and the List voltages + balancing to 3.40V. In the Summer this would help. In Winter i Must Switch to the bms soc.

Dr-Gigavolt, 2025-03-05:
Do you know the real capacity of your cells? Have you measured it? A couple of months ago I got 20 cells to extend my batteries and cannot use them. They are around 8% below nominal. A misunderstanding with the seller who calls B grade A- grade. And - is minus, not a connection sign, this was overseen by me. My explicit capacity requirement was either forgotten or ignored by him.

Rellikon83, 2025-03-06:
I‘ve Scanned the qr codes. The production date differs, but i bought eve grade A cells. Ok i dont know it but i measured a very low inner resistance when the cells arrived. Anyway the capacity should be the same if the cell Voltage difference is Max. 0.050V when the cells reaching the top Levels. I‘ve checked the datas and aggregate gives me a lower Ah for Both batteries.
How do you calculate the soc while the battery is loading or unloading? The drift happens if the cells are between Full and empty. The soc is always behind the real Situation. E.g. If the battery is empty, soc is at 10-20%. If the battery is Full, soc is at 65-85%. Cyclecount is 280. 
is there any effect if i would reduce the capacity in the Script?

Dr-Gigavolt, 2025-03-06:
It seems the real capacity is lower or there is an issue with current measurement. Have you activated Current from Victron? Charge is calculated as a time integral of current at discharging. In case of changing there is a multiplication by efficiency.

Dr-Gigavolt, 2025-06-04:
If your BMS current measurement is not accurate and you use the Serial Battery calculation you'll have troubles with all features that are SoC-controlled. My program doesn't send own SoC to Serial Battery. At the next opportunity I can check if there some simple way to reset the own SoC in your case. But I'm too busy now, it will take some time.

TazerReloaded, 2025-06-04:
Thank you for the quick response!

I don't use any SoC controlled features in the battery driver, just cell voltages. If a cell starts to get full, the driver of that pack will dynamically reduce the CVL, which will feed the excess to grid (or reduce solar production on grid loss). If a cell gets close to empty, DCL is reduced. This works well, because the aggregator forwards those limits to the system, and the drivers have a few neat features around these functionalities (charge and discharge current tables, voltage adjustments with an integral control loop for each pack).

Therefore, I want to leave your charge features disabled, but still use the SoC calculation. I think a simple voltage threshold would be sufficient, like for example `RESET_FULL = 3.45`, `RESET_EMPTY = 3.00` and `RESET_AVERAGE = True`, and the counter then resets when the cell average (or the first cell for False) reaches those thresholds.

TazerReloaded, 2025-06-07:
I wrote a simple reset logic that uses `MIN_CELL_VOLTAGE` / `MAX_CELL_VOLTAGE` (aggregatebatteries.py, line 1120):
```python
        elif settings.OWN_SOC:
            # OWN_CHARGE_PARAMETERS is disabled, but OWN_SOC is used
            # reset SoC on MIN_CELL_VOLTAGE / MAX_CELL_VOLTAGE if present
            if settings.MIN_CELL_VOLTAGE and MinCellVoltage <= settings.MIN_CELL_VOLTAGE:
                self._ownCharge = 0  # reset Coulumb counter to 0%
            elif settings.MAX_CELL_VOLTAGE and MaxCellVoltage >= settings.MAX_CELL_VOLTAGE:
                self._ownCharge = InstalledCapacity  # reset Coulumb counter to 100%
```
I will open a PR after testing, could take some time though.