Dr-Gigavolt/dbus-aggregate-batteries #81: battery.agggregator has no device instance

felkep, 2025-01-13:
Hello Dr-Gigavol,

I ran through you issue board and solved most of my mistakes.

But there is still (*at least*) one left which is blocking the instance from running.

I've managed that all batteries where found (with the changed names) and that there was a = missing for the charging Voltage.

Can you please help me with this problem?

I'm running a system with 4 batteries (JBD BMS) and 3 MPPTs

The SerialBattery Firmware is 1.0.20230531
The BatteryAggregate Version is 3.1.

[image]

Dr-Gigavolt, 2025-01-15:
Perhaps you've mixed new aggregatebatteries.py with an old settings.py? The missing item looks like this:

# Set up how full the battery has to be charged in given month
CHARGE_VOLTAGE_LIST = [
    2.45,  # January
    2.45,  # February
    2.42,  # March
    2.42,  # April
    2.40,  # May
    2.40,  # June
    2.40,  # July
    2.40,  # August
    2.42,  # September
    2.42,  # October
    2.45,  # November
    2.45,  # December
]

Dr-Gigavolt, 2025-01-19:
Have you already solved the problem? I recommend you to install the newest version, there some old bugs fixed. Due to lack of time I cannot follow the old stuff.

felkep, 2025-01-19:
Thank you! 
It finally worked.
I deleted the whole old installation.
And reinstalled it again

Thank you! 
BR Pasxale

closed 2025-01-19 by felkep